tsduck/tsduck #797: [BUG]  Timeref Problem

Sicand, 2021-05-28:
#### Bug description:
Hi , when i use timeref to auto update TOT & TDT tables i had a problem . After some time it go out of sync , and after 3 days the time on tables there is 3~4 minutes  of real clock (example 15:25 when UTC is 15:28) aand i have problem whit EPG generated by an external script . Any Suggestion or there is a way to sync time every time interval ? 

I'm using a Rockchip Board whit Armbian (Ubuntu)

lelegard, 2021-05-28:
Hi @Sicand 

Please be specific. If you want to enter a bug report, please use the "Bug Report" template after clicking on "New Issue". Please provide the exact tsp command, the context (offline processing, real time stream), the type of stream, the TSDuck version, the operating system.

It is virtually impossible to diagnose an issue with so few details.

Sicand, 2021-05-28:
#### Bug description:
Hi , when i use timeref to auto update TOT & TDT tables i had a problem . After some time it go out of sync , and after 3 days the time on tables there is 3~4 minutes of real clock (example 15:25 when UTC is 15:28) aand i have problem whit EPG generated by an external script 

#### How to reproduce:
tsp --realtime -m --ignore-joint-termination --bitrate 8042499 --control-port 4445 \
-I null \
-P regulate --bitrate 8042499 --packet-burst 14 \
-P mux -b 400000 ts/h265400.ts \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P merge --no-psi-merge --pcr-reset-backwards --incremental-pcr-restamp --no-smoothing 'tsp -r --verbose -m -I ip --timestamp-priority kernel-tsp 230.0.0.1:4000 -O file' \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P merge --no-psi-merge --pcr-reset-backwards --incremental-pcr-restamp --no-smoothing 'tsp -r --verbose -m -I ip --timestamp-priority kernel-tsp 230.0.0.2:4000 -O file' \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P duplicate 259=359 \
-P merge --bitrate 900000 --no-psi-merge --pcr-reset-backwards --incremental-pcr-restamp --no-smoothing 'tsp -r --verbose -m -I ip --timestamp-priority kernel-tsp 230.0.0.3:4000 -O file' \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P merge --no-psi-merge --pcr-reset-backwards --incremental-pcr-restamp --no-smoothing 'tsp -r --verbose -m -I file ts/music.ts --infinite -P regulate -O file' \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P mux -j -b 400000 ts/net.ts \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P mux -j -b 500000 ts/service.ts \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P merge --no-psi-merge --pcr-reset-backwards --incremental-pcr-restamp --no-smoothing 'ffmpeg -loglevel quiet -re -y -i "xxxxxxxxxxx.mp3" -vn -acodec mp2 -ac 2 -b:a 128000 -ar 44100 -fflags genpts -f mpegts -mpegts_start_pid 0x104 -mpegts_flags "system_b" - ' \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P merge --no-psi-merge --pcr-reset-backwards --incremental-pcr-restamp --no-smoothing 'ffmpeg -loglevel quiet -re -y -i "xzsxxxxxxxxxxxxxxxxxxxxxxxxxxxx.m3u8" -vn -acodec mp2 -ac 2 -b:a 128000 -ar 44100 -fflags genpts -f mpegts -mpegts_start_pid 0x107 -mpegts_flags "system_b" - ' \
-P filter --negate -p 0 -s \
-P filter --negate -p 17 -s \
-P filter --negate -p 100 -s \
-P filter --negate -p 4096 -s \
-P inject pat.xml --pid 0 --bitrate 4500 \
-P inject sdt.xml --pid 17 --bitrate 10000 \
-P inject nit.xml --pid 16 --bitrate 3000 \
-P inject tdt.xml=1000 tot.xml=15000 --pid 20 --bitrate 3000 \
-P timeref --start system \
-P mux -b 48500 EPG/epg.ts \
-P inject pmt-street1.xml --pid 100 --bitrate 15000 \
-P inject pmt-street2.xml --pid 200 --bitrate 15000 \
-P inject pmt-street3.xml --pid 300 --bitrate 15000 \
-P inject pmt-street4.xml --pid 110 --bitrate 15000 \
-P inject pmt-street5.xml --pid 120 --bitrate 15000 \
-P inject pmt-street6.xml --pid 130 --bitrate 15000 \
-P inject pmt-street7.xml --pid 86 --bitrate 15000 \
-P inject pmt-street9.xml --pid 201 --bitrate 15000 \
-P inject pmt-street10.xml --pid 140 --bitrate 15000 \
-P inject pmt-street12.xml --pid 111 --bitrate 15000 \
-P inject pmt-street13.xml --pid 112 --bitrate 15000 \
-P pcradjust --pid 258 --pid 267 --pid 269 --pid 265 \
-P pcrverify \
-P pcrbitrate \
-P continuity -f \ ......

Output realtime whit hides

#### Expected behavior:
An auto correct time after some hour 

#### Errors and logs:
None
~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: 
 - OS version: <Ubuntu 20.10 ARM64>
 - TSDuck full version: <!.24-2086>
 - Installation type: <build from source -->

#### Additional information:
Now i updated to 3.27-2399 and reduced tdt/tot bitrate to test if it correct the problem

lelegard, 2021-05-28:
Thanks for the details. Currently, the `timeref` plugin and most other TSDuck plugins are designed to work on live streams and offline files indifferently. The only time reference for the stream is based on the number of packets and nominal bitrates. For offline files, this is the only reasonable reference. For live streams, it is fine for all test activities. But for 24x7 production, small imprecisions accumulate over days. This is probably what you are experimenting. 

With version 3.26, all computations on bitrates are done on fixed-point types (one decimal) over a 64-bit representation. The precision is better. So, you should see some improvements with version 3.26 and higher.

That being said, I see that you are using this plugin as a permanent time generator over a template XML representation of the TOT and TDT. For those usages, I should probably add a specific option. I will consider this.

Sicand, 2021-05-28:
Thanks for your fast reply . Now i have this annoying  problem . Whit last version after some minutes all stop whit this error . 
"tsp: dtv/codec/tsAccessUnitIterator.cpp:133: bool ts::AccessUnitIterator::next(): Assertion `_nalunit < _data + _data_size' failed."

lelegard, 2021-05-28:
Could you extract a piece of TS reproducing this?

Sicand, 2021-05-28:
No problem but send it tomorrow .

Sicand, 2021-05-29:
This is the test file ; the end of file is the moment when tsduck stop.
https://drive.google.com/file/d/1s2kuMLV_3779eSKd9gaNFsY9BpUD2bjN/view?usp=sharing

lelegard, 2021-05-30:
Could you indicate which command makes tsp crash using that input file?

I tried various combinations of plugins without a crash and I do not have the environment to reproduce your large command with many merged sources. I need some reproducible tsp command which crashes, using that test file.

My tests are done on a Mac with an Intel CPU, compiled using clang 12.0.5. You mentioned you used an Arm board. Is this a 64 or 32-bit distro and with which version of gcc? The only Arm environment I can use is a Raspberry Pi using a 32-bit Raspbian distro using gcc 8.3.0.

Sicand, 2021-05-30:
I'm trying to reproduce the error but it is random . Before the update i launch the command posted , and the other 2 input separately for a mpeg2 and H264 preencoded file in sequence whitout problem . Now i'm trying disabling one of the 2 service (mpeg2) and it go normaly . When launch this after some minutes it go off . I will post command later.

lelegard, 2021-05-30:
Hi @Sicand,

Given the randomness of the error, I tried another approach, finding in which conditions the assertion shall fail. In fact, the assert was a bit too strict (`<` instead of `<=`) but it was never triggered so far. I tried to understand which conditions should be met to trigger the failure.

In theory, the failure occurs in case of a truncated NALunit in an AVC/HEVC/VVC stream when the truncation occurs right after a start code prefix 00 00 01. Using the plugin `craft`, I recreated such a packet (see `test-097` in the [test suite](https://github.com/tsduck/tsduck-test)) and I got the same assertion failure.

So, the only hypothesis I can find is that your streams occasionally have such truncated NALunits. 

I have fixed the assertion. You may pull and rebuild. Please let us know if the problem disappears.

lelegard, 2021-05-30:
And BTW I added an option `--system-synchronous` in the `timeref` plugin so that the TOT and TDT remain synchronous with the system time (applicable to live streams only, of course, not offline file processing).

Sicand, 2021-05-30:
Many thanks for the solution and your creation , and for the added option , very glad for this . Compiled on I3 ,started and whit new option and all services active no problem for 40 minutes in waiting for compile end on Arm64 board for testing there . 

_

> So, the only hypothesis I can find is that your streams occasionally have such truncated NALunits.

_

I think yes becouse there are a lot of pre encoded andf think cutted file randomly put on service in sequence and some in infinty loop .

Sicand, 2021-05-31:
All returned perfectly now , thanks.

A question , analizing the muxed ts whit TSreader in the h264 e 265 services it report a bitrate too large for the value input. 1..95Mbit instead of 0.9Mbit & 3,85 instead of 0.4 . Is that a problem of muxer ?

lelegard, 2021-05-31:
> A question , analizing the muxed ts whit TSreader in the h264 e 265 services it report a bitrate too large for the value input. 1..95Mbit instead of 0.9Mbit & 3,85 instead of 0.4 . Is that a problem of muxer ?

The `merge` plugin merges packets as they come. The plugin cannot "invent" new packets. So, either the problem comes from the merge source (higher than what you expect) or the way TSReader evaluates the bitrate of the various elementary streams. The `merge` plugin is rather a _merger_ than a _muxer_ (this has been debated many times here). Depending on options and stability of the input source, the merge can be not very smooth, alternating bursts and quieter periods. This may influence the way TSReader evaluates the bitrate. Does it report an average bitrate, a burst / maximum bitrate, over which time granularity?

lars18th, 2021-05-31:
Hi @Sicand ,

Only to know it: You say that you're using an external script to generate the EPG data. Can you share this script? It will be very interesting if the TSDuck tool can inject EPG data from XMLTV or similar sources (in fact because a lot of opensource tools exist to work with this format). Futhermore, I'm still waiting for a plugin to generate the p/f data from the schedule EPG (see #727).

Sicand, 2021-05-31:
Before using tsduck i used opencaster and created some basic bash & python
script (ffmpeg for clip duration) for on-air/next music video and same for
a other channel with parental control ; when i find your TS-duck and tested
the more stability i adopted It instead of opencaster and adapted epg part
. No problem to share the scripts , but i have to clean the code

Sicand, 2021-05-31:
For tsreader "problem" i explain Better later ; Only the nominal value Is
wrong for AVC and Hevc (no problem for MPEG2) , the real bitrate Is
correct. Try the recorded TS on tsreader