tsduck/tsduck #798: [Q]-[ERROR] * Error: srt: error during srt_connect: Connection setup failure: connection rejected

AlexandreLicinio, 2021-05-29:
Hello all, 

I have issue when processing a **MPTS** srt, **zap** a specific service then outputs the service to a **listener srt**.

```
tsp -I srt --caller 127.0.0.1:20151 -P zap "Euronews English HD" -P regulate -O srt --listener 12345 --transtype live --messageapi
```
i get this error:
`* Error: srt: error during srt_connect: Connection setup failure: connection rejected`

But the service exists and tsp process correctly the whole mpts by doing:
`tsp -I srt --caller 127.0.0.1:20151 --transtype live --messageapi -P until --seconds 10 -P analyze -O drop`

[image]
The service currently running and decoding by an Ateme DR5000, srt (from satellite astra feed, asi to srt with a sencore gateway).

The server where tsp is installed has no firewall or any weird network rule, i just don't get why i have this error.
```
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.27-2403
Built May 28 2021 - 23:09:42
Using GCC 9.3.0, C++ std 2011.03
Web library: libcurl: 7.68.0, ssl: OpenSSL/1.1.1f, libz: 1.2.11
SRT library: libsrt version 1.4.2
Dektec: This version of TSDuck was compiled without Dektec support
```

Thanks for you help

lelegard, 2021-05-29:
Hi @AlexandreLicinio 

It is an SRT error and I am not an SRT specialist but I can try to help.

> `* Error: srt: error during srt_connect: Connection setup failure: connection rejected`

Since you use the `srt` plugin on input and output, we don't know which one reports the error. You may use `tsp --log-plugin-index` to display the plugin index with each error message coming from a plugin (index 0 is input, index n-1 is output, ie. the last one). That being said, `srt_connect` is used in caller mode, not in listener mode. So, the error must come from the input one.

So, you mention that this one fails:
~~~
tsp -I srt --caller 127.0.0.1:20151 ...
~~~
while this one succeeds:
~~~
tsp -I srt --caller 127.0.0.1:20151 --transtype live --messageapi ...
~~~
Did you try the first command with `--transtype live --messageapi` ?

lars18th, 2021-05-31:
Hi @AlexandreLicinio ,

Another comment (not related to your issue): If you're reading from a realtime source (in this case a SRT source) then **not** use the `-P regulate` plugin. Perhaps you (as me too time ago) think that this will work as a "jitter remover" plugin. But this is not true. In fact, it only can add delays when the source is realtime, and I'm sure you don't want this. Futhermore, the SRT protocol will handle the jitter, so it's useless to try to de-jitter the stream.

I hope this helps you.
Regards.

ethouris, 2021-06-02:
@lelegard BTW you might want to read and display the rejection reason. When this rejection error code is reported, you can read additionally the rejection reason using `srt_getrejectreason` and turn it into a readable message by `srt_rejectreason_str`.

lelegard, 2021-06-02:
@ethouris, thanks, I will improve the error reporting in that case.

lelegard, 2021-06-02:
The rejection reason has been added in commit b85904f241bc0309f29b366a53604600e23e69fd.
I did find a way to exhibit this error message though (but I admit I didn't search much to reproduce a rejection failure)

ethouris, 2021-06-03:
SRT can reject connections typically by the following reasons:
 - the transmission mode and transmission-related settings are different or incompatible (such as those for file/live and message/stream mode, as well as incompatible packetfilter/FEC settings)
 - no password or incorrect password
 
So, it's enough to make this difference in settings and you should see the message.

lelegard, 2021-06-03:
Thanks @ethouris, it works as expected:
~~~
$ tsversion
3.27-2413
$ tsp -I srt --caller localhost:12345 --passphrase 1234567890 -O drop
* Error: srt: error during srt_connect: Connection setup failure: connection rejected, reject reason: Incorrect passphrase
~~~