tsduck/tsduck #800: [BUG] --add-input-stuffing does not work in combination with -P zap

moving-digital, 2021-06-02:
Hi @lelegard 

#### Bug description:
I am receiving a MPTS with 8 services (80 Mbps), which I want to extract only 1 (~ 8Mbps), and add a little bit a stuffing, say 500 Kbps.

I would like to avoid using `--stuffing`, because I do not need 70 Mbps of null packets, just 500kbps will do.

I am trying to use --add-input-stuffing but it seems that it is not taken into effect when combined with zap plugin.

The below does not produce any null packets:

```
tsp --verbose --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 14/24 --max-flushed-packets 7 " +
"-I ip " + msg.input_multicast_ip + " -l " + msg.local_address_input + " " +
"-P zap " + msg.input_service_id + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port + " --delete-files " + 
"-O ip " + msg.output_multicast_ip + " -l " + msg.local_address_output + " --enforce-burst";
```

I have tried with different stuffing, the result is the same:
```
--add-input-stuffing 1/10
--add-input-stuffing 1/15
--add-input-stuffing 14/24 
```

#### How to reproduce:
The TSP command is attached here-above.

#### Expected behavior:
--add-input-stuffing should work in combination with `zap` plugin

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: Linux 
 - OS version: Raspbian 10 Buster
 - TSDuck full version: 3.26-2349
 - Installation type: official binary

lelegard, 2021-06-02:
Hi @moving-digital 

This is the intended documented behaviour, not a bug.

The option `--add-input-stuffing` is a global tsp option which adds stuffing at the input of the chain, between packets from the input plugin. Then, all plugins process the stream as usual, considering the additional null packets as part of the stream, not making any difference with other null packets coming from the input plugin.

The plugin `zap` extracts one service and removes everything else. Why would it differentiate null packets which were added because of the option `--add-input-stuffing` and null packets which came from the input plugin?

If you want to zap on one service and keeps some stuffing, I would suggest something like this:
~~~
tsp -I ... -P zap --stuffing ... -P reduce --target-bitrate 8,500,000 ...
~~~
No need to use `--add-input-stuffing`  at tsp level, `-P zap --stuffing` will leave ~72 Mb/s of stuffing.

You may read the user's guide for other options in plugin `reduce`.

moving-digital, 2021-06-02:
Thank you @lelegard 

Actually there will be many use cases. What I explained earlier on was just an example. 

What if I want to keep 500 kbps of null packets when the bitrate after zap is used, is not known in advance?

lelegard, 2021-06-02:
> Alike reduce which reduces bitrate, I believe my use-case would need a kind of increase plugin, which basically, increases the bitrate...

As explained in the user's guide (tsp reference section), a plugin can modify or delete a packet, not create one. So, there is no possibility for an `increase` plugin.

>  I may not have any null packet, and still need to add 500 Kbps, while not knowing the bitrate in advance.

It depends on what you mean by "not knowing the bitrate in advance". If you have a VBR stream and need to adjust +500 kb/s in realtime, I have no solution. But if you mean that you want a script which runs on any stream the bitrate of which you do not know in advance, there is a solution: scripting. Tools like `tsanalyze` or the corresponding `analyze` plugin have "normalized" outputs which are designed to be used by scripts. The chapter 5 of the user's guide contains many examples of scripting using an initial analysis phase (offline TS file or a few seconds of a live stream), using its normalized output to get what you need (PID values, bitrates of TS, service or PID, etc) and then run tsp commands using those values.

In your case, get the bitrate of the service, add 500000 and use that value in plugin `reduce`.

Many examples use the old "normalized output" which was designed to be used with grep & sed. Now, the TSDuck commands and plugins can output their analysis in JSON, to be used with jq in scripts.

moving-digital, 2021-06-03:
> It depends on what you mean by "not knowing the bitrate in advance". If you have a VBR stream and need to adjust +500 kb/s in realtime, I have no solution. 

Basically, this is the use case. I should have be clearer from the start. However, the idea it is to:
- Downlink a MPTS containing say 10 VBR services
- Extract 1 service with `zap`
- Only keep the video PID, and remove all the audio + subs PIDs, but keep the stuffing of these PIDs
- Use the stuffing ("generated" from the audio + subs) in order to create a SCTE-35 PID

The reason behind the 500 Kbps (I took some margin) is to make sure I have enough null packets to inject SCTE-35. Since the first idea I had won't work because of the VBR scenario, will the above (using the stuffing generated from removal of audio+subs PID) work?

lelegard, 2021-06-03:
OK, if your ultimate goal is SCTE 35 insertion, then my feeling is that you are trying to over-control while a more pragmatic approach would be simpler.

The bandwidth of a SCTE 35 PID is extremely small, definitely the smallest bandwidth ever seen in digital TV. The average bandwidth is almost zero and the peak bandwidth is one packet per second (1.5 kb/s). So, trying to control a precise large stuffing bandwidth just for later SCTE 35 insertion is useless.

With tsp, the common practice to insert data in the stream is the following in sequence:

- Ensure you have enough stuffing, either with `--add-input-stuffing` if you cannot remove anything from a stream without stuffing or replacing useless PID's with stuffing (`-P zap --stuffing` in your case). You don't care about the amount of stuffing, the more you have, the more freedom you will have to insert packets at precise points.
- Insert what you have to insert (`spliceinject` in your case).
- Remove all unused stuffing: `-P filter --negate --pid 0x1FFF`

Pragmatism, not over-engineering...

moving-digital, 2021-06-03:
Gotcha,

You're a genius! I'm using the below command and it just works like a charm and does exactly what I need. 

tsp --verbose --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 1/10 --max-flushed-packets 7 " +
"-I ip " + msg.input_multicast_ip + " -l " + msg.local_address_input + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 6000/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port + " --delete-files " + 
"-P filter --negate --pid 0x1FFF " +
"-P zap " + msg.input_service_id + " " +
"-O ip " + msg.output_multicast_ip + " -l " + msg.local_address_output + " --enforce-burst";

A big thanks again!

lelegard, 2021-06-03:
I would suggest:
~~~
tsp --verbose --bitrate " + msg.input_bitrate_mbps*1000000 + " --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 " +
"-I ip " + msg.input_multicast_ip + " -l " + msg.local_address_input + " " +
"-P zap --stuffing " + msg.input_service_id + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 6000/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port + " --delete-files " + 
"-P filter --negate --pid 0x1FFF " +
"-O ip " + msg.output_multicast_ip + " -l " + msg.local_address_output + " --enforce-burst";
~~~
Modifications:
- Remove `--add-input-stuffing`, useless with `-P zap --stuffing`.
- Fine tuning of input latency: ` --initial-input-packets 7 --max-input-packets 7`.
- Use `-P zap --stuffing` from the beginning to create stuffing while preserving the bitrate.

moving-digital, 2021-06-03:
HI @lelegard 

After thorough tests, I found out that using `--initial-input-packets 7` and `--max-input-packets 7` actually breaks TSDuck.
it gets KO after ~10min with the following error message: `* ip: leaving multicast group`.

However, when I do not use these 2 options, TSDuck never stops and works as expected. 

Would you like to get the TS capture ? I can provide you with.

lelegard, 2021-06-03:
> After thorough tests, I found out that using --initial-input-packets 7 and --max-input-packets 7 actually breaks TSDuck.
It gets KO after ~10min with the following error message: * ip: leaving multicast group.

This is quite unexpected. Isn't there any other error message? The message you mention is just informational and is issued in verbose mode when a multicast socket is closed. Since an IP input plugin has no "end of file", it runs for ever. If the socket is closed, it must be because either there was another error on input or some other plugin decided to terminate. So, there must be some other error.

The options `--initial-input-packets 7 --max-input-packets 7` are just a way to make sure that we never wait for more than one UDP packet (usually 7 TS packets) to pass data. In the case of UDP/IP, this is the best way to decrease the overall latency (at the expense of some CPU power).

moving-digital, 2021-06-03:
Hi @lelegard 

I can confirm that TSDuck breaks when using my TS capture, with a **packet size = 7.**

The messages (the list is exhaustive) I keep getting when it breaks is:
* spliceinject: received message, 444 bytes, from 127.0.0.1:36407
* spliceinject: enqueuing SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate
* spliceinject: injecting SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate, current PTS: 0x0E9C871EE
* ip: leaving multicast group 225.1.1.1 from local address 192.168.1.123

The complete TSDuck command is:

```
pkt = 7;
tsp --verbose --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 1/10 --initial-input-packets " + pkt + " --max-input-packets " + pkt + " --max-flushed-packets " + pkt + " " +
"-I ip " + msg.input_multicast_ip + " -l " + msg.local_address_input + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port + " --delete-files " + 
"-P zap " + msg.input_service_id + " " +
"-P filter --negate --pid 0x1FFF " +
"-P regulate --packet-burst " + pkt + " " +
"-O ip " + msg.output_multicast_ip + " -l " + msg.local_address_output + " --packet-burst " + pkt + " --enforce-burst";
```

Let me know if you want me to provide you with the TS, for analysis/troubleshoot purpose.

lelegard, 2021-06-03:
> Let me know if you want me to provide you with the TS, for analysis/troubleshoot purpose.

Yes please. Please make sure that you can reproduce the problem with the provided materials.

This is weird because this does not make sense. All messages are informational (verbose mode) only. There is no error. And all entities are independent. The two plugins `ip` and `spliceinject` executes independently. The tsp input tuning options affect only the tsp layer and are completely independent from the plugin execution. So, all in all, it just does not make sense. There may be something worrying that must be investigated.

moving-digital, 2021-06-03:
Hi @lelegard 

Tried again just now, and it broke again after ~10min and packet size = 7.

The generic command is:

```
tsp --verbose --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 1/10 --initial-input-packets " + pkt + " --max-input-packets " + pkt + " --max-flushed-packets " + pkt + " " +
"-I ip " + msg.input_multicast_ip + " -l " + msg.local_address_input + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port + " --delete-files " + 
"-P zap " + msg.input_service_id + " " +
"-P filter --negate --pid 0x1FFF " +
"-P regulate --packet-burst 7 " +
"-O ip " + msg.output_multicast_ip + " -l " + msg.local_address_output + " --packet-burst 7 --enforce-burst"
```

The command with **packet size = 7** and all the parameters instantiated (TSDuck breaks):
`tsp --verbose --bitrate 100000000 --add-input-stuffing 1/10 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 192.168.1.123 -P pmt --service 1004 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject --service 1004 --udp 60001 --delete-files -P zap 1004 -P filter --negate --pid 0x1FFF -P regulate --packet-burst 7 -O ip 225.1.1.1:22222 -l 192.168.1.123 --packet-burst 7 --enforce-burst`

The command with **packet size = 14** and all the parameters instantiated (TSDuck does not break):
`tsp --verbose --bitrate 100000000 --add-input-stuffing 1/10 --initial-input-packets 14 --max-input-packets 14 --max-flushed-packets 14 -I ip 225.1.1.1:11111 -l 192.168.1.123 -P pmt --service 1004 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject --service 1004 --udp 60001 --delete-files -P zap 1004 -P filter --negate --pid 0x1FFF -P regulate --packet-burst 7 -O ip 225.1.1.1:22222 -l 192.168.1.123 --packet-burst 7 --enforce-burst`

Download link containing the TS capture (720MB):
https://wetransfer.com/downloads/a7d4a5df756aae89febdb9bbc1bf119420210603141640/af9f07c10d12f8c467fbe8a69351964720210603141704/92ca5c

lelegard, 2021-06-03:
Thanks, I have reproduced and analyzed the issue. There was indeed something worrying, a rare bug which happened only when all the following conditions were simultaneously met:

- Using `--add-input-stuffing`.
- The next read operation is exactly before the wrap-up point of the large circular buffer
- This read is at start of interlaced input stuffing, not plugin input.
- The amount of packets at the end of the buffer exactly fits the amount of interlaced input stuffing for that read.

Needless to say that those conditions were rarely encountered and even more rarely reproduceable, except when all input operations always had a deterministic constant size (UDP input with 7 TS packets read size from the buffer) and the modular arithmetic of the read size combined with the buffer size and the two proportions of the `--add-input-stuffing` ended up with the above combination of conditions after a deterministic amount of input packets (and consequently a deterministic amount of time depending on the input bitrate).

I just fixed it. 

I must thank you for accidentally running into this one and apologize for the loss of time.

That being said, if you do not want to rebuild (or wait for tomorrow's nightly build), you should follow my advice from a previous post in this thread. You do not need `--add-input-stuffing` but, instead, play with the stuffing from the zap plugin.

The correct command should be:
~~~
tsp --verbose --bitrate " + msg.input_bitrate_mbps*1000000 + " --initial-input-packets " + pkt + " --max-input-packets " + pkt + " --max-flushed-packets " + pkt + " " +
"-I ip " + msg.input_multicast_ip + " -l " + msg.local_address_input + " " +
"-P zap --stuffing " + msg.input_service_id + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port + " --delete-files " + 
"-P filter --negate --pid 0x1FFF " +
"-O ip " + msg.output_multicast_ip + " -l " + msg.local_address_output + " --packet-burst 7 --enforce-burst"
~~~

Notes:
- No more `--add-input-stuffing` (no more termination issue as a nice side effect).
- Create stuffing from `zap --stuffing` as soon as the first plugin.
- Remove stuffing later (in your case, it was useless because it was set after a zap which already removed all stuffing).
- Removed the `regulate` plugin. Your stream is already self-regulated from the input (this is not an offline file). Adding a `regulate` plugin on a self-regulated stream can only make things worse.

moving-digital, 2021-06-04:
Hi @lelegard 

Ho wow, lightning speed fix as usual. You must not thank me: **I must thank you once again!**

- I may still need to use `--add-input-stuffing` from the beginning as the input stream can actually be anything: MPTS, SPTS, CBR, VBR, some null packets, no null packets at all, etc.
- Meaning to say, I need to generate a tiny bit of stuffing in order to insert the SCTE-35 PID in the selected service
- Once the service has its SCTE-35 PID inserted, I can finally use zap to select it
- And finally I can finish with `filter --negate --pid 0x1FFF` in order to reduce the overall bitrate of the TS, to the bitrate of the zapped service only. 

Noted for regulate, that was part of the test I was doing to narrow down/try to find where the issue came from. Will remove it, thanks heaps!

moving-digital, 2021-06-05:
Hello @lelegard 

May I know in which build the fix will be available?

lelegard, 2021-06-05:
Version 3.27-2415 or higher. 
Available here: https://tsduck.io/download/prerelease/

moving-digital, 2021-06-07:
> Version 3.27-2415 or higher.
> Available here: https://tsduck.io/download/prerelease/
> 
> EDIT 1: I started a manual build on Raspbian
> EDIT 2: Raspbian 10 packages are available

Thank you so much !

moving-digital, 2021-06-08:
> > Version 3.27-2415 or higher.
> > Available here: https://tsduck.io/download/prerelease/
> > EDIT 1: I started a manual build on Raspbian
> > EDIT 2: Raspbian 10 packages are available
> 
> Thank you so much @lelegard ! I'm gonna try this all now.

Working like a charm @lelegard , thanks again, I'm gonna close this ticket!

moving-digital, 2021-06-27:
Hello @lelegard 

Everything is still working fine but I have few questions with regards to the below three Global tsp options:
--initial-input-packets 
--max-input-packets 
--max-flushed-packets

The command I am using is the following:
```
tsp --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 1/10 --initial-input-packets msg.pkts --max-input-packets msg.pkts --max-flushed-packets msg.pkts " +
"-I ip " + msg.mcast_in_main_sw + " -l " + msg.local_address_input_main_sw + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port_main_sw + " --delete-files " + 
"-P zap " + msg.input_service_id + " " +
"-P filter --negate --pid 0x1FFF " +
"-O ip " + msg.mcast_out_main_sw + " -l " + msg.local_address_output_main_sw + " --packet-burst 7 --enforce-burst";
```

With some Live TS streams I am using, I got the following results at the output of TSP:
- Having a number of packets = 7 for the above three global tsp options create a lot of continuity errors. 
- However, if I increase the number of packets to 128, there is no continuity errors anymore.

I would need some guidance to understand if playing with --packet-burst and --enforce-burst for the `ip output` plugin may help to reduce the number of CC errors if increasing that number, while reducing the number of packets for the global TSP options?

Basically, I would like to know if there is a relationship with the three options from the global TSP settings, and the packet burst options from the ip output plugin?

My end game is to be able to get the a CC error count = 0, while trying to maintain the lowest latency.

lelegard, 2021-06-27:
> Having a number of packets = 7 for the above three global tsp options create a lot of continuity errors.

This would never happen on a regular PC. I just think that the Raspberry Pi CPU is too slow for the requested reactiveness. Please try the same command on a Linux PC with a decent Intel CPU and let us know the result.

The CC errors are obviously due to missed UDP datagrams. A UDP datagram usually contains 7 TS packets. So, using 7 for all parameters means that maximum reactiveness, maximum throughput, minimum latency are guaranteed. But this comes with a a cost. The 8 threads (6 plugin threads, 1 main, 1 logging) are shaking the poor CPU with few cores. The input plugin thread is probably not scheduled on time and UDP datagrams are missed.

By increasing the buffering, you reduce the CPU load (and the risk of input thread not being ready on time) but you increase the latency and the output will burst (not a good idea with `ip` output plugin).

So, if you want to keep the low latency, smooth output, use 7 everywhere and a more powerful CPU (a regular PC).

Alternatively, before changing the hardware, you may try to increase the socket buffer size (option `--buffer-size`) in the input `ip` plugin. More missed UDP datagrams will be buffered in the kernel. This is worth a try, but without guarantee.

If you want to keep your Raspberry Pi, increase the parameters. I cannot give a value, you need to experiment in your configuration. But the output will burst. In other words, nothing is free...

moving-digital, 2021-06-29:
Thanks @lelegard 

I have done two tests:

1. Deleted as many plugins as possible (Only kept `input` and `output` plugin), and keeping the three global tsp options with packets =7. The problem remains (the number of CC errors keep increasing) while the CPU and RAM usage is super duper low, as shown in the screenshot:

[image]

Command is now:

```
tsp --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 1/10 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 " +
"-I ip " + msg.mcast_in_main_sw + " -l " + msg.local_address_input_main_sw + " " +
"-O ip " + msg.mcast_out_main_sw + " -l " + msg.local_address_output_main_sw + " --packet-burst 7 --enforce-burst";
```

2. Kept all the plugins, the CPU and RAM are still very low:

```
tsp --bitrate " + msg.input_bitrate_mbps*1000000 + " --add-input-stuffing 1/10 --initial-input-packets 128 --max-input-packets 128 --max-flushed-packets 128 " +
"-I ip " + msg.mcast_in_main_sw + " -l " + msg.local_address_input_main_sw + " " +
"-P pmt --service " + msg.input_service_id + " --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service " + msg.input_service_id + " --udp " + msg.scte35_udp_port_main_sw + " --delete-files " + 
"-P zap " + msg.input_service_id + " " +
"-P filter --negate --pid 0x1FFF " +
"-O ip " + msg.mcast_out_main_sw + " -l " + msg.local_address_output_main_sw + " --packet-burst 7 --enforce-burst";
```

[image]

I will try on a decent server once I can put my hand on it, but in the meantime, do you think it could be related to CPU/RAM issues?

lelegard, 2021-06-29:
Did you try to increase the socket buffer size in the input plugin, as indicated in my previous post?

The analysis is simple: if packets are missing, you fail to receive all UDP datagrams. This means that the input plugin fails to read new UDP packets fast enough. The input plugin executes in its own thread. It reads UDP datagrams, extract TS packets from it, put the packets in the global tsp buffer and loops back to read new UDP datagrams. There is nothing else. If it fails to do that, there is a system tuning issue. Maybe the CPU is busy and the input thread is not scheduled fast enough. It does not seem to be the case. With short flush values, all threads read and write more often in the global tsp buffer, leading to more contention on the mutex which synchronizes access to the buffer. The wait time will be very short but more frequent. If this is enough to miss UDP packets, this means that the kernel socket buffer is not large enough for the input bitrate. This is why I suggested to increase the socket buffer size.

lelegard, 2021-06-29:
> I will try the --buffer-size option, but is it located in the input ip plugin or in the global tsp options?

Please read the documentation ! There are so many plugins and options that one single volunteer (me) cannot answer that kind of question all the time for free. I made a huge effort to maintain the documentation. The users should provide the effort to read it.

> If in the global tsp options, can I use both --buffer size in combination with the three other options which are --add-input-stuffing --initial-input-packets --max-input-packets and --max-flushed-packets ?

All options are individually documented and, unless specifically documented otherwise, are independent 

> Last but not least, can the three global TSP options have a packets number that is different than the one from the output plugin option?

Global options are global. The options of each specific plugin are specific to that plugin. There is no relationship between the two.

moving-digital, 2021-06-29:
Sorry @lelegard 

My bad, I couldn't find the `--buffer-size` because I was looking for `--buffer-size-mb` somehow. And that is located in the global tsp options... Very sorry again. My apologies.

I've tried with `--buffer-size` = 128 Mb and it just fixed the issue. You are a genius!

`"-I ip " + msg.mcast_in_main_sw + " -l " + msg.local_address_input_main_sw + " --buffer-size 131072000 " 

Thanks again!!!
`

moving-digital, 2021-08-21:
Hello @lelegard 

I'm briefly reopening the topic. Would you be able to share the range of values accepted by --buffer-size? I've actually been trying at random because I couldn't find the proper range in the user manual...

Thanks !

lelegard, 2021-08-21:
Hi @moving-digital 

The unit is in bytes. It should probably be specified in the user's guide. The range of value is system-dependent.

In practice, this value is directly passed to the system as "socket receive buffer size". You have to refer to the operating system documentation (ie. google in practice) for precise details and the potential system-level tuning options.

moving-digital, 2021-09-01:
Hello @lelegard 

Once again thank you so much for your help. I'm afraid I tried everything I could and I still couldn't get a decent bandwidth using UDP. However in TCP, there is no issue at all.

Using IPERF3 gives the following bitrate for UDP with the Raspberry Pi. 

[image]

While as per TCP:

[image]

I'm really puzzled, have you ever seen this?

closed 2021-10-11 by moving-digital