tsduck/tsduck #802: [BUG] Compile error from -Werror=shadow gcc 7.5.0

ethouris, 2021-06-02:
#### Bug description:

gcc compiler warns about the shadowing a field by a function parameter, and the warning is turned into error, which results in a compile error.

#### How to reproduce:

Linux, gcc 7.5.0, compiled from qmake (although `make NOPCSC=1 NODTAPI=1` gives the same result)

#### Expected behavior:

Compiled with no errors.

#### Errors and logs:

~~~
../../../src/tsplugins/tsplugin_merge.cpp: In member function 'bool ts::MergePlugin::processDropPassOption(const UChar*, bool)':
../../../src/tsplugins/tsplugin_merge.cpp:368:78: error: declaration of 'option' shadows a member of 'ts::MergePlugin' [-Werror=shadow]
 bool ts::MergePlugin::processDropPassOption(const UChar* option, bool allowed)
                                                                              ^
In file included from ../../../src/libtsduck/plugin/tsPlugin.h:36:0,
                 from ../../../src/libtsduck/plugin/tsInputPlugin.h:36,
                 from ../../../src/libtsduck/plugin/tsPluginRepository.h:36,
                 from ../../../src/tsplugins/tsplugin_merge.cpp:39:
../../../src/libtsduck/base/tsArgs.h:409:15: note: shadowed declaration is here
         Args& option(const UChar* name,
               ^~~~~~

~~~

#### Environment:
 - OS: Linux
 - OS version: openSUSE Leap 15.1
 - TSDuck full version: 3.27-2411
 - Installation type: not yet installed

#### Additional information:

Under the following locations:
`tsplugin_merge.cpp:368`
`tsplugin_slice.cpp:187`
`tsplugin_time.cpp:210`

The `option` parameter was renamed to `opt` and this fixed the problem.

lelegard, 2021-06-02:
Thanks for reporting this. Will be fixed.

GCC 7.x is no longer part of the latest version of any distro. CentOS, usually the distro with the oldest software packages, uses GCC 8.1. Fedora uses GCC 11.1. I still have several older GCC versions that I rebuilt on a CentOS VM but I use them only to diagnose specific problems like this one. The "regular" builds use the latest version of each distro with the GCC that comes with it.

lelegard, 2021-06-02:
Fixed

closed 2021-06-02 by lelegard