tsduck/tsduck #803: [Q] SRT: multiple dynamic inputs for tsswitch with listener mode?

ethouris, 2021-06-03:
The description mentions SRT with listener mode, but there's nothing said explicitly as to whether it allows only one connection on this endpoint for a single input, or multiple connections are also handled. This is especially important for the `tsswitch` application.

lelegard, 2021-06-03:
In listener mode, only one incoming connection is accepted at a time, either in the input and output plugin.

Generally speaking, a TSDuck input plugin opens one transport stream source and reads packets until the "end of the stream", whatever it means for that plugin.

In the case of the SRT input plugin with listener mode, "opening a transport stream input" means opening the socket and waiting for one incoming connection. The end of the stream is declared when the remote peer disconnects.

ethouris, 2021-06-03:
Ok, I understand. Is it then even predicted in the TSSwitcher class such a functionality as adding or removing an input "on the fly"? I know it's not implemented (after reviewing the code), so I'm just asking if the architecture is at least prepared to handle such a case correctly.

lelegard, 2021-06-03:
> adding or removing an input "on the fly"?

The architecture does not prevent this. There would be quite some code to review but nothing impossible.

I am not satisfied anyway with the implementation of `tsswitch`, especially the class `ts::tsswitch::Core`. There is a long standing bug in the interaction between `--delayed-switch` and `--receive-timeout` (issue #324) which was never fixed and cannot be fixed with the current state machine. It should probably be completely redesigned but I never found the time and the motivation to do it.

closed 2026-04-26 by github-actions[bot]