tsduck/tsduck #804: [Q] Switching between SRT Input + weird SRT usage

ethouris, 2021-06-03:
I compiled TSDuck with SRT support and tried out tsswitch.

The intended usage is: I have 2 SRT listeners, each one is supposed to receive its own separate stream, but project to the output only one of them. The output is projected to another SRT listener, so I connect to it and receive the stream. The 4999 port is used for reading "commands". This is the command I used:

```
./tsswitch -f -r 4999 -I srt --listener 5001 --transtype live --messageapi -I srt --listener 5000 --transtype live --messageapi -O srt --listener 5005 --transtype live --messageapi
```

Questions:

1. The SRTO_MESSAGEAPI=true and SRTO_TRANSTYPE=live should be default parameters, moreover, tsduck is to be normally used for live transmission, why are the socket options overridden with non-default values for file transmission?
2. What I have observed after both streams were connected was that the first defined (port 5001) was being projected to the output. But when I tried to do `echo 0 >/dev/udp/<localhost-ip>/4999` - at least I found this somewhere in the documentation as the method to pass the command to switch to the next stream - nothing happens. What's the proper way to make the inputs switched on demand?

lelegard, 2021-06-03:
Hi @ethouris 

I will investigate the first point. I am not an SRT specialist and the `srt` plugin was initially a contribution from a user. So, let me check this.

The second point is easier. `echo 0 >/dev/udp/...` means "switch to input index 0". If you are already using the first input, you switch again and again on the same one. Use "next" or "previous" for next and previous input. Use an integer to switch to a specific plugin. The details are in chapter 3 of the user's guide, reference doc for `tsswitch`, "remote control" section.

lelegard, 2021-06-04:
> The SRTO_MESSAGEAPI=true and SRTO_TRANSTYPE=live should be default parameters, moreover, tsduck is to be normally used for live transmission, why are the socket options overridden with non-default values for file transmission?

@ethouris, from a "SRT for dummies" perspective (I am the dummy), what would you recommend for TSDuck SRT plugins? Input and ouput. Independently of what is currently defined in TSDuck.

I am interested in
- general approach and configuration
- default configuration and alternative options
- what should be avoided
- any useful remark

Thanks

ethouris, 2021-06-04:
Ok, after having ability to test it correctly, I found that the switching works (the documentation is kinda poor here, but I found what I needed in the sources), but there are some other issues:

1. Switching isn't smooth. This likely could be fixed by an additional processing block, but it is unclear for me, how tsswitch can get installed processing plugins that are attached to every input channel, not just processes the resulting stream. Something like that:
```
/-------\  /-------\
|       |  |       |
| SRT 1 |--| PROC  |\
|       |  |       | \       /-------\
\-------/  \-------/  \------|       |
                             |SWITCH |------- OUTPUT...
/-------\  /-------\   ------|       |
|       |  |       |  /      \-------/
| SRT 1 |--| PROC  | /
|       |  |       |
\-------/  \-------/ 
```

The PROC blocks should be also started or stopped on demand, while the inputs should be running always, then the switcher will only execute channel switching, when the processing block has responded with data (that is, until then it should read from the "previous" input).

2. SRT input isn't used exactly correct way. Once the SRT connection is established in live mode, the application must read the incoming data immediately and shall never stop unless the connection is broken, no matter if it has idea what to do with the incoming data or not - otherwise it will lead to explosion of the SRT receiver buffer and cause an irrecoverable problem. Currently I saw that when the output is also SRT, once you break the connection for the output, the application stops reading the input.
3. This can be connected to what is described in 2, as long as there's no connection on the output, the defined input SRT listener doesn't accept the connection. Likely this is because the input plugins aren't started - but that's kinda weird, as the `-f` option declares that - unlike default, where inputs aren't started when not selected - all input plugins should be started and should be read from, even if the payload were to fall on the floor.

ethouris, 2021-06-04:
As per "how to use SRT" I'm ready to help however I can. In general, if you use live mode, then options like `SRTO_TRANSTYPE`, `SRTO_CONGESTION` and `SRTO_MESSAGEAPI` should not be altered their default values.

Also, as I said above: once you get a connection in live mode for reading, the application must read the data as they come (even if they should be thrown away). We might also talk in person about the details, if you have some medium we can use for it.

lelegard, 2021-06-04:
Hi @ethouris,

Thanks for your feedback and your help.

As a preamble, please keep in mind that TSDuck was primarily designed for tests, debug, integration, demo, etc. This is not a fully integrated high-level application which is ready to be used out of the box in production by operators without intimate knowledge of the MPEG/DVB/ISDB/whatever technology. TSDuck is a low-level toolbox, literally hammers and screw-drivers, to be used by engineers who exactly know what they want to do and who are responsible for doing the right suite of low-level operations in the right order. Specifically, not enforcing the consistency of several elementary operations can be used on purpose to recreate incorrect operating conditions, typically to reproduce problems from the field.

> I found that the switching works (the documentation is kinda poor here, but I found what I needed in the sources)

What should be improved in the user's guide regarding this? Implementation details?

Some people are complaining that the user's guide is already too big (~500 pages). But they would also probably complain if there were no doc at all (like many open source projects unfortunately).

> Switching isn't smooth

How would you define "not smooth"? Even with `--fast-switch`?

> SRT input isn't used exactly correct way. Once the SRT connection is established in live mode, the application must read the incoming data immediately and shall never stop unless the connection is [...]

What do you mean by "irrecoverable problem"? If this is just losing packets, then that's OK: you are not fast enough, you lose packets, fair enough. But if you mean that this creates functional problems at the SRT socket level, then this is not acceptable (__anything__ can happen on a non-realtime system such as Linux or Windows which can lead to infrequent but unexpectedly long delays in process response time).

> Currently I saw that when the output is also SRT, once you break the connection for the output, the application stops reading the input.

This is the intended behaviour in `tsp` and `tsswitch`. When the output plugin reports an error to the upper level (`tsp` or `tsswitch`), the application terminates and all previous plugins are properly terminated. This is the test bed approach.

At the plugin level, however, the SRT output plugin has a `--multiple` option. This has been recently added on a user's request. In case of peer disconnection, the current send operation blocks until a reconnection is possible, at which time the pending packets are sent to the new peer. This is all internal to the plugin. From a `tsp` or `tsswitch` standpoint, this is just a lengthy call to `plugin->send(...)`.

But maybe you mean that, during this blocking period with the `--multiple` option, input packets are no longer read. This is true. In fact, input continues until the input buffer is full. In all cases, in `tsp` and `tsswitch`, there is a flow control in the buffer. When the buffer is full, input is stopped, until there is some free space again in the buffer. In the case of `tsp`, there is one single larger common buffer and packets are never moved. In the case of `tsswitch`, there is one buffer per input plugin (essentially because of `--fast-switch` and `--delayed-switch`) and packets are copied from the active input buffer into the output buffer.

> This can be connected to what is described in 2, as long as there's no connection on the output, the defined input SRT listener doesn't accept the connection. Likely this is because the input [...]

Again, this is the intended behaviour in `tsp` and `tsswitch`. This comes from the basic assumption that the very purpose of an application is to "produce" something, to output packets. So, the first real operation (beyond command line parsing) on any plugin is to start the output plugin. Because if it fails the application cannot produce anything, the application becomes useless and there no reason to continue in any way. In the case of the SRT output plugin, starting the plugin means connecting to the server (caller mode) or waiting for a client (listener mode). As long as the output connection is not completed, the `start()` operation of the output plugin does not return and nothing else is started (included the input plugins).

> As per "how to use SRT" I'm ready to help however I can. In general, if you use live mode, then options like SRTO_TRANSTYPE, SRTO_CONGESTION and SRTO_MESSAGEAPI should not be altered their default values.

Yes, please, let's discuss this. I really need to understand the "natural" configuration for a TS packet transmission with SRT and the other alternatives.

> Also, as I said above: once you get a connection in live mode for reading, the application must read the data as they come (even if they should be thrown away).

As mentioned earlier, if "must read" means "otherwise you won't receive packets correctly", this is fair game. If this means "otherwise the SRT socket will break or the library will crash", this is not acceptable. Nobody does this. Failing to read fast enough can result in packet loss (UDP, DVB broadcast) or full flow control (TCP) or even some intermediate form of flow control (try to recover up to some amount of packets, then lose packets) but not an application failure.

Generally speaking, because the initial fields of application were DVB broadcast and UDP, `tsp` and `tsswitch` are designed to read their input as fast as possible in nominal sustained conditions (asyn [... cut ...]
If this is really the _only_ way for SRT to work, that impacts the software architecture of the application. That would mean that the whole application should be built around this constraint. This means that SRT can be mostly used in "SRT-centric applications" but hardly in generic applications for which SRT is "just another form of input/output".

> We might also talk in person about the details, if you have some medium we can use for it.

Please send an email using https://tsduck.io/ => "Web resources" => "Contact" and we will find a way. You are right, fundamental conceptions must be directly discussed.