tsduck/tsduck #805: [FEAT] Craft with PES header manipulation

lars18th, 2021-06-07:
#### Is your feature request related to a problem?
With the current implementation of the TSDuck toolkit the modification of a PES header it's complicated. The reason is because we don't have specific parameters for options in the PES header in the `-P craft` plugin. For example, I want to change the PES_PRIORITY, the STREAM_ID and the PES_CEC_FLAG, and now I can't do it.

#### Proposed solution:
Add new options to the `craft` plugin adding the multiple options available in the PES Header. Examples:
- `--clear-pes-priority` and `--pes-priority`.
- `--clear-pes-crc-flag` and `--pes-crc-flag`.
- `--pes-stream-id value`.
- etc.

#### Alternatives:
The only alternative is to READ the current PES HEADER and patch it _manually_.

#### Additional information:
If the idea seems confusing as then the list of parameters for the `craft` plugin perhaps will be very large, then I suggest to implement a new `craft-pes` plugin.

I hope you want to consider this.
Thank you @lelegard in advance! :wink:

lars18th, 2021-06-07:
Hi @lelegard ,

I found that we need almost the option `--pes-header` in the `craft` plugin. When using something like `-P craft --only-label 1 --pes-payload --payload-pattern "1234"` the modified part of the TS packet is the PES PAYLOAD only. So we can't change the PES HEADER. This can be the expected behaviour, but then where is the `--pes-header` option to modify the PES HEADER?

Any idea to patch only the PES header at time?
Regards.

lars18th, 2021-06-07:
Hi @lelegard ,

Even you can consider that the `--pes-header` can be irrelevant, I see that you need to add another optional parameter for the `craft` plugin:
- `--no-repeat-pattern`: This indicate that the pattern is not repeated, so it only applies to the size of the binary string.

Why that? Imagine that you whant to change something in the PES header. Then you can work without the `--pes-payload`, so it applies to all the payload (including then the PES header). But if you want to change ONLY one BYTE, you want to do this:
- `-P craft --no-repeat-pattern --offset-pattern 3 --payload-pattern "0xE0"` to only changing the BYTE at POS @3 of the PES HEADER.

But in fact of this, at time you can only do that ALL the payload from the byte @3 has the byte `0xE0` repeated. Then I'm missing the option to limit the pattern repetion.

Futhermore, perhaps you want to add some utility to apply the pattern at bit level. Perhaps with a `mask-pattern` parameter. Then you can use this value to apply an AND operator between the content of the packet and the pattern before replacing the content.

Please, can you add this?
Thank you!

lars18th, 2021-06-07:
Hi @lelegard ,

After recompiling the current source removing this loop https://github.com/tsduck/tsduck/blob/31dc486b414294fc310a09756a61f6ce3f58f382/src/tsplugins/tsplugin_craft.cpp#L765 I can finally patch the PES header section as I need. So that's my current suggestion for changes in the `craft` plugin:

- `-P craft --no-repeat-pattern`: Flag to disable the repetion of the pattern. Useful to patch only a part of the packet.
- `-P craft --mask-and-pattern`: With this flag instead of _replacing_ the content with the pattern, you write the result of a binary AND between the current content and the pattern.
- `-P craft --mask-or-pattern`: Similar to the previous but with a binary OR between the current content and the pattern.

With only these three new parameters, and including in the Documentation that without using `--pes-payload` you can patch the PES Header, then I feel we'll have sufficient options to modify/path whatever we need. Simple and clear.

You agree?
Regards.

lelegard, 2021-06-07:
The plugin `craft` was initially an input plugin only (generating specially "crafted" packets). In the case of the input plugin, there was nothing pre-existing. So, not repeating or AND/OR operators did not make sense.

Later, the packet processing capabilities were added and the same options as the input plugin were copied. But, in the case of the packet processing version, the three new options make sense. Will do...

This input plugin will remain unchanged.

lelegard, 2021-06-19:
Done

closed 2022-09-23 by lelegard