tsduck/tsduck #812: [Q] EIT (possibly also other) minimum section gap of 25 ms not being respected by "inject"

kriip, 2021-06-10:
Hi,

Im trying to use tsp to create a TS with SI tables (NIT a, SDT a/o, EIT pf/sch a/o, TOT/TDT). The PID's from the TS is intented to be muxed by AppearTV equipment and eventually broadcast as DVB-C.

In general i have this working, but using Bridgetech VB220 for ETR290 analysis, I see failures for "Section gap check". Specifically this means section 5.1.4 from EN 300 468 "Repetition rates and random access" is not being respected:

"For SI specified within the present document the minimum time interval between the arrival of the last byte of a section to the first byte of the next transmitted section with the same PID, table_id and table_id_extension and with the same or different section_number shall be 25 ms"

I have tried alot of tweaking, to see if I can find a stable solution, but so far have been unsuccessful. I have slimmed down my PoC to make it easier to replicate the issue, and I now have this test case:

EIT P/F actual source in .xml format: test-ts306-eit-pf-a.xml
Precompiled to binary format using tstabcomp: tstabcomp --compile --europe --eit-normalization --eit-base-date 2021/06/10 --output test-ts306-eit-pf-a.bin test-ts306-eit-pf-a.xml

Binary EIT file injected in stream to be played each 1000ms using following tsp command:

~~~
tsp --verbose --bitrate 600000 --max-flushed-packets 112 --monitor \
 --control-local 127.0.0.1 --control-port 1234 \
 -I null \
 -P regulate --packet-burst 112 \
 -P inject --pid 18 test-ts306-eit-pf-a.bin=1000 --poll-files --stuffing \
 -O ip 232.20.60.1:5900 --ttl 16 --packet-burst 7 --enforce-burst
~~~

Verified with ETR290 analyzer Section Gap issues were being reported, see etr290 analysis.png
Recorded 20 seconds from my stream using tsp for offline analysis, see section-gap-too-low.ts

Question:
Im not sure if Im hitting limitation of tsduck or a bug. Maybe there is a better way to achieve what Im trying to do? Hope you can help enlighten me :)

For reference, the full command for generating all my tables would be something like:

~~~
tsp --verbose --bitrate 600000 --max-flushed-packets 112 --monitor \
 --control-local 127.0.0.1 --control-port 1234 \
 -I null \
 -P regulate --packet-burst 112 \
 -P inject --pid 16 test-nit-a.bin=2000 --poll-files --stuffing \
 -P inject --pid 20 test-tdt.bin=15000 --poll-files --stuffing  \
 -P timeref --start system \
 -P inject --pid 17 test-ts306-sdt-a.bin=500 test-ts306-sdt-o.bin=5000 --poll-files --stuffing \
 -P inject --pid 18 test-ts306-eit-pf-a.bin=1000 test-ts306-eit-pf-o.bin=5000 test-ts306-eit-sch.bin --bitrate 500000 --poll-files --stuffing \
 -O ip 232.20.60.1:5900 --ttl 16 --packet-burst 7 --enforce-burst
~~~

#### Errors and logs:
Files: [case.zip](https://github.com/tsduck/tsduck/files/6629773/case.zip)
~~~
root@iptvanalyzer2:/data/dumps/tg# tsp --verbose --bitrate 600000 --max-flushed-packets 112 --monitor \
>  --control-local 127.0.0.1 --control-port 1234 \
>  -I null \
>  -P regulate --packet-burst 112 \
>  -P inject --pid 18 test-ts306-eit-pf-a.bin=1000 --poll-files --stuffing \
>  -O ip 232.20.60.1:5900 --ttl 16 --packet-burst 7 --enforce-burst
* [MON] 2021/06/10 09:47, resource monitoring started
* inject: loaded 8 sections from test-ts306-eit-pf-a.bin, repetition rate: 1,000 ms
* inject: target bitrate from repetition rates: 12,032 b/s
* regulate: minimum packet burst duration is 10,000,000 nano-seconds
* null: initial input bitrate is 600,000 b/s
* regulate: regulated at bitrate 600,000 b/s
* inject: transport bitrate: 600,000 b/s, packet interval: 49
* inject: file test-ts306-eit-pf-a.bin created
* inject: loaded 8 sections from test-ts306-eit-pf-a.bin, repetition rate: 1,000 ms
* inject: target bitrate from repetition rates: 12,032 b/s
* inject: transport bitrate: 600,000 b/s, packet interval: 49
* [MON] 2021/06/10 09:47, VM: 526 MB (+346 MB), CPU:0.20% (average:0.20%)
* [MON] 2021/06/10 09:47, VM: 526 MB (stabilizing), CPU:0.10% (average:0.15%)
* [MON] 2021/06/10 09:47, VM: 526 MB (stabilizing), CPU:0.20% (average:0.16%)
* [MON] 2021/06/10 09:47, VM: 526 MB (stabilizing), CPU:0.20% (average:0.17%)
* [MON] 2021/06/10 09:47, VM: 526 MB (stabilizing), CPU:0.20% (average:0.17%)
* [MON] 2021/06/10 09:48, VM: 526 MB (stabilizing), CPU:0.20% (average:0.18%)
* [MON] 2021/06/10 09:48, VM: 526 MB (stabilizing), CPU:0.10% (average:0.17%)
* [MON] 2021/06/10 09:48, VM: 526 MB (stabilizing), CPU:0.20% (average:0.17%)
* [MON] 2021/06/10 09:48, VM: 526 MB (stabilizing), CPU:0.10% (average:0.16%)
* [MON] 2021/06/10 09:48, VM: 526 MB (stabilizing), CPU:0.10% (average:0.15%)
* [MON] 2021/06/10 09:48, VM: 526 MB (stabilizing), CPU:0.20% (average:0.16%)
* [MON] 2021/06/10 09:49, VM: 526 MB (stabilizing), CPU:0.20% (average:0.16%)
* [MON] 2021/06/10 09:50, VM: 526 MB (stabilizing), CPU:0.15% (average:0.16%)
^C* tsp: user interrupt, terminating...
* [MON] 2021/06/10 09:50, resource monitoring terminated
root@iptvanalyzer2:/data/dumps/tg#
~~~

#### Environment:
 - OS: Linux
 - OS version: Debian 10
 - TSDuck full version: 3.26-2349
 - Installation type: Official debian binary: tsduck_3.26-2349.debian10_amd64.deb

Thanks in advance,
Kristian

lelegard, 2021-06-10:
Hi @kriip 

Quoting https://tsduck.io, TSDuck is made "for test, monitoring, integration, debug, lab, demo." TSDuck has been designed to troubleshoot operational issues, not to be a production-grade software. 

Maybe some fine tuning may improve what you are seeing but TSDuck has never been designed to comply with ETSI TR 101 290. TSDuck is developed for free on spare time by one single person (me). Complying with strict constraints as listed in TR 101 290 requires a lot more development, a large test infrastructure and availability to process many support requests like yours. This is out of reach in the TSDuck economical environment. The level of precision and support you request is typically a matter of professional (ie. non-free) software or at least seriously sponsored open source projects. TSDuck is neither, sorry.

kriip, 2021-06-11:
Hi @lelegard

Thank you for your fast reply :) Ofcause not the reply I had hoped for, but i totally understand and respect your answer.

Thank you also, for creating an awesome piece of software! The toolkit approach and scriptablity TSDuck offers really catered to us much more than just the price point compared to the commercial offerings we also are considering. Being able to interface programatically and scale horizontally by combining TSDuck with tools like Docker would be so much better than a point-n-click webgui for setting up many DVB networks. I hope you will some day consider offering some kind of commercial support :)

Regards,
Kristian

closed 2021-06-11 by kriip