tsduck/tsduck #813: [Q] HLS input: Up to v8 support and SCTE-35 extraction ?

moving-digital, 2021-06-11:
Hi @lelegard 

I got two questions related to the HLS input plugin:

- Does it support up to v8 implementation?
- Is it able to extract the SCTE-35 markers present in the HLS, and reinject them in a TS?
#EXT-X-CUE-OUT:DURATION=7,ID=xxxxxxxxxx
#EXT-X-CUE-IN:DURATION=7,ID=yyyyyyyyy

lelegard, 2021-06-11:
Hi @moving-digital 

The HLS input plugin, like any other input plugin, is "just another source of transport stream". It only grabs the HLS segments as a continuous source of TS packets.

It does not implement all features of HLS. For a start, it only supports TS format, not fMP4. The only relevant tags in the playlists are the identification of the segments (for a media playlist) and the bandwidth and resolution information (for a master playlist, in order to select the appropriate version).

I don't see where you get the `#EXT-X-CUE-OUT` or `-IN` tags from. I cannot find them in [HLS version 9](https://datatracker.ietf.org/doc/html/draft-pantos-hls-rfc8216bis-09). I cannot see them either in [SCTE 35 2020](https://scte-cms-resource-storage.s3.amazonaws.com/ANSI_SCTE-35-2020-1619708851007.pdf) where section 12.2.3. "HLS cue tags" says "The #EXT-X-SCTE35 is the only tag defined by this standard".

Anyway, the purpose of an input plugin is to grab input TS packets, not to modify the transport stream.

moving-digital, 2021-06-11:
Thanks @lelegard 

`#EXT-X-CUE-OUT or -IN` are tags used in HLS for the POIS (Placement of opportunity), aka ad insertion. Some links here:
- https://docs.aws.amazon.com/medialive/latest/ug/sample-manifests-hls.html
- https://docs.aws.amazon.com/mediatailor/latest/ug/hls-ad-markers.html
- https://helpx.adobe.com/primetime/kb/what-is--ext-x-cue-out-cont-used-for-in-the-ssai-manifest-.html

Would TSDuck be able to convert such tags back to a proper SCTE-35 PID available in a "basic" TS stream?

lelegard, 2021-06-11:
Amazing, the syntax of the `#EXT-X-CUE-OUT` is not even compatible between AWS Elemental MediaLive and AWS Elemental MediaTaylor. And Adobe PrimeTime has a third syntax. The three syntax manipulate the same concepts but in slightly different ways. And all of them are obviously completely different from what the SCTE 35 standard says. What a mess...

But,, again, an input plugin is a source of TS packets, nothing more. Inserting SCTE-35 table means creating new packets (or replacing null packets but HLS streams usually don't have any). Artificial null packet insertion (`--add-input-stuffing` option) occurs at `tsp` level, outside the input plugin. The interactions and synchronization with the rest of the stream are also important. This is outside the scope of an input plugin.

bbgdzxng1, 2021-06-27:
There are many different syntax formats for manifest decoration.  It is a mess.

There are only two "standard" ones, `#EXT-X-DATERANGE` (as per RFC8216 bis) and `#EXT-X-SCTE35` (ANSI/SCTE 35 2019a).  However, support for these by segmenting encoder vendors and ad insertion vendors has historically been very low, although support for `#EXT-X-DATERANGE` is now growing.

Then there are the _vendor_ implementations.  Each detecting vendor does their own thing, but both Google Ad Manager and MediaTailor do a reasonable of documenting the formats they can detect:
https://docs.aws.amazon.com/mediatailor/latest/ug/hls-ad-markers.html
https://docs.aws.amazon.com/medialive/latest/ug/sample-manifests-hls.html
https://support.google.com/admanager/answer/7245661?hl=en#supported-ad-markers

- `#EXT-X-CUE` (Adobe Primetime)
- `#EXT-X-CUE-OUT`, & `#EXT-X-CUE-OUT-CONT` with two sub variations
  - `#EXT-X-CUE-OUT:DURATION=10.50`
  - `#EXT-X-CUE-OUT:10.50`
- `#EXT-OATCLS-SCTE35` (aka Enhanced, same as Elemental but with additional binary payload)
- `#EXT-X-SPLICEPOINT-SCTE35`
- And there are several other proprietary segmenting encoder vendor implementations as well.

Encoder support for #EXT-X-DATERANGE (as per RFC8216) is on the rise, because it is RFC.  And it is the real SCTE message.
#EXT-X-CUE-OUT/IN is the simplest markup format, since an encoder/segmenter can implement this without even reading the SCTE-35 documentation.  This is the most common "simple markup".

The danger point for the implementer is that a manifest can declare the duration in multiple tag events.  The duration is often included in the OUT message, the duration is included as a rolling update in the CONT message, the duration is included in the binary payload.  The binary payload may then be repeated/updated between tags.  A client such as TS Duck needs to choose one as the point of authority, decide some level of precedence if the same contradictory info is included in the _same tag_ (yes it happens), and decide whether "last known" wins.  And log them all so that it is debuggable.

To complicate it further, any declared duration should be trumped by a CUE-IN, because that is conceptually a splice immediate return to network.  But the duration could also expire before the CUE-IN, especially when using non-integer framerates like 30000/1001.  So in that case, the declared duration should win over the CUE-IN.

Worse than that, a segmenting encoder may choose to send more than one manifest decoration tag format.  Because more is better, right?

So which one could a tool choose, support and even implement?
- Allow user to control whether the SCTE-35 data track or HLS manifest is considered the reference
- If HLS manifest, which markup formats/tags should be supported?
- When contradictory information is received, what takes precedence?  And log it.
- How do you debug contradictory info?
- Log unsupported tags events, so it is clear that unsupported tags are ignored.

Supporting all the various vendor formats would be a Sisyphean task.  @lelegard is correct, it is a mess and should be handled at HLS level.

There is absolutely a need for a de-facto tool for debugging HLS ad markers, but whether that is the responsibility of a transport stream analyzer like TS Duck or should really the responsibility of a HLS manifest parser / analyzer project is a difficult question - and there is no de-facto, open source HLS parser / analyzer /toolkit project or library that has the clear momentum - the industry needs one of the many HLS parsers/validators to become a "HLS Duck", like a cross-platform implementation of Apple Media Stream Validator which supports outbound pipes and translation from manifest to SCTE datatrack, with logging and HLS validation - it is a whole project in itself.
- Robert Pantos' Media Stream Validator for macOS does not support ad markers, because it is not a validator of the HLS drafts or RFC, it is a validator of playback compatibility with Apple devices.
- FFmpeg is lacking in both inbound HLS ad marker and SCTE data track support.
- https://github.com/futzu/threefive is a very cool project and has some very limited support for HLS markers, against a local playlist.  (Off topic - Ignoring HLS, from purely a data track perspective, threefive and TS Duck's rmsplice would be an amazing hookup, as threefive has some awesome debug/parsing - check it out @lelegard).

Whether the requirement for manifest ad marker parsing should be satisfied in a specialized transport stream analyzer or a separate HLS manifest tool is a very valid question.

closed 2021-10-13 by moving-digital

superkabuki, 2025-04-01:
@moving-digital https://github.com/superkabuki/threefive3  
supports live HLS and all the HLS tag variations #EXT-X-DATERANGE, #EXT-X-CUE-*, #EXT-OUTCLS-SCTE35 and #EXT-X-SCTE35