tsduck/tsduck #814: > Considering the extraction of a table such as the NIT in your example, you should use a `ts::SectionDemux`. This is a class which does all section demuxing, table reconstruction and notifies the application when a new complete table is available. By accumulating raw TS packets as you are doing, you will have to manually recreate the complicated stuff in `ts::SectionDemux`. TSDuck classes are all designed to hide the complexity and provide a high-level and simple-to-use interface to applications or plugins.

fabricioizumi, 2021-06-15:
> Considering the extraction of a table such as the NIT in your example, you should use a `ts::SectionDemux`. This is a class which does all section demuxing, table reconstruction and notifies the application when a new complete table is available. By accumulating raw TS packets as you are doing, you will have to manually recreate the complicated stuff in `ts::SectionDemux`. TSDuck classes are all designed to hide the complexity and provide a high-level and simple-to-use interface to applications or plugins.

Where Can I find a sample code for using SectionDemux like you describe? I trying to use TSDuck as a library in my project to extract some tables from a TS file.

_Originally posted by @fabricioizumi in https://github.com/tsduck/tsduck/issues/317#issuecomment-861611623_

lelegard, 2021-06-16:
Hi @fabricioizumi 

In short, you need to develop a C++ class which implements `TableHandlerInterface`, use an instance of `SectionDemux` with an instance of your class as table handler, sets the PID's to demux, pass all TS packets to the demux. Whenever a new version of a table is detected on one of the demuxed PID's, your class implementing `TableHandlerInterface` will be called with that table.

TSDuck is the largest pool of sample code. `SectionDemux` is used everywhere. I would suggest to start with:
- `src/tsplugins/tsplugin_nitscan.cpp`
- `src/libtsduck/dtv/broadcast/tsTSScanner.h` & `.cpp`

fabricioizumi, 2021-06-25:
Ok, thanks for your answer. I study the code in tsTSScanner.cpp and I have success in implement a class based in TableHandleInterface. Thanks!

closed 2021-06-25 by fabricioizumi