tsduck/tsduck #815: MPEInjecct with UDP size greater than 4052 bytes

jasmitch, 2021-06-16:
Am looking to encapsulate a multicast data stream using mpeinject with a length of 41421bytes, I receive the following error from tsp:

Error: mpeinject: error creating MPE section from UDP datagram, source: 192.168.66.124:2122, destination: 224.1.2.111:2122, size: 4141 bytes

Noticed that I am able to send with a length up to 4052bytes, I can see that the Max_IP_SIZE is set to 65535 in tsplugin_mpeinject.cpp, however cannot see where the 4052bytes limit is? Is there any reason for this limit? Am happy to recompile if a larger value is possible?

lelegard, 2021-06-16:
Hi @jasmitch 

4052 is not a hard-coded limit, this is an implicit limit to encapsulate a UDP datagram into a DSM-CC section. The maximum size of a section is 4096 bytes (length field on 12 bits), minus 8 (section header) + 4 (MAC) + 20 (min IP header) + 8 (UDP header) + 4 (section CRC) equals 4052 for the UDP payload.

It is unclear to me if ETSI EN 301 192 allows an IP datagram to span over several sections. If it is allowed (to be confirmed), then this is a limitation of the `mpeinject` plugin.

This issue has never been raised for one good reason: although the maximum size of an IP datagram is 65536 bytes, it is a common practice to never generate datagrams larger than 1500 bytes (standard MTU). Over this limit, the datagram will be fragmented and the risk of data loss is increased by a factor which is proportional to the size.

This is why, for instance, the standard TS encapsulation into UDP (multicast or unicast) uses 7 TS packets per UDP datagram. This is the largest number of TS packets below 1500 bytes.

So, I would say:
- 4052 bytes is the maximum UDP payload size to fit into a DSM-CC section for MPE.
- It is unclear to me if more than one section per UDP datagram is allowed.
- Even if it is allowed, I wouldn't bet that all receivers will handle that correctly because it is a common good practice to never generate UDP datagrams which are larger than 1500 bytes.
- This issue is the symptom that your UDP source, although valid, does not respect the networking good practices.

If you can control the UDP source, I would suggest to review it to avoid general networking performance issues in addition to MPE limitations.

jasmitch, 2021-06-17:
Thanks @lelegard,

I don't have any control in the software for the UDP payload settings, so will follow up. The reason for the larger payload is for improved efficiency/through-put over a satellite channel as this application is for a file broadcast requirement. I guess ULE might be something to consider too.

Appreciate your detailed response.

lelegard, 2021-06-17:
>The reason for the larger payload is for improved efficiency/through-put over a satellite channel 

This assumption is generally considered incorrect. The peak of efficiency of UDP is at 1500 bytes, there is some consensus about this. Larger payloads increase throughput but significantly reduce the reliability, thus decreasing the overall efficiency.

> this application is for a file broadcast requirement.

This is even more a good reason for not increasing the payload size over 1500 bytes. Transfering files requires reliability. Unlike video, losing file data more or less means losing the whole file. Transport stream, especially over satellite, is error prone. Packet losses are frequent. So, first, you need to add some form of redundancy (see the FLUTE protocol, for instance, for redundancy in file broadcasting). Then, you need to limit the size of lost chunks to limit the size and usage of the redundancy overhead. So you _must not_ increase the payload size.

> I guess ULE might be something to consider too.

I remember that ULE vs MPE was a debate long ago. But ULE never made its way. I cannot tell why. I just retrieved the [latest ULE draft](https://datatracker.ietf.org/doc/html/draft-fair-ipdvb-ule-02). It dates back from 2003 and never became a standard AFAIK while MPE is a widely used standard.

jasmitch, 2021-06-18:
Interesting, I wasn't able to find too much documentation/studies on payload v's efficiency for mpe, all 
however showed efficiency converging after ~1200Bytes, which makes sense. The application does employ a 
form of FLUTE protocol, it has a configurable application layer FEC and a back channel request capability 
for missed packets.

ULE is probably where some efficiency savings could be made, thanks for your help.

gslin, 2021-07-08:
If ignoring the efficiency, is there any way to support larger UDP size?

lelegard, 2021-07-08:
> If ignoring the efficiency, is there any way to support larger UDP size?

Currently not. It would require modification of the code to allow the generation and injection of several sections for one UDP datagram. Given that this is useless in practice (inadequate performance and risk that multi-section UDP datagrams are not correctly extracted by some receivers), I see no incentive to implement that.