tsduck/tsduck #816: [FEAT] pcrshift plugin

lars18th, 2021-06-22:
#### Is your feature request related to a problem?
My problem is to found a _simple_ solution to patch/edit/modify PCR and/or PTS/DTS values _independtly_. I want for example to add/substract one offset to the clock marks (all or only some). I request something similar in more than one case (i.e. #809, #347, etc.).

#### Proposed solution:
Similar (in the esence) to the plugin `timeshift` I suggest to create a new one called `pcrshift`. The objective of this new plugin is edit the PCR/PTS/DTS time marks. Instead of _shifting_ the packets it _moves in time_ the packet without physically moving it. Why you could want this? One example: you have a simple (in the sense of small) mis-synchronization of one substream (for example, one audio pid). And then you want to _restore_ this error. You then can calculate the TIME offset and the POSITION offset (please, take note that audio is CBR). So using the `timeshift` you can solve the _position_ error. However, to solve the _time_ error you lack the counterpart plugin.

#### Alternatives:
We don't have at time any method to free patch/edit/change PCR/PTS/DTS values. The current tools are useful only to solve typical errors related to the container format of the MPEG-TS. So none of the current tools enables the option to edit time-marks in TS packets.

#### Additional information:
From my point of view, this kind of functionality is a major missing feature in the toolbox. I hope you understand why we want it.
Regards.

lars18th, 2021-06-28:
Hi @lelegard ,

What's your point of view regarding the lack of tools to patch/edit/change PCR/PTS/DTS values?

lelegard, 2021-06-28:
Please try the new `pcredit` plugin. This is the basis for PCR/PTS/DTS editing. New  features could be added when needed.

lars18th, 2021-07-02:
Hi @lelegard ,

> New features could be added when needed.

I'm not sure if this feature could be implemented in the `pcredit` plugin. But in any case I'll detail it here:

- The objective is to calculate the _time difference_ (the offset) of two different streams (inside the same MPTS). The strategy is to analyze (with external tools) two streams (for example two videos or audios with some marks), and obtain the exact TS packet where is the starting PES packet that corresponds to frames of the same instant time. The idea with these two marks is that the plugin will print the _time difference_ of the streams to use this value with the `pcredit` plugin.

So, perhaps the plugin could do something like this: Given a pid and a time, search for the PUSI mark more close to this time. Do the same for a second pid and time. And when your have both packets identified calculate the PTS difference and print it.

My idea is to do this:
```
--------/ Muticast MPTS /-+---> TSDuck pcredit-offset-calculator ----> TSDcuk pcredit (resync)
                          |                          ^
                          ----> External analyzer ---|
```

After the resync, peharps an additional remuxing could be necessary if the position of the TS packets (inside the MPTS) with new timestamps excess the decoding buffer size. But this is out-the-scope of the functionality.

What you think about this functionality? Very complex?
Regards.

dvbshka, 2021-09-13:
hi @lelegard 
I am trying to use the `pcredit` plugin but it tells me that no such plugin exists.
`* Error: pcredit.so: cannot open shared object file: No such file or directory
* Error: processor plugin pcredit not found`
tsversion
3.27-2438

lelegard, 2021-09-13:
> I am trying to use the pcredit plugin but it tells me that no such plugin exists.

As you can see from this discussion, the `pcredit` plugin has been recently added. It will be present in version 3.28 (release to be expected soon). In the meantime, you may either [rebuild TSDuck yourself from the latest state of the repo](https://tsduck.io/doxy/building.html) or grab the [latest nightly builds](https://tsduck.io/download/prerelease/) (Ubuntu and Windows only).