tsduck/tsduck #817: [Q] Understanding JSON report from plugin Analyze

Anas92230, 2021-06-24:
Hello,
I need some help to understanding information on json report. I used analyze plugin from ip multicast input. for this multicast i used "services->Is-Scrambled to determine if the service is scrambled or not. But when the field "is-scrambled" is false the service can be scrambled! for example the plugin : 
-find ECM in stream
-find audio component scrambled!
thank you for your help. I put a json export.
[export_plugin_analyze.txt](https://github.com/tsduck/tsduck/files/6708016/export_plugin_analyze.txt)

lelegard, 2021-06-24:
Hi @Anas92230 

The text file you provide does not exhibit the reported behaviour. The JSON contains only one service:
~~~
  "services": [
   {
    "bitrate-204": 0,
    "type-name": "MPEG-1 Video",
    ...
    "pids": [
     100,1000,1001,5456,6456],
    "name": "TV",
    "is-scrambled": false,
    ...
    }
   }],
~~~

I looked in the JSON description of the 5 PIDs of the service (100,1000,1001,5456,6456). They all have `"is-scrambled": false`. It is consequently normal that the service is marked as not scrambled as well.

Could you provide a binary TS file which has this specific issue?

Anas92230, 2021-06-24:
Hello,
Thanks for your comment.
If you look audio pid you can see is scrambled True and also you can see ecm pid.
That's why I was confused. To be complete the ts is scrambled.
I'll do a ts record for you.

Anas92230, 2021-06-24:
[tv.zip](https://github.com/tsduck/tsduck/files/6710206/tv.zip)
You can find in attach file an export from ts recording. i just replace ts extension by zip extension :-)
telle me if it's ok for you

lelegard, 2021-06-24:
Thanks for the TS. The service is correctly marked as scrambled in the JSON:

~~~
$ tsanalyze tv.ts --json >tv.json 
$ jq <tv.json -r '.services[0]."is-scrambled"'
true
~~~

Or the complete services entry:
~~~
  "services": [
    {
      "bitrate": 1819196,
      "bitrate-204": 1974021,
      "components": {
        "clear": 3,
        "scrambled": 2,
        "total": 5
      },
      "id": 456,
      "is-scrambled": true,
      "name": "HANNIBAL TV",
      "original-network-id": 1,
      "packets": 12105,
      "pcr-pid": 1000,
      "pids": [
        100,
        1000,
        1001,
        5456,
        6456
      ],
      "pmt-pid": 100,
      "provider": "Harmonic",
      "ssu": false,
      "t2mi": false,
      "tsid": 456,
      "type": 1,
      "type-name": "MPEG-1 Video"
    }
  ],
~~~

Anas92230, 2021-06-24:
I'm so confused i don't understand...the previous json give me an another info.
especially since there was contradictory information ecm pid with no scrambling.
Thank you so much.

closed 2021-06-24 by Anas92230

lelegard, 2021-06-25:
> there was contradictory information ecm pid with no scrambling.

The presence of an ECM component and the scrambling of any component are two independent topics.

Anas92230, 2021-06-25:
Yes it's true
One possibility is when I scan this multicast the service had an issue in configuration.
Finally you confirm if the field is-scrambled in the service was true or false is always right.
If I can I want to know did you give formation on tsduck?

lelegard, 2021-06-25:
> One possibility is when I scan this multicast the service had an issue in configuration.

This is not necessarily a misconfiguration. This is a standard configuration in the case of a clear/scrambled transition (see the transmission diagrams in annex G of ETSI TS 103 197) or during short clear periods (ads for instance).

> Finally you confirm if the field is-scrambled in the service was true or false is always right.

The field is-scrambled is set in a service when at least one scrambled TS packet was seen in one of the PMT-declared component PID's of the service during the time of analysis.

> If I can I want to know did you give formation on tsduck?

What do you mean by that?

Anas92230, 2021-06-25:
You can send me a private message at Aknazzaya@gmail.com
I'll explain you with more detail