tsduck/tsduck #818: [Q] How to correctly configure DekTec output for DVB-T2

st-channel24, 2021-07-02:
Starting from a working configuration in DekTec's StreamXpress for the play-out of a transport stream in DVB-T2, I failed to copy the set-up to TSDuck .
Problems:
- at the time of writing the commands,  I seem to be missing TSDuck equivalents for [DekTec's](url) "Frame structure : NUM_DATA_SYMBOLS" and "PLP0 : NUM_BLOCKS"
- output result on DTU-315 : no error messages in the cmd console, but the stream is played out at a BW of 10 MHz iso the configured 8 MHz (and consequently cannot be received on a TV)
- output result on DTU-215 : cmd console gives an error from the DekTec driver : "*Error: dektec: output device start send error: DTAPI_E_INVALID_SAMPRATE (DTAPI status 4306)"
[TSDUCK_DVBT2.txt](https://github.com/tsduck/tsduck/files/6755363/TSDUCK_DVBT2.txt)

command input 👍 

`tsp --add-input-stuffing 1/3 
-I file C:\data\scratch\streams\DVB-T2_Bonn_514.000MHz_CH26_PLP0.ts -i 
-P limit --bitrate 21983722 
-O dektec  --device 0 --frequency 514000000 --level -350 --modulation DVB-T2 --bandwidth 8 --bandwidth-extension --cell-id 0  -–papr none --pilot-pattern 2 --plp0-code-rate 3/5 --plp0-fec-type 64k --plp0-group-id 1 --plp0-high-efficiency --plp0-id 0 --plp0-il-length 3 --plp0-il-type 0 --plp0-issy none --plp0-modulation 64-QAM --plp0-rotation --plp0-type 1 --t2-fpsf 2 --t2-guard-interval 19/128 --t2-l1-modulation 64-QAM --t2-network-id 12421 --t2-system-id 32769
`

What could I be doing wrong ? Are there any examples for DVB-T2 play-out on DekTec ?

lelegard, 2021-07-02:
Reopening issue since it seems legitimate.

> at the time of writing the commands, I seem to be missing TSDuck equivalents for DekTec's "Frame structure : NUM_DATA_SYMBOLS" and "PLP0 : NUM_BLOCKS"

These two fields are automatically computed from other parameters using the Dektec function `DtDvbT2Pars::OptimisePlpNumBlocks`.

The results are documented as follow in Dektec's API documentation:
- OptPlpNumBlocks: optimum value for the number of FEC blocks per IL frame for PLP0 to maximise the DVB-T2 channel’s bitrate.
- OptNumDataSyms : optimum value for the number of data OFDM symbols per T2 frame to maximise the DVB-T2 channel’s bitrate.

> output result on DTU-315 : no error messages in the cmd console, but the stream is played out at a BW of 10 MHz iso the configured 8 MHz

Do you mean that you specify `--bandwidth 8` but, when you analyse the RF signal, you get 10 MHz bandwidth ?

Could you try with `tsp --debug ...` It should display all DVB-T2 parameters which are passed to the Dektec API. The bandwidth is marked as `m_Bandwidth =`. For 8 MHz bandwidth, the integer value should be 4. If this is the case, either your measurement is incorrect or there is a bug in the Dektec API.

> output result on DTU-215 : cmd console gives an error from the DekTec driver : "*Error: dektec: output device start send error: DTAPI_E_INVALID_SAMPRATE (DTAPI status 4306)"

This is a Dektec error. I have no idea what it means. It is not documented as a possible error code for `DtOutpChannel::SetTxControl`.  I have no DTU-215 to try. Please use `tsp --debug` and post the complete output.

Also post the output of `tsp --version=all` to make sure we can specify the right DTAPI version if the problem should be reported to Dektec.

st-channel24, 2021-07-05:
Indeed, your understanding concerning BW is correct : using StreamXpress I get an 8 MHz channel that can be received on the TV.
Using the TSDuck commands as described, yields either an error (DTU-215) or a 10 MHz channel that cannot be received (DTU-315) (seen on a spectrum analyser with identical settings for the 8MHz and 10 MHz observation)
"tsp -- debug " shows that m_Bandwidth = 4 (as expected).
Somebody suggested me to add the bitrate in the output processor, e.g. -o dektec --bitrate=21983722.
And YES, this works !!!  It is however not very practical, because the value is extemely critical : deviations of merely
[feedback_TSDuck_debug.txt](https://github.com/tsduck/tsduck/files/6763213/feedback_TSDuck_debug.txt)
 more than +/- 100 ppm  already (!) cause reception to fail on the TV, and deviations of a couple of % generate again the same dektec error. 
NOTE : the value was obtained from StreamXpress.

lelegard, 2021-07-05:
From your debug trace file, I see that you are using TSDuck version 3.17, which is quite old. The latest version is 3.27.

There is maybe an inconsistency between the Dektec library (linked with TSDuck, quite old: 5.33.0.117 while the latest one is 5.44.1.162) and the Dektec device drivers (separately installed, maybe newer: 4.13.4.81 but still not up to date since the latest one is 4.28.2).

I would recommend to install the latest versions of both. This would ensure that the Dektec library and drivers are aligned. Please retry and let us know the result. Also retry `tsp --version=all` to check the TSDuck, Dektec libraru and Dektec drivers versions.

TSDuck (with embedded Dektec library): https://tsduck.io/download/tsduck/
Dektec device drivers: https://www.dektec.com/downloads/SDK/

st-channel24, 2021-07-05:
updating SW and drivers according to above links : same result.
[feedback_TSDuck_updSW.txt](https://github.com/tsduck/tsduck/files/6763981/feedback_TSDuck_updSW.txt)

lelegard, 2021-07-05:
OK, I think I found an issue with the bitrate with DVB-T2. In the Dektec output plugin with modulators, the default bitrate is computed from the modulation parameters (normally, each specific set of modulation parameters is arithmetically associated with a bitrate). Each modulation type uses a distinct algorithm and the computation was not done with DVB-T2. This is fixed now. The appropriate bitrate should be computed without being forced to specify it.

You may try to rebuild from source or wait until tomorrow for the next nightly build on Windows, [available here](https://tsduck.io/download/prerelease/). You need version 3.28-2446 or higher. Please try tomorrow with the new version.

Now, the 10-MHz vs 8-MHz bandwidth is another issue. How do you conclude that the signal has a 10-MHz bandwidth instead of 8-MHz?

st-channel24, 2021-07-06:
Using the nightly build, shows that DVB-T2 is now playing correctly on both DTU-215 and DTU-315.  This means the 10 MHz /8MHz problem is now also solved.  I think it wa
[feedback_TSDuck_nightly.txt](https://github.com/tsduck/tsduck/files/6769579/feedback_TSDuck_nightly.txt)
s due to something with Dektec that causes it to switch over to 10 MHz mode when some of the configuration parameters are missing or incorrect.  With only the hardware of DTU-315 supporting this mode, DTU-215 gives the DTAPI_E_INVALID_SAMPRATE error instead.

lelegard, 2021-07-06:
OK, so shall we conclude that the issue is solved and can be closed ?

st-channel24, 2021-07-06:
Sure, for me the issue is fully solved.  The solution from TSDuck is far more elegant (and flexible) than the one of sending the bitrate with the output processor.
Thank you !

closed 2021-07-06 by st-channel24