tsduck/tsduck #819: [Q] Ask if it's possible to do IP over DVB (or MPEG-TS) with tsduck utilities

gslin, 2021-07-03:
I want to transport data stream over the satellite network.  The current idea is to listen an UDP port to receive raw UDP data and send to the MUX server.

Currently I could generate a null stream and we could confirm that MUX server (`225.0.0.1:1234`) side had received this null stream:

    tsp --bitrate 1000000 -I null -P mpeinject 224.1.2.3:12345 --pid 8123 -P regulate -O ip 225.0.0.1:1234

Then I want to listen 6666/udp and send to the MUX server, but have failed:

    tsp --bitrate 1000000 -I ip 6666 -P mpe --udp-forward -P mpeinject 224.1.2.3:12345 --pid 8123 -P regulate -O ip 225.0.0.1:1234

I guess it's because `tsp` only accepts TS UDP packet, not raw UDP packet.  Any other way to do something like "IP over DVB" or "IP over MPEG-TS"?

lelegard, 2021-07-03:
There is a confusion between the `ip` input and output plugins which manipulate "TS over IP" and the `mpe` and `mpeinject` plugins which manipulate "IP over TS".

At a higher level, what you want to _precisely_ achieve is unclear. The details are unclear and the devil is in the details.

See a complete example of transporting raw UDP data over TS, with an architecture diagram and detailed commands in section 5.2.21 of the TSDuck user's guide.

Concerning your last example `tsp -I ip ...  -P mpe ... -P mpeinject`, I see a few potential issues.

First, the `mpe` plugin after an `ip` input plugin means that you extract raw UDP datagrams which were encapsulated in TS packets which were in turn encapsulated into UDP. In short, the input must be "IP over TS over IP". Is this really what you mean?

Second, the `mpeinject` plugin means that you insert a second raw UDP stream inside the TS, in addition to the first one, already present on input. Is that what you mean?

Third, the `mpeinject` plugin injects data into the TS. This assume that there is some free space in the TS (null packets). Is that the case? To be sure to understand the various difficulties when manipulating TS, I would suggest to carefully read the reference section on the `tsp` utility in the TSDuck user's guide, chapter 3. Be careful to read the section "Modifying, inserting and deleting packets".

gslin, 2021-07-04:
Thanks for the quick response.

I want to transmit (i.e. broadcast) arbitrary UDP packets via satellite networks, and then received by my set-top boxes with dishes & TV tuner cards.  These UDP packets are not in any standard format, instead, the format is defined by myself, which is encrypted by AES-GCM.

The satellite vendor offers a MUX server and its endpoint (in this case it's `225.0.0.1:1234`), so I try to build the transmitting side.

Currently my idea is to build an UDP listener with `tsp`, but it seems not working in this way.

lelegard, 2021-07-04:
> Currently my idea is to build an UDP listener with tsp, but it seems not working in this way.

On the contrary, it works _exactly_ that way.

What you want to do seems identical to the example in section 5.2.21.2 of the TSDuck user's guide. You need to adapt the PID and service id and name allocation of course.

You also need to understand what your MUX server is _exactly_ expecting on its UDP port: just one MPE PID, a service containing the MPE PID or a full dual service including the IP/MAC notification table (as in the example from the user's guide). Then, adapt the example accordingly.

gslin, 2021-07-04:
Thanks, but my test shows it does not support raw UDP data packet:

    tsp --debug=1 -I ip 5353 -P mpeinject 224.250.250.1:9000 --pid 5003 -O ip 227.0.0.227:12345

And I was using `dig` to generate some sort of UDP packets (and apparently it's a DNS packet, not a TS packet):

    dig @10.0.0.211 -p 5353

The debug logs show that `tsp` was trying to recognize it as a TS packet but failed:

```
* Debug: ip: binding socket to 0.0.0.0:5353
* Debug: mpeinject: server thread started
* Debug: ip: no TS packet in message, 40 bytes
* Warning: ip: detected multiple sources for the same destination 10.0.0.211:5353 with potentially distinct streams
* Warning: ip: detected source: 10.0.0.211:55187
* Warning: ip: detected source: 10.0.0.211:43495
* Debug: ip: no TS packet in message, 40 bytes
```

In the mean time, I was running `tcpdump -n -i eth1 multicast` and see no packet generated.  Still not sure whether my understnading on `mpeinject` is correct.

gslin, 2021-07-05:
Thanks, it's my fault to misunderstand the `mpeinject` command.

`mpeinject` command will listen an UDP port and expect raw UDP packets, and I should not use `-I ip port_num` to listen an UDP port, which will expect UDP packets in TS format.

closed 2021-07-08 by gslin