tsduck/tsduck #820: [Q] Please How can I extract TV Channels that are MPE Encapsulated

picatchoof, 2021-07-05:
Please I Stack in an issue.
**My target is to extract TV Channel that are MPE Encapsulated.**
I succeed to do so using offline tools _(mpe2ts, dvp-ip-mpe2ts, in cmd Lines and also Smartbeam Extractor under Windows)_ to extract the video files. But I need help to use TSDuck to allow extract the streams Live from DVB Tuner. the Extracted TVs must be streamed over LAN (Multicast or even Unicast).

I succeed to extract the the MPE TS as bundle file using the CMD Line:
tsp -v -I dvb --freq 11158000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --modulation 8-PSK -P mpe -p 2002 -P zap 'sb demo' -O file mpe.ts
this is on Eutelsat 7WA

Please the same case can be tested on the TS sample in the TS repository at https://tsduck.io/streams/?name=russia-tricolor-mpe

Thnx a lot

lelegard, 2021-07-06:
Hi @picatchoof,

In your command, you save one service of the outer TS, not the inner TS which is encapsulated in the MPE PID.

If you want to extract the inner MPE-encapsulated TS and stream it over UDP/IP, this must be done in the `mpe` plugin using the option `--udp-forward`. The outer TS can then just be dropped.

Your command should be something like this. Use option `--redirect` to stream it to an address which is different from the one in the MPE-encapsulated UDP datagrams.
~~~
tsp -I dvb ... -P mpe -p 2002 --udp-forward ... -O drop
~~~

To save the content of the MPE-encapsulated TS (and not the outer TS), use:
~~~
tsp -I dvb ... -P mpe -p 2002 -o mpe.ts -O drop
~~~

Please read the TSDuck user's guide, 1) reference section for the `mpe` plugin, 2) MPE examples in section 5.2.21

picatchoof, 2021-07-06:
Hello.
Thanks a lot for your reply.
Forgive me I tried a lot of options (udp-Forward, pid Zap,....) but no result.
I'm not sur if the issue is in my TSDuck cmd or in the Player.
I tried vlc, ffplay, mplayer but unfortunately no relevant results.

Thank you very much.

lelegard, 2021-07-06:
Please _precisely_ describe:
- What is the structure of the input TS
- Which command you use
- What you expect to happen

Please also provide a sample of the full input TS

picatchoof, 2021-07-06:
Hello
Please the Sample File is the same as in the repository.  https://tsduck.io/streams/?name=russia-tricolor-mpe

**_The TS Structure:_**
**My TS sample can be downloaded from:**
https://we.tl/t-RZ8qBQGEtA

**the TS analysis can be downloaded from:**
https://we.tl/t-22hgximTHb

**_The command I run:_** and its result is:
_smartbeam@ubuntu-lab:~$ tsp -I dvb --freq 11158000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --modulation 8-PSK -P mpe --udp-forwar -O drop  ^C* tsp: user interrupt, terminating...
smartbeam@ubuntu-lab:~$ tsp -vvv -I dvb --freq 11158000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --modulation 8-PSK -P mpe --udp-forwar -O drop
* dvb: using /dev/dvb/adapter0 (DVB-S, DVB-S2)
* dvb: tuned to transponder --delivery-system DVB-S2 --frequency 11,158,000,000 --symbol-rate 5,000,000 --fec-inner 9/10 --polarity vertical --modulation 8-PSK --pilots off --roll-off 0.35 --lnb Extended --satellite-number 0
* dvb: signal locked: yes, strength: 36%
* dvb: actual tuning options: --delivery-system DVB-S2 --frequency 11,158,000,000 --symbol-rate 5,000,000 --fec-inner 9/10 --polarity vertical --modulation 8-PSK --pilots off --roll-off 0.35 --isi 255 --pls-code 262143 --pls-mode 63 --lnb Extended --satellite-number 0
* dvb: initial input bitrate is -465,085 b/s
* mpe: extract new MPE PID 0x07D1 (2001), service 0x00C8 (200)
* mpe: extract new MPE PID 0x07D2 (2002), service 0x00C8 (200)_

**_Eexpect to happen_**
in My TS File (the same as the TP received from Satellite) Contains 4 Channels that are MPE.
I'm looking to extract them and make stream on Local Network.
the Stream can be either Unicast or even Multicast.
after That I have to Play them on VLC or other Player.

FYI The TS in the repository (36.0E_12168.835_V_27500.ts) contain 50 Channels.

Best Regards:

lelegard, 2021-07-06:
MPE encapsulates UDP datagrams into a TS. But this can be _any_ form of UDP traffic, not only transport stream, any traffic.

In your description, you seem to expect that a media player can play this UDP content. Do you exactly know the format of the MPE-encapsulated UDP content or do you only speculate that it should be a valid transport stream?

I downloaded your file and extracted one MPE PID, using an hex dump of the UDP payload:
~~~
tsp -I file e7wa-b10.ts -P until --packet 2000 -P mpe --pid 2002 --dump-udp -O drop &>pid2002.txt
~~~

Looking into one UDP datagram (beginning of dump below), you can see that it does not look like a standard TS over UDP, neither RTP. Searching for 0x47 synch bytes, we see some form of TS, but not a standard encapsulation.
~~~
* mpe: PID 0x07D2 (2002), src: 192.168.30.216:51295, dest: 224.30.2.118:6000 (01:00:5E:1E:02:76), 1468 bytes, fragment: 0x4000
  0000:  80 21 21 12 93 90 BF EF 3C 04 EC 00 01 03 89 80  .!!.....<.......
  0010:  00 05 D5 40 B2 33 C1 55 11 84 FF C0 2C 72 66 3A  ...@.3.U....,rf:
  0020:  06 70 F5 3F 0D BB FE 1F FD 86 70 5F B4 71 D6 43  .p.?......p_.q.C
  0030:  4C E8 E7 1C 7A 3F 7E 5A E7 D2 A3 B0 4B E1 B9 18  L...z?~Z....K...
  0040:  81 D2 E6 AA CD 89 19 9E 50 DA FA C2 9D F9 78 C1  ........P.....x.
  0050:  5A 91 7E D9 82 CC 57 9D 8E CC 64 DE 39 BE BF 12  Z.~...W...d.9...
  0060:  D3 E6 E1 73 9F D2 45 61 9F 6A F3 DC 2F 2F C7 2F  ...s..Ea.j..//./
  0070:  C6 A3 C7 82 8C 3A F1 55 C6 D4 26 2C 3A E6 76 47  .....:.U..&,:.vG
  0080:  AA C9 AB E8 C7 BF D5 25 34 43 06 C3 47 00 21 1E  .......%4C..G.!.
  0090:  5A B2 EC 25 66 D1 84 17 06 6D 53 B8 88 16 26 07  Z..%f....mS...&.
  00A0:  FC AC 04 FA 48 7B 43 E5 3B 56 E7 81 01 26 80 CE  ....H{C.;V...&..
  00B0:  F6 59 AA B7 76 B9 CB EC 91 CA 0E 82 5E D4 94 6B  .Y..v.......^..k
  00C0:  10 D4 11 C4 40 0B 24 79 22 DB 18 CC 9C 3F 85 E0  ....@.$y"....?..
  00D0:  61 55 18 C2 E0 4D 29 49 DA 95 7F 4A C6 CE 92 CE  aU...M)I...J....
  00E0:  B9 70 A3 A8 AE B2 A3 6B 96 FF 46 AB B5 B0 56 5A  .p.....k..F...VZ
  00F0:  96 45 2E AD 43 9B 63 FA 0B 36 8C F7 BC 7E 7F AD  .E..C.c..6...~..
  0100:  25 42 78 A0 0F 75 B0 6E 5D 83 88 AB 96 D2 EF 4C  %Bx..u.n]......L
  0110:  0E 6D C0 41 6C E7 5B 62 8E 1E 2D CA 53 C7 31 43  .m.Al.[b..-.S.1C
  0120:  FC 32 FB 0C 87 B0 35 99 A8 DD 16 8C DB 74 6C 2C  .2....5......tl,
  0130:  2D 7D DD CB 2B E5 F0 F6 FA 01 45 A6 4E 9C 7A 1D  -}..+.....E.N.z.
  0140:  4C ED E9 91 FF 6A DB BB 47 00 21 1F B6 2B 3F EC  L....j..G.!..+?.
  0150:  B2 0F 8B A4 04 DA EE 9F 7C B3 FD 80 E8 BF A1 9E  ........|.......
  0160:  93 1D 5A 86 DD 3A 10 5E E7 F7 C2 38 20 59 C3 77  ..Z..:.^...8 Y.w
  0170:  75 41 67 18 A3 1B A4 DE 39 F5 24 BE 37 0B CF F4  uAg.....9.$.7...
  0180:  A0 50 AC 68 0C A6 B5 61 EC 4C 63 1F 1A 6A 6B 10  .P.h...a.Lc..jk.
  0190:  F1 6B CD 85 DB B8 FD 3B 1D 60 E7 FD 2E 82 82 08  .k.....;.`......
  01A0:  60 DC C5 74 64 84 D7 F9 D6 2E C8 86 83 D6 E6 43  `..td..........C
  01B0:  E6 E2 B4 CA 92 B9 79 8A BB 96 98 05 05 39 BA C5  ......y......9..
  01C0:  69 D2 20 23 EB 4B 11 C9 90 5F AF 1F 4B 17 A0 AA  i. #.K..._..K...
  01D0:  BB 39 A0 9C AE 08 05 02 E6 9B 3E 3F D4 70 95 4C  .9........>?.p.L
  01E0:  8C 1E 2F 6A 90 70 29 69 D4 EC 05 D5 04 24 21 F0  ../j.p)i.....$!.
  01F0:  4D 2F 42 93 8B 57 B2 BD 8C E6 CF 39 EC DD 48 F7  M/B..W.....9..H.
  0200:  D8 54 E8 BD 47 00 21 10 C7 E2 A1 00 B5 5A 8B E9  .T..G.!......Z..
  0210:  88 26 D5 A1 87 74 9D 80 10 DF 37 7B 7D D7 66 46  .&...t....7{}.fF
...
~~~

Now, we simulate a UDP forwarding, redirecting on local host. First window (sender):
~~~
tsp -I file e7wa-b10.ts -P regulate --packet-burst 16 -P mpe --pid 2002 --udp-forward --redirect localhost:1234 -O drop
~~~
Second window (receiver):
~~~
tsp -I ip 1234 -O file forwarded.ts
~~~

There is some kind of TS structure:
~~~
$ tspsi forwarded.ts 

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:   16727 (0x4157)
    Program: 16727 (0x4157)  PID:   32 (0x0020)

* PMT, TID 2 (0x02), PID 32 (0x0020)
  Version: 0, sections: 1, total size: 31 bytes
  - Section 0:
    Program: 16727 (0x4157), PCR PID: 33 (0x0021)
    Elementary stream: type 0x0F (MPEG-2 AAC Audio), PID: 34 (0x0022)
    - Descriptor 0: Maximum Bitrate (0x0E, 14), 3 bytes
      Maximum bitrate: 0x000005A0 (1440), 576,000 bits/second
    Elementary stream: type 0x1B (AVC video), PID: 33 (0x0021)

~~~

But `tsanalyze` reports a lot of unreferenced PID's and other inconsistencies. Trying to play the file with VLC, there is some lines of video, then nothing.

So, my conclusion is that the MPE-encapsulated UDP content is _not_ a standard transport stream. This is probably some form of proprietary encapsulation of a TS and the MPE-decapsulated content can only be played or decapsulated in the corresponding proprietary receiver.

Since you mention the sample TS in https://tsduck.io/streams/?name=russia-tricolor-mpe, please note that this file is just an example of MPE encapsulation. The encapsulated UDP content is not a valid TS. As mentioned on the page _"The UDP stream inside each MPE stream contains a proprietary encapsulation of an HLS stream."_ Since HLS contains TS packets, it looks like TS but it is not. This may be the case in your stream.

As a conclusion, before trying to decapsulate the content of an MPE stream, you must exactly know the format of the encapsulated data.

picatchoof, 2021-07-06:
Hello Thanx a lot for your response.

I Executed this command :

`sp -vvv -I dvb --freq 11158000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --modulation 8-PSK -P mpe -p 2002 --udp-forward -O drop
`
to check I run mplayer udp://224.30.2.115:6000

after almost One minute I get a sound and very bad video screen and errors on the Terminal that runs mplayer.
you can view on the pdf at:
BTW You are right and according to the Cahnnels Provider the stream is IPTV-DASH  (they use DASH and HLS Protocol)
https://vimeo.com/288226812

Best regards

-----------------ERRORS ON THE MPLAYER-------------
[aac @ 0x7f6fdd7764c0]channel element 2.7 is not allocated
[h264 @ 0x7f6fdd7764c0]error while decoding MB 5 1, bytestream -10
[h264 @ 0x7f6fdd7764c0]concealing 920 DC, 920 AC, 920 MV errors in P frame
A:9828.2 V:9818.0 A-V: 10.225 ct:  1.140   0/  0  4%  3% 48.6% 20 0 
[aac @ 0x7f6fdd7764c0]decode_band_types: Input buffer exhausted before END element found
[h264 @ 0x7f6fdd7764c0]concealing 427 DC, 427 AC, 427 MV errors in B frame
A:9828.2 V:9818.0 A-V: 10.188 ct:  1.144   0/  0  4%  3% 48.4% 20 0 
[aac @ 0x7f6fdd7764c0]Number of bands (39) exceeds limit (32).
[h264 @ 0x7f6fdd7764c0]cabac decode of qscale diff failed at 11 9
[h264 @ 0x7f6fdd7764c0]error while decoding MB 11 9, bytestream 1363
[h264 @ 0x7f6fdd7764c0]concealing 598 DC, 598 AC, 598 MV errors in B frame
A:9828.3 V:9818.1 A-V: 10.191 ct:  1.148   0/  0  4%  3% 48.2% 20 0 
[aac @ 0x7f6fdd7764c0]decode_band_types: Input buffer exhausted before END element found

MPlayer interrupted by signal 2 in module: decode_audio
nop_streaming_read error : Interrupted system call
[mpegts @ 0x7f6fde1f3600]Invalid timestamps stream=0, pts=884588560, dts=6688971693, size=6459
[aac @ 0x7f6fdd7764c0]Number of bands (49) exceeds limit (41).
[aac @ 0x7f6fdd7764c0]channel element 3.5 is not allocated
[h264 @ 0x7f6fdd7764c0]cabac decode of qscale diff failed at 28 13
[h264 @ 0x7f6fdd7764c0]error while decoding MB 28 13, bytestream 1146
[h264 @ 0x7f6fdd7764c0]concealing 421 DC, 421 AC, 421 MV errors in B frame
A:9828.4 V:9818.1 A-V: 10.254 ct:  1.152   0/  0  4%  3% 49.6% 20 0 

Exiting... (Quit)

lelegard, 2021-07-07:
That is exactly what I suspected. The Broadpeak nanoCDN is now a well-established solution for the so-called "multicast ABR" strategy (a very bad oxymoron in my opinion). This was also the case for the sample stream from the repository.

Since DASH and HLS are bidirectional unicast protocols, the MPE-encapsulated UDP stream is a proprietary unidirectional feed to the Broadpeak nanoCDN receiver, which in turn will act as a DASH or HLS server on the local network of the receiver. In other words, you can easily decapsulate the UDP stream from MPE but there is nothing you can do with it, at least without a Broadpeak nanoCDN receiver.

picatchoof, 2021-07-07:
Dear Lelegard I appreciate your involvement and help.

Can you please check the tools in this Git.

https://github.com/piratfm/eti-tools

The tools such **mpe2ts** and the **dvb-ip-mpe2ts** do extract the TV channels from the The Big TS sample.

My questioning is :
Can we use such tool as piping or similar usage within TSDUCK to extract those channels?

Best regards

EnoSat, 2023-11-17:
13 East - 12207 H
MPE pid - 1012 , one separate video stream (224.30.6.4:6000) + four separate audio streams (224.30.6.4:6020/6021/6022/6023)
- would it be possible to combine these separate streams into one stream using TSduck?
MPE pid - 1013 maybe DASH (224.31.6.9:6000/6001/6002/6003/6004/6005/6020/6021/6022/6023 and 224.31.6.10:600/6020)

picatchoof, 2023-11-17:
Hi All
Good to Know
BTW you can take a look at this webpage it can give you some details about the TPP Content
https://www.digitalbitrate.com/dtv.php?mux=12207&liste=1&live=65&lang=fr

There is PIDs 
**1011
1012
1013**

Is it possible to tell if this TP is DVB-NIP or not?
Also Can you please supply a TS sample?
Thnx a lot

EnoSat, 2023-11-17:
1011 - INT
1012 - IP
1013 - DASH
`
<MPD xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="urn:mpeg:dash:schema:mpd:2011" xsi:schemaLocation="urn:mpeg:dash:schema:mpd:2011 http://standards.iso.org/ittf/PubliclyAvailableStandards/MPEG-DASH_schema_files/DASH-MPD.xsd" type="dynamic" availabilityStartTime="1970-01-01T00:00:00Z" publishTime="2023-11-17T10:05:45.223699Z" minimumUpdatePeriod="PT6S" timeShiftBufferDepth="PT36S" maxSegmentDuration="PT7S" minBufferTime="PT2S" profiles="urn:mpeg:dash:profile:isoff-live:2011,urn:com:dashif:dash264,urn:hbbtv:dash:profile:isoff-live:2012">
  <Period id="1" start="PT0S">
    <BaseURL>dash/</BaseURL>
    <AdaptationSet id="1" group="1" contentType="audio" lang="en" segmentAlignment="true" audioSamplingRate="24000" mimeType="audio/mp4" codecs="mp4a.40.2" startWithSAP="1">
      <AudioChannelConfiguration schemeIdUri="urn:mpeg:dash:23003:3:audio_channel_configuration:2011" value="2"/>
      <Role schemeIdUri="urn:mpeg:dash:role:2011" value="main"/>
      <SegmentTemplate timescale="48000" initialization="orp01-ten012-ch010-$RepresentationID$.dash" media="orp01-ten012-ch010-$RepresentationID$-$Time$.dash">
        
        <SegmentTimeline>
          <S t="81610344145599" d="288768"/>
          <S d="286720"/>
          <S d="288768" r="1"/>
          <S d="286721"/>
          <S d="288768"/>
        </SegmentTimeline>
      </SegmentTemplate>
      <Representation id="audio_133600_eng=131600" bandwidth="131600">
      </Representation>
    </AdaptationSet>
    <AdaptationSet id="2" group="2" contentType="video" par="16:9" segmentAlignment="true" width="1280" height="720" sar="1:1" frameRate="25" mimeType="video/mp4" codecs="avc1.4D001F" startWithSAP="1">
      <Role schemeIdUri="urn:mpeg:dash:role:2011" value="main"/>
      <SegmentTemplate timescale="600" initialization="orp01-ten012-ch010-$RepresentationID$.dash" media="orp01-ten012-ch010-$RepresentationID$-$Time$.dash">
        
        <SegmentTimeline>
          <S t="1020129301811" d="3600" r="5"/>
        </SegmentTimeline>
      </SegmentTemplate>
      <Representation id="video=3621600" bandwidth="3621600" scanType="progressive">
      </Representation>
    </AdaptationSet>
  </Period>
  <UTCTiming schemeIdUri="urn:mpeg:dash:utc:http-iso:2014" value="https://time.akamai.com/?iso"/>
</MPD>`

picatchoof, 2023-11-17:
Thnx a lot for that.
Also is there a tools (Linux or Windows) that allow to extract and view what encapsulated in such Streams either Live or for Offline TS files
Thnx

picatchoof, 2023-11-17:
@EnoSat Can you please tell what the plateform that shows you such details?
I tried the https://flysat.com/en/channel/64847/eutelsat-poc  but don't give enough details like what you have on the Screenshoot
BR

EnoSat, 2023-11-17:
Own creation, originally intended for Flysat updates, but it stopped providing details about DVB-IP some time ago
http://enosat.euweb.cz/dvbip.htm

picatchoof, 2023-11-27:
Hi all
Hi @EnoSat 
Can you please tell more about this two TPs on Hotbird.

TP-6 11317 V 27500 8psk.
TP-8 11355 V 27500 8psk.
Best regards.

EnoSat, 2023-11-27:
11317 V tested DVB-NIP - https://dvb.org/?standard=native-ip-broadcasting
11355 V tested DVB-I - https://dvb-i.tv/ (one could also talk about some kind of OTT modification)

andimik, 2025-09-06:
Just for your information

On Hotbird 13E there is again a transponder with DVB-NIP. @EnoSat have you tried to use tsduck for it? 

 (12226 V, 27500, 3/4, 8PSK)

https://www.satellites.co.uk/forums/threads/dvb-nip-on-hotbird.183386/

lelegard, 2025-09-06:
I am not knowledgeable on DVB-NIP. As far as I understand, the way IP packets are encapsulated depends on the modulation:

- DVB-T2 and DVB-S2X: GSE, in BBFRAMEs
- DVB-S2: MPE, in transport stream packets

By definition, TSDuck can only process transport stream packets. So, if that transponder is DVB-S2, you should get a TS with MPE streams (which can be extracted using the plugin `mpe`).

I not longer have access to satellite reception. It would be great if someone could record a few minutes of that transport stream (if it is DVB-S2 and if I am not wrong about the encapsulation of DVB-NIP over DVB-S2).

EnoSat, 2025-09-07:
[image]
https://mega.nz/file/FRUHHDIb#hUGiuYnwFPlQ8kAdJgLr4R0AR-cBn5jh4VcGoB_5Whw

picatchoof, 2025-09-07:
Hi All
Great Job. I think it's put on Air especially for the IBC, bacause all are doing Demos and testing this new Era of boroadcasting.
Eutelsat from their side and also the Others Enensys and Inverto.
but be carefull. It might be not fully complaint DVB-NIP format. becaus if it generated using **bpk** Systems it's NOT but if its other big chance to be the Right one.
I know that they have other TPs runing in 7°West but I think not easily visible from Amstredam.
Good luck

picatchoof, 2025-09-07:
The Idea is to deliver Dedicated Boxex Qualified as Pro within sufficent Storage.
The Box can receive from Satellite the Live Channels, make them available for users that are on the Same Wifi / LAN.
Those users can view both Live Channels and can also view Movies that are stored and remain available offline (Satellite Only No Internet). 
Yes kind of Push VOD.

picatchoof, 2025-09-07:
BTW My TS sasmple is
https://mega.nz/file/LLhTAZjA#gqvnjO4x132TsKwCpE9vlUazbIAysxz6uVpeEX9DO1s

This Sample was made while doing demos of file sent to satellite using Inverto DVB-NIP Plateform.
I'm Looking for tools to be able to extract the content from this kind of streams

Thanx

picatchoof, 2025-09-08:
> tsp -I file ts-12032025.ts -P mpe --pid 0x0190 --summary -O drop

This is **not working** under windows.
Need to restart my Lab (Linux) and do testing.

BTW ther is two DVB-NIP ongoing signals on Hotbirb 13.

1st: Hot Bird (13.0E) **12225 V / TP-76**  
DVBS2 TSid:7600 ONid:318 8PSK 27500 3/4 CCM Pilots:ON Roll_off:0.20
The Up Stream is done using Enensys MediaCast for Encapsulation and data casting

2nd: Hot Bird (13.0E) **12380 V / TP-84**
DVBS2 TSid:8400 ONid:318 8PSK 30000 5/6 CCM Pilots:ON Roll_off:0.20
The Up Stream is done using Inverto Airstreamt for Encapsulation and Data casting

lelegard, 2025-11-28:
@EnoSat, @andimik, @picatchoof, there is now a plugin `nip` which analyses the content of the DVB-NIP carousel, extracts specific files or the entire `http://dvb.gw/` carousel.

The plugin `nip` works on a DVB-NIP stream in a MPE PID in the source TS.

There is also an equivalent command `tsnip` which works on a pcap file where the UDP packets of the DVB-NIP stream are captured. This is useful when the DVB-NIP stream is over GSE instead of MPE.