tsduck/tsduck #821: Thought for Teletext injection?

SonnyWalkman, 2021-07-08:
I require a way to inject teletext into a MPTS used at a venue.

Is there any plans to develop a plugin or inject teletext as a PES with TS Duck?

Currently the only way to do it is using Open caster however the teletext is static and very convoluted using a python script to produce the PES. 

Here  https://github.com/peterkvt80/vbit2 is a novel way to generate the raw teletext data (VBI) using .tti files however, how to format these and "inject" the PES in the MPTS is the enigma.

Not sure if there is any similar minded people wanting to keep teletext alive?

SonnyWalkman, 2021-07-28:
Dear Lelegard,
I know I keep pestering about adding a plug-in to allow adding a Teletext stream into a MPTS?
I understand there is maybe no call ambition other than me to add such however, I'm using TSDuck and found it relatively easy to implement once reading the documentation a few times.

I'm happy to donate some $$ to assist with time to develop a plug-in for TS duck if that's something you prefer?

You may think teletext is dead? Many do however I'd like to use teletext to show CPU load, memory and a list of other parameters as a page on our DATV repeater project. It's a Novel way to show information and have other pages of txt containing other useful information. Weather etc. I have affinity for the teletext service long live CEEFAX!!

I'm wanting to use TTI files from vbit2 another project use output from to inject into PES. 

Opencaster allows such however, I feel TSDuck is easier to automate using python than opencaster? I.e generation the SI tables etc.

Let me know how you feel about it?

Thanks Sonny.

lelegard, 2021-07-31:
Hi @SonnyWalkman,

Developing such a plugin is quite possible, although more complex than its extraction counterpart. It is just a matter of time and motivation. I have currently none of them. Maybe someone else...

SonnyWalkman, 2021-09-27:
Thanks lelegard.. I understand your time is very important and devoted to improving TSduck and many other projects your working on such as the new SRT features and bug fixes.. I appreciate your ongoing commitment to TSduck's ecosystem!   Do you remember Disco Duck in the 80's! hahaha TSDuck-> DISCODuck.

I have a project where I'm encoding  2 x HD channels both raw YUV and S16 via fifo from  running gstreamer pipelines res -1280x720 piped into ffmpeg with options -b, -maxrate & -minrate.   side note- Gstreamer's 1.14.4 mpegtsmux element seems not produce CBR as smooth as ffmpeg in testing. 

I'm using a Nvidia Xavier NX board (AVR based) which is dedicated to video encoding utilising its power of hardware cuda cores muscle to 2 pass encoding to improve CBR streams.
TSduck does the rest for replacing PAT, PMT and SDT and injecting fresh NIT, BAT, EIT, TDT and TOT.. Pretty much happy with how it going so far. I'm using the recent python wrappers. Great!

I'm 'wishing' to inject Teletext data (10 of so teletext pages) into the stream to show statistics such as stream/s PID's values, CPU %, GPU% (cuda usage) and lastly memory% as service information via these teletext pages.
Teletext to most is pretty much dead these days, I personally have a affiliation with teletext and request to use it as a tool in my project as for others down the track.

I understand, TSduck already extracts the SRT (subtitles) from the stream from the designated stream PID to file? A plugin (Ttsplugin_teletextinject.cpp) could twofold,  serve the purpose of injecting both the SRT for subtitles and the standard text pages. Use a file?? (no sure of the format consider *.tti ) and the ability to push pages via UDP as a method of input. option switches to give choice of reverse PES and inject timing into null packets?

I've recently made a donation for your continued work and happy to donate when for the primary purpose of:
1. Keep coffee in your hand. (good stuff not instant cough cough.. )
2.  Persuading you to consider writing a plugin  😅 😃  

Kind regards
Sonny.

SonnyWalkman, 2022-12-20:
> Developing such a plugin is quite possible, although more complex than its extraction counterpart. It is just a matter of time and motivation. I have currently none of them. Maybe someone else...

Hello [lelegard],
Has there been any thought on a teletext inject plugin to inject data into a TS stream for both SRT (Teletext subtitles) and standard Teletext PES? 

I understand there's the plugin 'Teletext' for the extraction of Teletext subtitles in SRT format however nothing to put it back or add Teletex PES date to a stream possible to adapt 'datainject' plugin?
I'd assume this could be done using available null packets and use either a static file to read from or use a UDP socket with a cache to hold data whilst in waits a predeterminate rate to inject in the TS.

Please reconsider.

I'm happy to donate $$ for your time.

lelegard, 2022-12-21:
> Has there been any thought on a teletext inject plugin to inject data into a TS stream for both SRT (Teletext subtitles) and standard Teletext PES?

"Thought" : yes, available time : no.

> I'd assume this could be done using available null packets and use either a static file to read from or use a UDP socket with a cache to hold data whilst in waits a predeterminate rate to inject in the TS.

Yes, that's the way it should be done, except that, instead of a predetermined rate, the Teletext frames should be synchronized with the PCR of the service, based on SRT timestamps.

> Please reconsider.
> I'm happy to donate $$ for your time.

This is not a matter of $$. This is a matter of time and I do not have any for that kind of demanding stuff.

This is the way free and open source software work. It is developed on good will by volunteers, mostly based on their own need. I developed TSDuck when I needed it and added new features when I needed them. I published it as open source in grateful return for all the valuable open source software I had used so far for free.

I also add new features from users requests when those features are not too time consuming and they bring a significant common value to the project. This one is too time consuming. I don't think you realize that injecting Teletext from SRT is way more complicated than injecting EMM's (plugin `datainject`).

Several TSDuck plugins or features were developed by users when they needed them. If you desperately need Teletext injection, please develop it and make sure you return the code back to the project, as a contribution for all the benefits you got from open source projects, TSDuck and others.

SonnyWalkman, 2022-12-21:
Thanks for your response Thierry,
I have read through most of the TSduck requests and have noticed actions on
other time consuming functions which get actioned. One example is
tsswitch.  I guess teletext is long gone with nobody ever wanting to have
in muxed into TS streams, I can see theres is less of an effort.

I’m not wanting to produce a full SRT subtitling soulution btw , which I
can see this being complicated task rightfully so.

My requirement to to push standard old teletext PES already formed from
another application via udp socket into the TS stream timed by PCR as you
have indicated?

https://en.m.wikipedia.org/wiki/Teletext

I’m not a C++ developer although I could take a look at what others have
done a far as a plugin? Your documentation is outstanding.  Between plugin
template and how open caster they implemented   injecting teletext which is
now unfortunately unsupported, I could work on a plugin. (Teletextinject)

Thank-you for your consideration
Btw, I haven’t use any TSduck functions so far. I like the concept and have
donated on two occasions.

Rob

SonnyWalkman, 2023-02-23:
> Here is my solution. It's not exactly what you're looking for, it inserts
> "static" carousel of teletext from .t42 file. And when the file ends, it
> seek(0) and transmit again.
> But I may try to modify it to get the pages from named pipe, instead of
> static file.
> What's the difference between .t42 and vbit format?
> teletext.zip

Hello omikron88,

Thank you for your email.
I’ll look into what you have done.
I’ve stepped back from using TSduck and are looking at a Gstreamer generic
way of muxing teletext.
Vbit GitHub.
https://github.com/peterkvt80/vbit2

Does the hard work of reading the .tti files in directory  and converting
each to .t42 pushing the .t42 file through stdout.
Using
https://github.com/peterkvt80/vbit-py

Could push it to a dedicated udp port other than using stdout.

There is no sync with video clock unfortunately.

Regards

Rob

omikron88, 2023-02-23:
My parser includes timestamp at start of each PES unit. But not a PCR.
The 0 and 3600 command line parameters says initial and increment of
timestamps, in 90 kHz clock.

Now I am testing vbit2 too.

Tom

SonnyWalkman, 2023-02-23:
Hi omikron88,
To add to my last email,
from my request vbit2 does actually generate an elementary stream, VBIT2
does all that is needed from a front end standpoint managing the carousel
directory, adding/modifying and deletion of .tti files etc.
Check out peterkvt80 <https://github.com/peterkvt80>/vbit2
<https://github.com/peterkvt80/vbit2>  on github, The comments and
documentation are well maintained. In addition, he's easy to contact and is
open to improvements. I see vbit2 does all what is needed. How to mux into
a running TS is another story.

Taken from the last vbit2 update:-
*"VBIT2 can also optionally generate output a Packetized Elementary Stream
for insertion into an MPEG transport stream for DVB teletext"*

It would be great to develop a tsduck backend plugin to inject the teletext
timed with video PTS. via a dedicated udp port bypassing stdout. In
addition to a full  C code implementation without needing to use a string
of commands to do such,  something on the lines of :-

*tsp -v -b $bitrate -I null \*
*-P textinject -p 6666 -b 4700  ...  (-p port number, -b bitrate  -<other
options etc etc>)*
*-P pmt ..*
*etc*
*etc*
*-O hides -b 8 -c ...*

I haven't looked at how tsduck plugins are developed so no help.
I've requested Thierry to look at developing such with a good donation
however, from his return comments, he's time poor or neither interested in
supporting such.

Anyway, thanks for your consideration and effort in a solution. Food for
thought, I will give it a try and see if I can make any improvements.

Cheers

Rob

SonnyWalkman, 2023-07-16:
> My parser includes timestamp at start of each PES unit. But not a PCR. The 0 and 3600 command line parameters says initial and increment of timestamps, in 90 kHz clock. Now I am testing vbit2 too. Tom Dne čt 23. 2. 2023 21:32 uživatel Sonny Walkman ***@***.***> napsal:
> […](#)

To create a proper PCR clocked TS stream, from looking at TSDuck documentation, I think the appropriate way and commented by a few is to create a new TS and inject the Teletext data.
Thierry, may be able to assist with the proper process?  There has to be a simple way? omikron88 method above works however, due to the lack of PCR when injecting, the PCR is inconsistent. pcrverify goes crazy..

BITRATE=400000
PCR_PER_SEC=5
PCR_DISTANCE=$(( 
(( $PCR_PER_SEC * 188 * 8 )) ))
PCR_PID=5004
tsp --verbose --bitrate $BITRATE --max-flushed-packets 70
-I null
-P regulate --packet-burst 14
-P filter --every $PCR_DISTANCE --set-label 1
-P craft --only-label 1 --pid $PCR_PID --no-payload --pcr 0
-P continuity --pid $PCR_PID --fix
-P pcradjust --pid $PCR_PID
-P inject PAT.xml --pid 0 --bitrate 15000
-P inject SDT.xml --pid 17 --bitrate 15000
-P merge -b 4700 "python ttxgen.py ttx.t42 2 0 3600 1024"
-P pmt --patch-xml TTX.xml
-P inject -p 16 NIT.xml=2000
-P inject -p 20 TDT.xml=1000
-P nit --build-service-list-descriptors
-P pcrverify
-O ip 224.0.0.3:4500 --packet-burst 7 --enforce-burst

What do you think?
My goal is to : "Keep Teletext ALIVE! on DVB"

SonnyWalkman, 2023-11-03:
Hello @omikron88 ,
Thank you for you incite in developing your cool code snippet. Works

I've ported your ttxgen.py python code into a Go executable adding cli ARG and error logging with quiet option.
 I also read in the T42 file into a buffer and read the buffer continually. to reduce the IO overhead with opening and closing file descriptors.

I'm now embarking on dynamically creating teletext pages and adding a header with data time and was hoping for your assistance in calculating the PTS to be dynamically calculated. 

> My parser includes timestamp at start of each PES unit. But not a PCR.
The 0 and 3600 command line parameters says initial and increment of
timestamps, in 90 kHz clock.

These settings ttxgen.py txt.t42 `2 0 3600 1024` are tailored for the two-page test txt.t42 file.  pages 898 and 899.

I'd like to calculate the PTS count and bitrate for(x) number of pages which is I calculate on startup know the total of pages to packetize to PES.  Are you able to guide me on what is required here? it be nice to add PCR to make the PES more in line with what ffmpeg would produce without the Video and Audio streams.

Regards

Rob

FlexMcMurphy, 2024-10-13:
@SonnyWalkman 

I've been playing with injecting DVB Subtitles into transport stream files using ffmpeg. Only out of curiosity I am trying to find out is it possible to create a teletext subtitle stream and mux that in to a .ts file.

I am interested in your comments in this thread and feel your pain about shouting for help but not getting much help it seems. Are you making any progress? Would you have any advice for me about my goal? It sounds like it might be related to what you are trying to do?

Cheers,

Flex

SonnyWalkman, 2024-10-13:
Hi @FlexMcMurphy 

I've had no success with Teletext with ffmpeg which I believe ffmpeg needs to be build against zvbi library to allow teletext and subtitles to be multiplexed. Despite early comments I'm yet to see any working example in doing so. As it stands both ffmpeg and mpegtsmux in Gstreamer aren't able to properly do teletext since teletext needs a PCR timed with video content however saying this, there is a hacky method to generate a PCR and have TSduck replace the teletext PCR with your video PCR. 

Subtitles using subrip files works with ffmpeg. Have you played with SRT format?Pleanty on the internet on how to create SRT files and mux using ffmpeg. There has been requests made years ago to properly implement dvb teletext however, I don't believe anything has happened since like myself, there are only a few working or wanting use dvb teletext?

https://trac.ffmpeg.org/ticket/5912

To mux teletext at the moment, I'm using vbit2 which has an output config mode -ts to generate a dvb ts stream via stdio with its own PCR to multiplex into TSduck which I use 

I find it works however, the bitrate needs to be consistent all the time filling each ts packet with enough pages to fulfill a complete ts packet. If pages are less, then the packet needs to be null to packed to keep the bitrate at 36000 in my implementation. 
- I'm hoping @lelegard to implement a teletextinject plugin to do it properly which take into account variable rate of teletext. Create a internal teletext PES buffer to monitor teletext PES data coming in and pack to match the required consistent bitrate and then use the PCR for the video to time the teletext correctly with service you selected to add teletext with. Modify the PAT to include the teletext service and add the service PMT needs to be changed.

FlexMcMurphy, 2024-10-13:
Hi @SonnyWalkman.

Yes I figured out how to add a .srt of subtitles to a .ts file. I got a bit too much in to subtitles in .ts files and was wondering since a dvb stream of subs can be added to a .ts container how can a stream of teletext subtitles be added to a .ts container?

As you say it looks like ffmpeg cannot do this at the moment and other software is needed that costs money. One person suggested I look in to [OpenCaster](https://github.com/aventuri/opencaster) but I have no idea where to start with setting that up and maybe other hardware is needed for it to work.

I wonder would OpenCaster be useful for your use case? It seems you already have a lot of ideas about how to progress your project. How about starting your own GitHub project about it and maybe others will offer to help? You could teach yourself C or or C++ and get something small working. There might be developers who are retired now with free time and experience of programming with teletext applications who would be more likely to advise if you have started something?

Cheers,

Flex

SonnyWalkman, 2024-10-13:
Hi @FlexMcMurphy 
I've used opencaster however hasn't been actively supported for years. Check out this fork of opencaster
https://github.com/DeYangLiu/OpenCaster

Gives example of muxing teletext however reading the manual show how to add teletext and subtitles. Page 33.

http://www.avalpa.com/assets/freesoft/opencaster/AvalpaBroadcastServerUserManual-v2.1.pdf

Opencaster is a set of command line tool which pipe to each other in series.
There is a python tool to generate the test teletext therefore this could be modified to create subtitles 0x02h teletext with 0x03h indicating subtitles.

I'm working with go programming which could recode the txt2pes and pes2ts tools to do the teletext? However there need to have code to generate subtitles and correctly sync with DTS and PCR to correct show with the video.

Someday I'll developed my own teletext encoder which correctly uses the video PCR to sync the teletext. Be great if ffmpeg could do it.