tsduck/tsduck #822: [Q] How create/use pop up notification in DVBT

antoineagscom, 2021-07-12:
Hi,

I had two questions about the dvbt. I would like to be able to display a notification (like a pop up). 

Is there a way to do this in the DVBT standard (scte-16 maybe) ?
And if yes how to inject this with tsduck ?

Thanks in advance

lelegard, 2021-07-12:
What do you mean by notification? On the TV set? In that case, this is a matter of application on the TV set or STB. Usually, if HbbTV is supported on your receivers, there should be a way.

Once you know what to insert in the TS, tsduck will probably be able to insert it.

I don't see how SCTE16 can help.

antoineagscom, 2021-07-13:
Hello, thank you for the answer.

I would like to send a message or a list of characters that would be displayed on a TV without interrupt a current stream.
(like when there is a channel update, a pop-up appears System Software Update Service (TS 102 006 [11])). 
Or send subtitles but I don't know where to start.

I was just wondering if it was possible since you have a better vision than me, 
I thought I read that the "Cable Emergency Alert Table (SCTE 18)" looked like what I wanted to do (p 448 of the user's guide)

The HbbTv only supports some Smart TVs I guess ?

lelegard, 2021-07-13:
SCTE 18 (not 16) is the US emergency alert system. It requires a corresponding EAS application on the receiver which is mandatory on the US market (ATSC receivers). DVB-T is used on the European market. I am not sure that the DVB-T receivers support EAS.

closed 2021-07-13 by antoineagscom