tsduck/tsduck #823: problem about HIDES ut100c [Q]

xlsau, 2021-07-12:
hi,

I try testing HIDES UT100c modulator on my laptop. my laptop work with Linux mint  Linux Mint 19.1 by Linux 4.15.0-147-generic kernel. my problem is when I use "tsp -I hls http://x.x.x.x:1935/ffmpeg/test1/playlist.m3u8 -O hides -a 0 -f 474000000" command, device give error message "tshides: error getting device type on /dev/usb-it950x0: 0x00000008 (RSD counter not ready)". and Than I checked my laptop via "tshide" and "tshides -v" what I get these result at the below side :

root@Canavar:~# tshides 
tshides: error getting device type on /dev/usb-it950x0: 0x00000008 (RSD counter not ready)
tshides: error getting device type on /dev/usb-it950x1: 0x00000008 (RSD counter not ready)
tshides: error getting device type on /dev/usb-it950x2: 0x00000008 (RSD counter not ready)
0: "usb-it950x0" (/dev/usb-it950x0)
1: "usb-it950x1" (/dev/usb-it950x1)
2: "usb-it950x2" (/dev/usb-it950x2)

root@Canavar:~# tshides -v
tshides: error getting device type on /dev/usb-it950x0: 0x00000008 (RSD counter not ready)
tshides: error getting device type on /dev/usb-it950x1: 0x00000008 (RSD counter not ready)
tshides: error getting device type on /dev/usb-it950x2: 0x00000008 (RSD counter not ready)
Found 3 HiDes devices

Index ........... 0
Name ............ "usb-it950x0"
Device .......... /dev/usb-it950x0
Chip type ....... 0x9507
Driver version .. v18.04.16.1w
API version ..... 1.3.20160929.0
Link firmware ... 255.39.2.0
OFDM firmware ... 255.9.11.0
Company ......... ITEtech
Hardware info ... Eagle DVBT

 Index ........... 1
 Name ............ "usb-it950x1"
 Device .......... /dev/usb-it950x1
 Chip type ....... 0x9507
 Driver version .. v18.04.16.1w
 API version ..... 1.3.20160929.0
 Link firmware ... 255.39.2.0
 OFDM firmware ... 255.9.11.0
 Company ......... ITEtech
 Hardware info ... Eagle DVBT

  Index ........... 2
  Name ............ "usb-it950x2"
  Device .......... /dev/usb-it950x2
  Chip type ....... 0x9507
  Driver version .. v18.04.16.1w
  API version ..... 1.3.20160929.0
  Link firmware ... 255.39.2.0
  OFDM firmware ... 255.9.11.0
  Company ......... ITEtech
  Hardware info ... Eagle DVBT

root@Canavar:~# lsusb
Bus 001 Device 005: ID 2232:1024 Silicon Motion 
Bus 001 Device 003: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 026: ID 048d:9507 Integrated Technology Express, Inc. ITE it9507 full featured DVB-T transmission chip [ccHDtv]
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I haven't been able to find anything similar to this on the internet yet. do you have any idea?

lelegard, 2021-07-12:
Where did you get the HiDes drivers from? It has been reported that the drivers on the CD which comes with the device are outdated. 

Up-to-date drivers are available here : https://tsduck.io/download/hides/

xlsau, 2021-07-13:
hi, my diriver is hides-dkms_18.04.16.1w_all.deb.. I useded same source.

lelegard, 2021-08-13:
Hi @xlsau,

Could you please check for device type error during the ITE driver initialization in the kernel?
Here is the command and my output. Please post yours.
~~~
$ sudo dmesg | grep -A2 getDeviceType
[  240.862967] EagleUser_getDeviceType
               	- Device Type: 11 load from EEPROM
[  240.892360] RX SlaveII address: 3A
~~~

From your report, I suppose that you have three different HiDes devices which are plugged in at the same time. Could you also try with only one instead of three? I have no idea if it makes a difference but I have only one HiDes device and never tested with more than one.

Since the chip and device type are informational only and not used elsewhere in TSDuck, I just made a modification to ignore errors during the fetch of these information. Such errors will now be logged at `--verbose` level or higher only and will not prevent using the device. Although this is not normal, there is no reason it should prevent the modulator from working.