tsduck/tsduck #824: [Q] UDP IP Stream (TS over IP) DVB S2.

maasuu, 2021-07-13:
I have UDP IP stream from Novra s400 pro with UDP paylaod of 1316 Bytes, how to extract the paylaod using tsduk,.

lelegard, 2021-07-13:
Use the `mpe` plugin. Please read the user's guide, the reference section for this plugin and the example chapter.

picatchoof, 2021-07-13:
Hello
I had maybe the Same Issue: I get Help from the Forum Below
I download the tool dvb-ip-mpe2ts that I used after TSDuck (Pipe)
https://www.sat-universe.com/archive/index.php?t-301973.html
https://github.com/piratfm/eti-tools

Using TSDUck I succeed to isolate the MPE TS (Thanx to te lelegard in his advises)
I used especially the ZAP and the MPE Plugins.
After that either you store the output in FIFO File File that will be used by dvb-ip-mpe2ts or you can Stream it internally directly to dvb-ip-mpe2ts 

dvb-ip-mpe2ts will extract the TS videos from IP/DVB  to output can be streamed or stored as local video.
take a look to my Question it may help 
https://github.com/tsduck/tsduck/issues/820
Good luck

picatchoof, 2021-07-13:
Can you share a Sample of TS file that contain the DATA expected to be extracted?

maasuu, 2021-07-14:
> Can you share a Sample of TS file that contain the DATA expected to be extracted?

sure sir, its is .pcap file captured by wireshark.

maasuu, 2021-07-14:
/home/satlab/Desktop/dump.ts
sir its that

maasuu, 2021-07-14:
[ts.zip](https://github.com/tsduck/tsduck/files/6813760/ts.

maasuu, 2021-07-14:
its is the ouput of TBS dvb s2 tuner, but in novra s400 pro  output is Ethernet interface. i capture it with wireshark, .pcap

maasuu, 2021-07-14:
sir, i used "tsp -P mpe -s 232.1.1.1:6000 -o d:\mpeg2.ts"  command the file name is created, but no data. how to capture the network stream using tsduck. 
2. next i used "tsp -I ip 192.168.1.2@232.1.1.1:6000" command, data is showing but cant able save that.

picatchoof, 2021-07-15:
Hello Again.
The ts dump is hardly downloaded because the URL was wrong : [ts.zip](**https://github.com/tsduck/tsduck/files/6813760/ts.**

instead I used https://github.com/tsduck/tsduck/files/6813760/**ts.zip**
Never mind. But this TS it can not be Analyzed by any tools;
I Used **tsanalyze** From TSDuck Emptu Result.
I opened the Dump.ts using Wireshark as I do within the TSs I generate Gives Error.
Even TSReader can not handle this sample.

I'm not sure if in your case you may need TSDuck to deal with IP output.
My be try other tools can give relevent results.
Good Luck.

picatchoof, 2021-07-15:
If you're sur that your Box gives MPE at the output you can use dvb-ip-mpe2ts it will help separate the content encapsulated in the Out IP Stream.

BTW did you tried to read the Output file using VLC, FFPLAY from ffmpeg?
Also try to analyse your Box output in this tool it's really powerful
https://www.tsreader.com/purchase/purchase/downloadlite.html

maasuu, 2021-07-16:
> Use the `mpe` plugin. Please read the user's guide, the reference section for this plugin and the example chapter.
hi iam using novra s400 pro at my end, the output captured by wireshark is only udp, how tsduck decap the stream, 
[image]

picatchoof, 2021-07-16:
Hello
Can you possibly give the scheme of what you want to achieve as a montage.
It seems that the Nova picks up the DVB-S2 signal and transmits it to the Ethernet (IP) port.
What is the nature of the input signal?
And what is the nature of the PIDs / TS Streams that you are looking to output via Ethernet?

maasuu, 2021-07-16:
> Hello
> Can you possibly give the scheme of what you want to achieve as a montage.
> It seems that the Nova picks up the DVB-S2 signal and transmits it to the Ethernet (IP) port.
> What is the nature of the input signal?
> And what is the nature of the PIDs / TS Streams that you are looking to output via Ethernet?
> 
> If possible provide the Satellite and the TP data
> BR

hi, that we receiving dvb-s2 signal is VCM coded, we don,t know the PIDs, "tsp -I ip 192.168.1.2@232.1.1.1:6000" command displaying some protocols on that stream in raw format.

picatchoof, 2021-07-16:
What about Satellite and Transponder you are using?

maasuu, 2021-07-16:
> What about Satellite and Transponder you are using?

turksat 4b/ tp 11582/H/2350

closed 2026-04-26 by github-actions[bot]