tsduck/tsduck #826: [Q] eitinject usage?

peter-stanton, 2021-07-14:
I'm working on some scripts that use tsduck to generate test TS files. I've managed to create an EIT xml file, inject the table using tsp and the TV picks up my EIT data. But the table it injects is static across the whole file and I'd like to simulate one event ending and the next one starting. 
Does the eitinject module take the stream time into consideration and modify the EIT content as the stream plays? And how do you even use it? The user guide mentions that it takes a database of event data, but doesn't say what format it's in.

lelegard, 2021-07-31:
Hi @peter-stanton,

> The user guide mentions that it takes a database of event data, but doesn't say what format it's in.

It does. Please read the user's guide carefully:

> __Electronic Program Guide (EPG) database__
>
> The events are loaded in an in-memory EPG database. There are several sources for events: files and incoming EIT’s (from the TS upstream).
>
> Event files shall contain EIT tables or sections in binary, XML or JSON format. The organization of events in the EIT’s and the type of EIT’s are ignored. Only the events descriptions and the DVB triplets (service id, transport stream id and original network id) are important, all the rest is ignored.
> 
> When event files are loaded or when incoming EIT’s are received, all events are individually extracted and stored in the in-memory EPG database. The EIT encapsulation is just a convenient pre-existing format to store events, nothing more.
>
> The event input files can be specified using wildcards (be sure to use quotes in order to avoid the interpretation of the wildcards by the shell). The eitinject plugin polls the corresponding files at regular intervals. Whenever a file matching the wildcards is created or updated, the file is loaded. Existing events are ignored.
>
> Events are automatically removed from the EPG database when they become obsolete (the current transport stream time goes beyong the end time of the event). There is currently no way to delete an event from the database before its completion.

In short, all your events must be described in XML, JSON or binary as if they were in an EIT. As mentioned _"The EIT encapsulation is just a convenient pre-existing format to store events, nothing more"_. The real EIT's which will be injected are rebuilt from all the input events, according to the current time..

deonstz, 2022-08-04:
Hi Peter, I have been struggling to get my xml file injected for abt 2 weeks now. Must be doing something very basic wrong. Would you mind sharing your file with me? Else I will post a new issue.