tsduck/tsduck #827: version `GLIBC_2.33' not found on Ubuntu 20

bholly, 2021-07-20:
- Precondition: Ubuntu 20.04.2 LTS with tsduck 3.26 installed.
- Step: Execute "tsversion --upgrade" on the command line.
- Result: After upgrade tsduck 3.27 does not work and shows following error: 
-- tsp: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /usr/lib/libtsduck.so)
- Expected: tsduck 3.27 should work after upgrade.
- Additional information: it seems, tsduck_3.27-2438.ubuntu21_amd64.deb is based on Ubuntu 21 with a higher glibc version. However, if this version is incompatible with Ubuntu 20 it should not be ignored by "tsversion --upgrade".

picatchoof, 2021-07-21:
Hello.
Unfortunately it happened to me.
I was obliged to downgrade TSDuck to continue working as usual.

Good luck

lelegard, 2021-07-31:
Hi @bholly and @picatchoof,

TSDuck pre-built binaries are provided for some distros only and are built for the latest version of the distro only (21.04 in the case of Ubuntu). For other distros or older versions of the distros, you have to rebuild TSDuck yourself. On Linux, this is quite easy and [explained here](https://tsduck.io/doxy/building.html). 

For Ubuntu and all distros which supports .deb or .rpm packages, just type `make installer`. After compilation, the .deb or .rpm packages are available in the subdirectory named `installers`.

Please understand that TSDuck is a non-funded open-source project which is maintained by one single person. In practice, it is impossible to produce binaries for all versions of all distros. This is why rebuilding the binaries and the packages has been made as simple as possible.

picatchoof, 2021-07-31:
Hello @lelegard 
Hello all.
Appreciate your involvement.
And the way you react to almost all queries.
I'm running the 3.26-2349 it ask for upgrade but needs some dependencies when applied.
I will just wait till finished the project and do installation from source to implement all the needed packages.
Thanks a lot. 
Good luck 🤞

beamerkun, 2022-01-12:
@lelegard I understand that one-man OSS project is a lot to handle, but I can't help myself but ask:
- Why not stick to at least latest Ubuntu LTS version?
- Why official release is build for Ubuntu 21, but nightlies are done for 20.04 ?

I can help with reconfiguring CI / build systems, if that's needed ;-)

lelegard, 2022-01-12:
The production of TSDuck is fully driven by the available resources.

A release is something I manually trigger every 3 or 4 months using my own infrastructure, namely a Mac and a Raspberry Pi 😉, nothing more. I use a set of VM's on my Mac with the most usual Linux distros and Windows. Those VM's always run the latest version of their OS. The production script boots each VM, builds TSDuck, collects the packages and shuts down the VM. The trigger is manual, the process is automated.

Nightly builds and web site updates being ... well ... nightly, I do not want to be involved. All operations are automatically performed using the GitHub Actions CI/CD. GitHub Actions only provides Ubuntu 20 (and Windows and macOS). There is no other Linux distro and no more recent version. That is the only reason.

This is explained in the last slides of [this presentation](https://tsduck.io/download/docs/tsduck-project.pdf).

beamerkun, 2022-01-12:
@lelegard thanks a lot for explanation, makes perfect sense now. You're doing a lot of great work here, so even more thanks for prompt response

lars18th, 2022-01-18:
FYI @lelegard ,

> I use a set of VM's on my Mac with the most usual Linux distros and Windows.

Perhaps you want to migrate your current _N_ Linux VM's to just one with Distrobox: https://distrobox.privatedns.org/

lelegard, 2022-01-18:
> Perhaps you want to migrate your current N Linux VM's to just one with Distrobox:

Thanks for the link. I am afraid that, unlike a VM, a container is never exactly identical to the corresponding distro. For a start, all containers use the same kernel. There are always subtle differences in package management and versioning which are important when trying to reproduce a specific issue from a user. The idea is to have bleeding edge distros (Fedora, Ubuntu) and old distros with old packages (Redhat / CentOS / Alma), nicely packaged distros (Ubuntu, Fedora) and awful old-fashioned rolling distros (Gentoo, Arch). This is not quite the same. Unfortunately.