tsduck/tsduck #829: Send to ip output multiple ports at once [Q]

antoineagscom, 2021-07-21:
Hello,

I would like to send a ts file in only one command on all the tsp listener (here port 1200 & 1201) 

Listeners :
1) tsp -v -I ip 1200 -l 127.0.0.1 -O drop
2)  tsp -v -I ip 1201 -l 127.0.0.1 -O drop

Sender :
tsp-v -I file file.ts -P regulate -O ip 127.0.0.1:1200 127.0.0.1:1201

I cannot add a range of port 1200 - 120X ; Have you any idea please ?

antoineagscom, 2021-07-21:
I use a fork plugin for this, I share information for all 

tsp-v -I file file.ts -P regulate -P fork "tsp -O ip 127.0.0.1:1200" -P fork "tsp -O ip 127.0.0.1:1201" -O drop

I can close the issue, thanks for these great tool !

closed 2021-07-21 by antoineagscom

lelegard, 2021-07-31:
Or, more simply (to avoid the useless creation of one process):

tsp-v -I file file.ts -P regulate -P fork "tsp -O ip 127.0.0.1:1200" -O ip 127.0.0.1:1201