tsduck/tsduck #830: [BUG] PES not counting the correct amount of PES and HEVC SubLayer parsing problem

thisyilife, 2021-07-23:
Hi,

I came across 2 bugs when using and I wanted your opinion on them. Unfortunately, I used personal files and cannot share it publicly.

#### Bug description:
I tried to use PES plugin to get the number of PES in a stream but it seems that i get less than the real number of PES present in the TS file. 

#### How to reproduce:
The command that I used to count the number of PES is the following one :
tsp -I file « filename » -P pes -t -O drop | wc -l

I compared the number of PES obtained with TSDuck with the number of PES obtained by counting the number of present DTS/PTS in PES. I made sure to use files with defined DTS/PTS and did not count it twice when both were present. For some files, the amount of PES counted by tsDuck was lesser than the amount of present DTS/PTS. 

#### Expected behavior:
Same number.

#### Environment:
 - OS: 
 - OS version: 
 - TSDuck full version: 
 - Installation type: 

#### Additional information:
I’ve seen that this problem occurs when the last packet from a PES has only 1 or 2 bytes of payload and it could be a problem with the test in **PESDemux::processPacket** function, line 331 :

```
if( len != 0 && pc.ts→size() >= 4 + len)
```

Replacing the number ‘4’ by ‘6’  correctly count the number of PES present in a TS file. 
Unfortunately, I tested with personal file and cannot share it publicly, but as I mentioned, the count problem appears whenever the last packet from a PES has only 1 or 2 bytes of payload.

-----------------------------------------------------------------------------------------------------------------------------------------------------

I don't know if I need to create another thread but if not here is the description of the second bug :

#### Bug description:
Another bug that I encountered is that I cannot access value ( i.e bit_rate and cpb_size) in SubLayerHRDParameters from a PES even if it exists.
I checked that they were correctly defined with DVB inspector first.

As a reminder the SubLayerHRDParameters is present in NALUnit SPS_NUT → seq_parameter_set_rbsp→ vui_parameters → hrd_parameters.

#### How to reproduce:
Use a TS file that transport HEVC video and type the following command :
tsp -I file “file” -P pes -i --avc-access-unit -O drop

(I used the -i option to compress the output, but even without -i, i could not found the SubLayerHRDParameters information)
The members from SubLayerHRDParameters does not appears even if they exist. I used a single layered HEVC stream when encountering this second bug.

#### Expected behavior:
Printing of members from SubLayerHRDParameters. 

#### Environment:
 - OS: 
 - OS version: 
 - TSDuck full version: 
 - Installation type: 

#### Additional information:
I checked with a debugger and I’ve found that the issue happens at line 122, file **HEVCHRDParam.cpp**. 

```
If (valid && sl.low_delay_hrd_flag == 0)
```

The flag low_delay_hrd_flag wasn’t present in the files I used and would not pass this condition even if they didn’t exists. Following the ITU-T Rec H.265 (10/2014), they can indeed not be defined when fixed_pic_rate_whithin_cvs_flag[i] flag is defined.

When removing the condition (sl.low_delay_hrd_flag == 0), I could correctly parse the SubLayerHRDParameter’s members but it’s not the correct way to fixed it.

I hope I gave enough information to help you.

lelegard, 2021-08-01:
Hi @thisyilife, thanks for the detailed and precise report, this is very much appreciated.

About the first issue, you are right, this is a bug. Replacing 4 by 6 is the right fix. There are indeed 6 and not 4 bytes before the area the size of which is given in the PES_packet_length field (unit start code: 4, length: 2). I have integrated the fix.

Let me investigate the second issue, no need to open another thread.

lelegard, 2021-08-01:
@thisyilife, could you please provide a sample TS which exhibits the second issue?

If the TS is private, there is no need to provide a large amount. One PES packet exhibiting the issue is enough. First, use option `--packet-index` in plugin `pes` to locate the first and last TS packet for each PES packet. After identifying the PES packet you would like to provide, cut the original TS around the first and last TS packet using plugin `skip` and `until`. Please make sure you reproduce the issue on the final TS extract.

Even if you consider the TS as private, just one PES packet does not disclose valid information.

lelegard, 2021-08-01:
OK, I think that I missed this in H.265 section E.3.2:

> __cpb_cnt_minus1__[ i ] plus 1 specifies the number of alternative CPB specifications in the bitstream of the CVS when HighestTid is equal to i. The value of cpb_cnt_minus1[ i ] shall be in the range of 0 to 31, inclusive. When not present, the value of cpb_cnt_minus1[ i ] is inferred to be equal to 0.

The important part is "When not present, the value of cpb_cnt_minus1[ i ] is inferred to be equal to 0", meaning that there is exactly one cpb. I erroneously assumed that in the absence of cpb_cnt_minus1[I] there was no cpb.

I understand the fix but I definitely prefer to have one PES packet to test, as requested in the previous post.

thisyilife, 2021-08-02:
Hi,

Indeed, I haven't thought about sending only one PES. So here are 2 sample files with the PES containing SPS information causing the bug. Both files are single layered HEVC type.

The expected SubLayerHRDParameters value for the file "1s.ts" are :
- **bit_rate_value_minus1[0]** = 0x1C9C2 (117186)
- **cpb_size_value_minus1[0]** = 0x1C9C2 (117186)
- **cbr_flag[0]** = 0x0 (0)  

And for the second file (2s.ts), expected SubLayerHRDParameters value are :
- **bit_rate_value_minus1[0]** = 0x81B2 (33202)
- **bit_rate_value_minus1[0]** = 0x81B2 (33202)
- **cbr_flag[0]** = 0x0 (0) 

I checked both file's value with DVB Inspector and tsp pes plugin. Unfortunately, the SubLayerHRDParameters values still don't appears with the latter.

Thank you for your time and consideration, tsDuck is an amazing toolkit and I'm glad to be helpful.

N.B : Github doesn't support attached .ts file, here is a link to download the 2 sample files : https://filetransfer.io/data-package/u9Jfiyis#link

lelegard, 2021-08-03:
Hi @thisyilife 

The second issue has been fixed in commit b931d4ba3c22f8150258f7db894bc6aa0dd6a98e.

Thanks again for reporting these two bugs and for the extremely precise diagnostic.

closed 2021-08-03 by thisyilife