tsduck/tsduck #833: [Q] Using TSDuck to output EIT event data

mcflyuk, 2021-07-29:
Wasn't sure whether to put this as a question or a feature request, stick with question for now.

I'm doing some investigation into transport streams and event information in the EIT. I can open the stream in an analyser (e.g. DVB Inspector) and then navigate to each event in turn but it would be useful to have some way of searching through all the events in a stream. For instance, search for number of instances of events with a particular genre or title.

After reading through the TSDuck manual, there's the "Analyze EIT sections" page but this produces an overview report rather than outputting the events themselves (_tsp -P eit [options]_).

Then there's the eitinject section, which will create EIT tables or sections from binary, XML or JSON format. Effectively what I'm trying to do is the inverse of this - export the EIT tables from a transport stream out to XML or JSON format that I could then search through. 

Didn't find any other possible method in the documentation so thought I'd put the question forward to see whether there was a solution with TSDuck?

lelegard, 2021-07-31:
If you do not need to reinject the EIT, there is no need for XML or JSON output. The default text output is sufficient. In that case, use the option `-a` or `--all-sections` in plugin `tables` or command `tstables`. All EIT sections are output and you can search through the output text.

If you need XML (or JSON), this a bit complicated since an XML structure can only represent a complete table, _all_ sections included. But an EIT is a bit different. This is not a true table in MPEG terms. While properly constructed EIT p/f are always complete tables with 2 sections, the vast majority of EIT schedule are never complete tables. There are missing sections. This is why you never see them in an XML output. The `tables` plugin waits for all sections of a given table to be received before outputting the complete XML structure. In the case of EIT schedule, some sections do not exist, they are never received, the table is never complete and the XML structure is never output.

If you really need an XML structure for all EIT sections, you may use the option `--pack-all-sections` (please RTFM) to force the production of one XML structure per section.

mcflyuk, 2021-08-05:
That's really helpful, thanks. Hadn't tried the tstables command before but can see it's very useful for searching through the stream, should be enough for me.

closed 2021-08-05 by mcflyuk