tsduck/tsduck #835: [Q] Qt creator develop new plugins workflow

nimblin, 2021-07-31:
I'm trying to develop a packet processor plugin and wanted to try out Qt creator on as you suggest but I'm having some issues with it and was hoping you could help.

So, I made a new source `src/tsplugins/tsplugin_myplugin.cpp` manually (a slightly modified [tsplugin_sample](https://github.com/tsduck/tsduck/blob/master/sample/sample-plugin/tsplugin_sample.cpp)). At this point I can run make and TSDuck compiles fine, all tests pass, and my plugin works fine - great!

I then opened the `build/qtcreator/tsduck.pro` on Qt Creator, and manually created a new folder tsplugin_myplugin in `build/qtcreator/` with a file `tsplugin_myplugin.pro` inside as every other plugin:

```
CONFIG += tsplugin
TARGET = tsplugin_myplugin
include(../tsduck.pri)

``` 

Qt creator then automatically includes the `.pri` file and finds my `tsplugin_myplugin.cpp` source, at this point everything looks the same as every other plugin.

**The problem is,** QtCreator reports errors everywhere even-though it has no problem finding the header files:
[image]

Whats the issue here? How should I use QtCreator for development? 

Thanks in advance.

lelegard, 2021-07-31:
Hi @nimblin,

I see that your error is on the `bat` plugin. Do you have the same problem with all existing plugins? I mean, is this a general QT Creator problem with all plugins or is it a specific problem for your plugin?

I guess that this is a generic problem for all plugins because, as you mentioned, creating the project file is trivial.

It is hard to diagnose. It depends on which distro you use, the origin of the QT Creator package (from qt.io or a distro package). There are many configuration issues with QT Creator, depending on the distro. I use QT Creator to develop TSDuck on macOS and Linux. I have installed it on more than ten Linux VM's, using various distros. On some cases, usually because of missing or incompatible package dependencies, I ran into various problems. They were all fixed after googling a while.

If this is specific to your plugin, try to run the script `build/build-qtcreator-project-files.sh`. It makes sure that all QT Creator project files are correct. If you are interested in building on Windows, the script `build/build-msvc-project-files.sh` does the same thing for Visual Studio project files.

nimblin, 2021-07-31:
Hello @lelegard - I should have been more clear, the errors you see depicted in the screenshot are project wide, I just used the `bat` plugin as an example.

Interestingly enough, a the freshly opened project on Qt Creator builds fine, it is only once you open a file on the editor that you I see errors for that specific file when building the complete project. Closing all opened files in editor and rebuilding the project even after it failed also works. Very odd! 

I'm running Ubuntu 20.04.2 LTS and it is a relatively fresh install, I got `qtcreator` through `apt`, the version is Qt Creator 4.11.0

I see from that `build/build-qtcreator-project-files.sh` script that my manual process of creating the sub-project structure inside QT Creator was appropriate, that's good, thanks for the reference.

I'll leave this issue open while I look for a solution and update on my findings.

nimblin, 2021-07-31:
All issues appear to be gone after disabling the code model: `help>About Plugins>Clang Code Model`. [Source](https://forum.qt.io/topic/96755/qt-creator-not-correctly-parsing-includes/5).

closed 2021-07-31 by nimblin

lelegard, 2021-07-31:
I have run into this particular issue once. I fixed it somehow (without disabling the code mode), but do not remember how. Make sure that clang and other qt dev packages are installed. You may find more info in google, maybe.