tsduck/tsduck #837: [Q] Timeout extend

picatchoof, 2021-08-04:
Dear all,
is it possible to extend the timeout that handle the Signal Lock.
I receive this error BUT the issue is that I'm sur that the TP carry a signal. (Can be seen by satellite Box).
Also is it possible to do scan and monitor the 
- SNR
- Eb/No
- BER
Best regards.

mylinux@SBlab:~$  tsp -vvv -I dvb --freq 11078000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --modulation 8-PSK -O file B06-.ts
* dvb: using /dev/dvb/adapter0 (DVB-S, DVB-S2)
* dvb: tuned to transponder --delivery-system DVB-S2 --frequency 11,078,000,000 --symbol-rate 5,000,000 --fec-inner 9/10 --polarity vertical --modulation 8-PSK --pilots off --roll-off 0.35 --lnb Extended --satellite-number 0
*** Error: dvb: no input signal lock after 5000 milliseconds**

BTW this TP is on the E7WA satellite.

Best regards

picatchoof, 2021-08-04:
Hello all.
BTW I succeed to bypass the previous error by disabling the signal Timeout using the parameter **--signal-timeout 0** , but I stil can not get the Data from the TP

the situation is that this TP carry MPE streams only, this at this moment the only reason for me that I can tell to justify why NO SIGNAL LOCKED is not Possible

mylinux@SBlab:~$ tsp -vvv -I dvb --freq 11078000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --pilots on --modulation 8-PSK --signal-timeout 0 -P mpe -p 3002 -o snrt.ts -O file tststs.ts
* dvb: using /dev/dvb/adapter0 (DVB-S, DVB-S2)
* dvb: tuned to transponder --delivery-system DVB-S2 --frequency 11,078,000,000 --symbol-rate 5,000,000 --fec-inner 9/10 --polarity vertical --modulation 8-PSK --pilots on --roll-off 0.35 --lnb Extended --satellite-number 0
* dvb: signal locked: no, strength: 32%

lelegard, 2021-08-05:
Hi @picatchoof 

The first hypothesis for "signal locked: no, strength: 32%" would be a Linux driver issue. Can you successfully receive another TS at another frequency on this sat using `tsp` on the same machine? If yes, the problem is elsewhere.

The second hypothesis is that your modulation parameters are slightly incorrect. Are you 100% sure that all parameters are correct and where do you get them from? Looking at the [Lyngsat page for E7WA](https://www.lyngsat.com/Eutelsat-7-West-A.html), there is no reference to 11.078 GHz. This does not mean that the frequency does not exist, especially if it carries MPE only (Lyngsat is a community effort, not an official source). But it looks suspect.

picatchoof, 2021-08-05:
Hi @lelegard.
**I'm using TBS6904 DVB-S2 Quad Input PCIe Card under Ubuntu 20.04**
I made some tests on TP that are technically identical to this one on the same satellite.
TP1= 11158V5000
TP2= 11033V5000
TP3= 10992V5000
I can see and record TS from Them All, using TSP & DVBLAST.
for this **TP=11078V5000** I get in touch with Satellite Operator Eutelsat they confirm all the technical issues from their side are normalized and must be OK from the customer side.

The only difference between the Three and this "faulty" is that the 11078/V/5000 carry only MPE content.
maybe if there is raw method to log the TS or MPE parameter that can help to Lock the Signal can be helpful.

Before I used to receive this MPE Content using TSP easily. But it was included in TP 11158V5000 with the Other TVs and Radios.
This web site shows the data that is DSM-CC from the previous configuration of the TP11158v5000.

https://www.digitalbitrate.com/dtv.php?mux=11158&liste=1&live=76&lang=fr  
The SID 200 has two PIDs 2001/2002 that embed the TVs encapsulated as MPE.

[image]

Best Regards

lelegard, 2021-08-05:
> The only difference between the Three and this "faulty" is that the 11078/V/5000 carry only MPE content.
maybe if there is raw method to log the TS or MPE parameter that can help to Lock the Signal can be helpful.

What does "only MPE content" means here? MPE is an encapsulation of IP datagrams inside TS packets in one PID. So, "only MPE" probably means that the TS contains only one service with one single component containing DSM-CC sections.

At transport level, as far as satellite reception is concerned, there is absolutely zero difference between the four TS. If the first three TS are received by `tsp` but not the fourth one, there is at least one modulation parameter which is incorrect. Please do check them all. If you have a contact at Eutelsat, ask them for _all_ details.

picatchoof, 2021-08-06:
Hello @lelegard 
I'm still struggling on this issue.
On the residential Receiver we can notice that there is a signal on this Frequency. but no traditional TV/Radio Channels.
also the Eutelsat Technicians confirm that its pure Data.
[image]

Today Afternoon I will change my Antenna/LNB/Eliminate the  DiSEqC Switch and redo the tests.

Maybe we need to move to an other plateforme.

Also I'm wondering to run my TBS Tuner under Windows to have more details about this issue, because it seems to me that TBS put more tools under Windows for such equipment.

Appreciate your involvement
and I will keep you Informed on the evolution of deeper diagnostics.