tsduck/tsduck #838: [BUG] undefined reference to 'dladdr' while building static on Ubuntu 21.04

jcurropena, 2021-08-05:
#### Bug description:
Trying to buid static version with custom plugins on Ubuntu 21.04. Installed both libsrt1-gnutls and libsrt1-openssl.

#### How to reproduce:
apt install -y wget vim curl apt-utils unzip libpcsclite1 libsrt1.4-gnutls srt-tools cmake sudo g++ gdb git vim cmake libcurl4-gnutls-dev
<copied cpp files for custom plugins on tsduck source directory from .zip files>
make NOPCSC=1 NODTAPI=1 STATIC=1 NOTEST=1 NOTELETEXT=1 static

#### Expected behavior:
- compilation works
- static versions of the ts* binaries are generated

#### Errors and logs:
~~~
  [DEP] dtv/demux/tsTimeTrackerDemux.cpp
  [DEP] dtv/demux/tsTableHandlerInterface.cpp
  [DEP] dtv/demux/tsStandaloneTableDemux.cpp
  [DEP] dtv/demux/tsSignalizationHandlerInterface.cpp
  [DEP] dtv/demux/tsSignalizationDemux.cpp
  [DEP] dtv/demux/tsSectionProviderInterface.cpp
  [DEP] dtv/demux/tsSectionHandlerInterface.cpp
  [DEP] dtv/demux/tsSectionDemux.cpp
  [DEP] dtv/demux/tsPacketizer.cpp
  [DEP] dtv/demux/tsOneShotPacketizer.cpp
  [DEP] dtv/demux/tsCyclingPacketizer.cpp
  [DEP] dtv/demux/tsAbstractPacketizer.cpp
  [DEP] dtv/demux/tsAbstractDemux.cpp
  [DEP] dtv/cas/tsTSScrambling.cpp
  [DEP] dtv/cas/tsPIDOperator.cpp
  [DEP] dtv/cas/tsPCSC.cpp
  [DEP] dtv/cas/tsKeyTable.cpp
  [DEP] dtv/cas/tsCASSelectionArgs.cpp
  [DEP] dtv/cas/tsCASMapper.cpp
  [DEP] dtv/cas/tsCASFamily.cpp
  [DEP] dtv/linux/tsTunerDeviceInfo.cpp
  [DEP] dtv/linux/tsTunerDevice.cpp
  [DEP] dtv/linux/tsHiDesDevice.cpp
  [DEP] dtv/linux/tsDTVProperties.cpp
  [DEP] dtv/tables/tsVCT.cpp
  [DEP] dtv/tables/tsUNT.cpp
  [DEP] dtv/tables/tsTVCT.cpp
  [DEP] dtv/tables/tsTSDT.cpp
  [DEP] dtv/tables/tsTOT.cpp
  [DEP] dtv/tables/tsTDT.cpp
  [DEP] dtv/tables/tsSTT.cpp
  [DEP] dtv/tables/tsSpliceInformationTable.cpp
  [DEP] dtv/tables/tsSelectionInformationTable.cpp
  [DEP] dtv/tables/tsSDT.cpp
  [DEP] dtv/tables/tsRST.cpp
  [DEP] dtv/tables/tsRRT.cpp
  [DEP] dtv/tables/tsRNT.cpp
  [DEP] dtv/tables/tsPMT.cpp
  [DEP] dtv/tables/tsPCAT.cpp
  [DEP] dtv/tables/tsPAT.cpp
  [DEP] dtv/tables/tsNIT.cpp
  [DEP] dtv/tables/tsNBIT.cpp
  [DEP] dtv/tables/tsMGT.cpp
  [DEP] dtv/tables/tsLIT.cpp
  [DEP] dtv/tables/tsLDT.cpp
  [DEP] dtv/tables/tsITT.cpp
  [DEP] dtv/tables/tsINT.cpp
  [DEP] dtv/tables/tsETT.cpp
  [DEP] dtv/tables/tsERT.cpp
  [DEP] dtv/tables/tsEIT.cpp
  [DEP] dtv/tables/tsDSMCCStreamDescriptorsTable.cpp
  [DEP] dtv/tables/tsDiscontinuityInformationTable.cpp
  [DEP] dtv/tables/tsDCCT.cpp
  [DEP] dtv/tables/tsDCCSCT.cpp
  [DEP] dtv/tables/tsCVCT.cpp
  [DEP] dtv/tables/tsCIT.cpp
[... 1543 lines cut ...]
  [DEP] tsdump.cpp
  [DEP] tsdektec.cpp
  [DEP] tsdate.cpp
  [DEP] tscmp.cpp
  [DEP] tscharset.cpp
  [DEP] tsbitrate.cpp
  [DEP] tsanalyze.cpp
  [CXX] tsanalyze.cpp
  [LD] /app/tsduck-3.26-2349/bin/release-x86_64-7ea2beb4432b-static/tsanalyze
/usr/bin/ld: /app/tsduck-3.26-2349/bin/release-x86_64-7ea2beb4432b-static/libtsduck.a(tsSysUtils.o): in function `ts::CallerLibraryFile()':
tsSysUtils.cpp:(.text+0x368): undefined reference to `dladdr'
collect2: error: ld returned 1 exit status
make[3]: *** [../../Makefile.common:1073: /app/tsduck-3.26-2349/bin/release-x86_64-7ea2beb4432b-static/tsanalyze] Error 1
make[2]: *** [Makefile:43: default] Error 2
make[1]: *** [Makefile:59: default] Error 2
make: *** [Makefile.tsduck:81: static] Error 2
~~~

#### Environment:
 - OS: Ubuntu (both hosted version and docker version>
 - OS version: Ubuntu 21.04
 - TSDuck full version: tested both on 3.27-2438 and 3.26-2349
 - Installation type:  source file, static compilation with custom plugins added.

#### Additional information:

lelegard, 2021-08-06:
Hi @jcurropena,

The first question is what do you want to do exactly? The word "static" can be used in many ways here:

1. Link `tsp` with all plugins being statically linked inside the `tsp` binary instead of being individual shared objects.
2. Same as 1. plus link against the TSDuck static library `libtsduck.a` instead of `libtsduck.so`.
3. Full standalone static `tsp` executable (same as 2. plus link everything against system static libraries, zero shared object used).

Options 1 and 2 and not directly available or supported. You have to tweak the makefiles yourself.

By using `make static`, you force option 3.

Full static builds were sometimes required in the past to avoid dependencies problems. On Windows, some people still want static builds to avoid "DLL hell" issues. On most Unix systems, however, static builds are now at best useless and at worst impossible because many basic system features now rely on shared objects. On macOS, for instance, full static builds are no longer supported at all.

On Linux, full static builds are still technically possible but create many constraints. For instance, it is no longer possible to resolve host names from a statically built binary. The host name resolution is fully based on "shared object providers" for `/etc/hosts`, DNS, etc. So, with a statically built TSDuck on Linux, you cannot use host names, you must use IP addresses only. Even "localhost" is not recognized. This also means that the source code itself must be highly customized to avoid system functions which prevent static builds such as `gethostbyname()` or `dl...()`. Also note that HTTP and SRT won't be usable either, as well as most network libraries which resolve host names (unless you get specially crafted versions of these libraries with limited features for static builds).

So, a fully static build on Linux is a really, really, bad idea. On a very small constrained device with a custom kernel, this may make sense. But on a Ubuntu distro, even on a small device, it does not make sense at all. Literally everything on a Ubuntu distro depend on shared objects. So, trying to get rid of them is useless.

Now, you may ask "so why providing `make static`"? This is an exercise I went through years ago when some users wanted static builds on Windows. I extended the exercise to Linux and macOS and went through all the above mentioned issues. It worked, more or less. The `make static` option is still there for reference but is no longer maintained.

New code which was added afterwards does not take into account the required conditional compilation for static build. This is typically the source of your error. I can try to fix those recent dependencies and allow full static builds again. But this is a really bad idea on Ubuntu.

> `make NOPCSC=1 NODTAPI=1 STATIC=1 NOTEST=1 NOTELETEXT=1 static`

Just use `make static`. All other definitions `xxx=1` are implicitly forced with a static build. As mentioned above, CURL and SRT are also disabled.

> Also tested using CFLAGS=--std=c++11 in make.

This is already done in TSDuck makefile. TSDuck requires C++11.

lelegard, 2021-08-06:
I have pushed a simple fix to allow again `make static` on Linux. But you know the limitations: no host names, no SRT, no HTTP, etc.

jcurropena, 2021-08-09:
Hi @lelegard ,

As always, thanks for replying in such a cooperative and informative way :)

Let's go over the (devil) details!

> Hi @jcurropena,
>
> The first question is what do you want to do exactly? The word "static" can be used in many ways here:
>
>     1. Link `tsp` with all plugins being statically linked inside the `tsp` [...]

Indeed, apologies for the bad wording. I was certainly meaning option 3, full standalone `tsp` and tool binaries with no dependencies to shared objects.

> Full static builds were sometimes required in the past to avoid dependencies problems. On Windows, some people still want static builds to avoid "DLL hell" issues. On most Unix systems, however, [...]

I'm fuly aware of the disadvantages, but in this case we are heavily constrained by 3rd parties and we have, most of the times, no other choice. 

As you might know, most of the video headends have tight constraints about internet connection; too tight, I could even say. So it's not the first time we need to perform analysis with tsduck + our plugins (drift calculation, etr290 monitoring, mostly) where we can't even install the minimal base packages, not to say tsduck, etc. 

Docker would be the easiest option, but the constraints apply there, too.

So, in this point, the only choice we end up facing is to build up a full standalone version, copy it over to the machine, and run it there. 

As you can imagine, it's a pain in the ass, you have mentioned some of the drawbacks above. But somehow we avoid the issues, since our analysis is focused on specific parts of the headend flow, so some of the drawbacks do not apply, and we can avoid others.

> Now, you may ask "so why providing `make static`"? This is an exercise I went through years ago when some users wanted static builds on Windows. I extended the exercise to Linux and macOS and went [...]

Good one, didn't know.

> 
> > Also tested using CFLAGS=--std=c++11 in make.
> 
> This is already done in TSDuck makefile. TSDuck requires C++11.

Just in case, I could see in the Makefile that c++11 should be required anyway.

lelegard, 2021-08-09:
OK, as long as you are aware of the constraints, that is fine. It is interesting to note that at least one person needs full static linking on Linux, despite the constraints. This means that this build option should be maintained in the future.

As mentioned in a previous post, this has been fixed in the latest repo. Do you confirm?

jcurropena, 2021-08-09:
> OK, as long as you are aware of the constraints, that is fine. It is interesting to note that at least one person needs full static linking on Linux, despite the constraints. This means that this [...]

Compiling right now, I'll confirm asap.

jcurropena, 2021-08-09:
Hi @lelegard, I can confirm it works again. Closing the issue.

Thanks

closed 2021-08-09 by jcurropena

lelegard, 2021-08-09:
> I can confirm it works again. 

Good news, thanks.

> tsduck + our plugins (drift calculation, etr290 monitoring, mostly) 

Proprietary and closed-source plugins are fully legitimate. This is allowed on purpose by the license. I have developed a few of them myself in CAS and security contexts. This principle goes even beyond plugins using the "TSDuck extension mechanism" (using shared libraries though...)

I do not know exactly what your plugins are doing on drift calculation or etr290 monitoring, but this sounds like generic useful features for everyone. TSDuck represents a huge amount of work across years that you can use for free. It could be fair to contribute back with generic useful features, not using secret protocols of course. Several plugins were already provided as contributions from users: `srt`, `bitrate_monitor`, complex modes in `encap` / `decap`, etc. This is the spirit of open source.

You may also consider providing some public feedback (see #729), not even mentioning [donating](https://tsduck.io/donate/).

No obligation, just a suggestion, your choice.

jcurropena, 2021-08-09:
> Good news, thanks.
>
>
> Proprietary and closed-source plugins are fully legitimate. This is allowed on purpose by the license. I have developed a few of them myself in CAS and security contexts. This [...]

I fully agree with you, and a request was put on management and legal team (this is a multinational company) to allow our team to opensource them (or the most part of, anyway). As you say, it's nothing else but mostly common  functionality, and there's no IP or trademarks involved.

So now it's beyond my acting or decision, will see.