tsduck/tsduck #839: [Q] AbstractTable deserializePayload() implementation

nimblin, 2021-08-06:
I am trying to understand how to extend tsduck with support for custom tables. I am looking at the [AIT class](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tables/tsAIT.h)  (ETSI TS 102 809)  for reference as I am most familiar with that table.

Consider the *deserializePayload()* implementation:
``` c++
void ts::AIT::deserializePayload(PSIBuffer& buf, const Section& section)
{

    // buf.currentReadBitOffset()   YIELDS 0

    // Get common properties (should be identical in all sections)
    const uint16_t tid_ext = section.tableIdExtension();
    test_application_flag = (tid_ext & 0x8000) != 0;
    application_type = tid_ext & 0x7fff;

    // buf.currentReadBitOffset() YIELDS 0

    // Get common descriptor list                                 [X]
    buf.getDescriptorListWithLength(descs); 

    // Application loop length.
    buf.skipBits(4);
    const size_t loop_length = buf.getBits<size_t>(12);
    const size_t end_loop = buf.currentReadByteOffset() + loop_length;

    // Get application descriptions.
    while (buf.canRead()) {
        const uint32_t org_id = buf.getUInt32();
        const uint16_t app_id = buf.getUInt16();
        Application& app(applications[ApplicationIdentifier(org_id, app_id)]);
        app.control_code = buf.getUInt8();
        buf.getDescriptorListWithLength(app.descs); 
    }

    // Make sure we exactly reached the end of transport stream loop.
    if (!buf.error() && buf.currentReadByteOffset() != end_loop) {
        buf.setUserError();
    }
}

```

And here is the AIT Section structure for reference ([TS 102 809](https://www.etsi.org/deliver/etsi_ts/102800_102899/102809/01.03.01_60/ts_102809v010301p.pdf) 5.3.4.6 Table 16):
```
                                                    No. of Bits
application_information_section() {               
    table_id ........................................ 8
    section_syntax_indicator ........................ 1
    reserved_future_use ............................. 1
    reserved ........................................ 2
    section_length .................................. 12
    test_application_flag ........................... 1
    application_type ................................ 15
    reserved ........................................ 2
    version_number .................................. 5
    current_next_indicator .......................... 1
    section_number .................................. 8
    last_section_number ............................. 8
    reserved_future_use ............................. 4
    common_descriptors_length ....................... 12  <------------------   [Z] 
    for(i=0;i<N;i++) {   
        descriptor()   
    }   
    reserved_future_use ............................. 4
    application_loop_length ......................... 12
    for(i=0;i<N;i++) { 
        application_identifier {   
            organisation_id ......................... 32
            application_id .......................... 16
        } 
        application_control_code .................... 8
        reserved_future_use ......................... 4
        application_descriptors_loop_length ......... 12
        for(j=0;j<N;j++) {   
            descriptor() 
        }   
    }   
    CRC_32 .......................................... 32
}   

```
 ---

My problem is specifically with the parsing of the beggining of the section: How is *buf.getDescriptorListWithLength(descs)* **[X]** able to parse the section if *buf.currentReadBitOffset()* yields 0 just before it? Shouldn't *getDescriptorListWithLength* expect to be called at buffer offset [Z]?

Everything after **[X]** makes sense.

I would also like to add that, while `-P tables` parses the sections perfectly (I am certain about this as I am generating tables from OpenCaster to test it properly), trying to parse the descriptors programmatically from a plugin as follows does not work:

```c++
    const DescriptorList& dlist = ait.descs;

    for (size_t i = dlist.search(DID_AIT_APPLICATION); i < dlist.count(); i = dlist.search(DID_AIT_APPLICATION, i + 1)) {

        // NEVER enters the loop, even though I know for sure there is a DID_AIT_APPLICATION descriptor

        const ApplicationDescriptor app_des(duck, *dlist[i]);
        if (app_des.isValid()) {
            tsp->verbose(u"app priority: %d", {app_des.application_priority});
        }
    }
```

Am I doing something wrong here or is this possibly proof of a misimplementation of *deserializePayload()*?

Thanks

lelegard, 2021-08-06:
Hi @nimblin,

> My problem is specifically with the parsing of the beggining of the section: How is buf.getDescriptorListWithLength(descs) [X] able to parse the section if buf.currentReadBitOffset() yields 0 just before it? Shouldn't getDescriptorListWithLength expect to be called at buffer offset [Z]?

See the [documentation of `ts::PSIBuffer::getDescriptorListWithLength`](https://tsduck.io/doxy/classts_1_1_p_s_i_buffer.html#a65efcec1e0dba0915dda8ec438fd310a), especially this:

> The 2-byte length field uses only its N least significant bits (length_bits) for the length of the descriptor list. If the current read pointer is byte-aligned, 16-N bits are skipped first. If the current read bit pointer is 16-N, the length is directly read after that bit. For all other read pointers, a read error is generated.

Given the large number of sections and descriptors to code, the class `PSIBuffer` has been designed to in a flexible way to be able to write small and safe C++ code, based on the most common use cases.

The two bytes preceding a descriptor list are split in 4 and 12 bits (N=12). The first 4 bits are sometimes used and most of the time unused, as in the AIT.

When the first 4 bits are used in a section type, the deserialization code reads them first using `getBits()`. The current read bit pointer is 4 (16-N) and `getDescriptorListWithLength` reads the length field from the N=12 following bits.

When the first 4 bits are unused, the current read bit pointer is 0 (byte-aligned) and `getDescriptorListWithLength` skips the first 4 (16-N) bits before reading the length field from the N=12 following bits.

If the current read bit pointer is neither 0 nor 4 (16-N), this is considered as a coding or input error and a read error is generated.

nimblin, 2021-08-06:
Hello @lelegard - thanks for your quick and detailed response, as usual. While your response provides some helpful insight into the *why* of length handling in `getDescriptorListWithLength`, I think I already understood the *how* beforehand. 

I will try to formulate my question in a different, simpler, manner.

Recall that, In the AIT section, the common descriptors loop begins 10 bytes into the table - this is the offset of **[Z]** in my first comment. In that case:

#### Why do this
``` C++
void ts::AIT::deserializePayload(PSIBuffer& buf, const Section& section)
{
    const uint16_t tid_ext = section.tableIdExtension();

    [...]

    buf.getDescriptorListWithLength(descs); 

    [..]
```

#### Instead of this
``` C++
void ts::AIT::deserializePayload(PSIBuffer& buf, const Section& section)
{
    const uint16_t tid_ext = section.tableIdExtension();

    [...]

    buf.skipBits(64); // bit offset of reserved_future_use (two bytes before the common descriptor loop)
    buf.getDescriptorListWithLength(descs); 

    [..]
```

How does `getDescriptorListWithLength` magically land itself at the correct bit offset if `buf.currentReadBitOffset` yields 0 right before calling `getDescriptorListWithLength`.

lelegard, 2021-08-06:
> Recall that, In the AIT section, the common descriptors loop begins 10 bytes into the table - this is the offset of [Z] in my first comment. In that case:

I see, I did not get that was your issue.

The reason is simple: the abstract method `deserializePayload()` ([documented here](https://tsduck.io/doxy/classts_1_1_abstract_table.html#ac2e7fd063107108a0388b15f985ef591)) is responsible for deserializing the _payload_ of one binary section (hence, header excluded). The common header and CRC are already processed and validated at the superclass level. The `PSIBuffer` object which is received by `deserializePayload()` is mapped over the section payload only.

For a "long section" (section_syntax_indicator == 1), the payload starts after the common last_section_number field and stops before the final CRC_32 field. In other words, the first 8 bytes and last 4 bytes of the section are not part of its payload. For "short sections" (section_syntax_indicator == 0), the header is 3 bytes long and there is no CRC_32, the payload starts at offset 3.

Again, the idea was to simplify the implementation of each individual type of table, putting every common processing in the common superclasses `AbstractTable` and `AbstractLongTable`. When a `BinaryTable` object (a set of binary sections) is deserialized, the common processing in the superclasses is:

- Validate all sections, make sure they are all here, of the same type and all CRC32 are correct.
- Cleanup the final C++ object (class `PAT`, `PMT`, `AIT`, etc). The subclass-specific part is cleaned using the abstract method `clearContent()` that you must implement.
- The common fields (table id, version, is_current) are set up.
- Finally, the abstract method `deserializePayload()` is invoked on the payload (I repeat, _payload_ only) of each section, one by one.
 
Your `deserializePayload()` is then responsible to deserialize one section payload and _add_ the extracted information in the C++ object. I mean _add_ and not _replace_ since the method is repeatedly called for each section in the table. If your table type stores useful information in the 16-bit "table id extension", you can get it using `section.tableIdExtension()`. The table id extension shall be the same in all sections of a table (MPEG rule).

nimblin, 2021-08-06:
Superb explanation @lelegard , everything is clear now. Much appreciated.

closed 2021-08-06 by nimblin