tsduck/tsduck #840: [BUG] Inconsistency between command line and config options when scanning ISDB-S

APTX, 2021-08-08:
#### Bug description:
I recently tried to scan ISDB-S BS again and I got the `FE_SET_TONE` error. I reread this issue #582 and found the `--lnb` option, but the error would remain:
```
tsscan --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels.xml
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
```

There was also mention of putting the options in `.tsduck` so I tried it again with this in the config file:
```
default.region = japan
default.lnb = japan
```
This got rid of the error and allowed scanning to continue. Is the way I passed the argument on the command line wrong?

#### How to reproduce:
Try to scan ISDB-S with no config file:
```
tsscan --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels.xml
```

#### Expected behavior:
Scanning would start normally, ideally services would be found:
```
* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16400, 0x4010
[... more frequencies ...]
* Frequency: 12,149,440,000 Hz R
tsscan: No PAT found, services are unknown

Name       Provider  ServId TSId   ONetId Type PMTPID
---------- --------- ------ ------ ------ ---- ------
BS朝日1      ビーエス朝日    0x0097 0x4010 0x0004 0x01 0x0101
BS朝日2      ビーエス朝日    0x0098 0x4010 0x0004 0x01 0x0201
BS朝日3      ビーエス朝日    0x0099 0x4010 0x0004 0x01 0x0203
[... more services ...]
```

#### Errors and logs:
Debug logs printed by `tsscan`:
~~~
tsscan: Debug: debug level set to 1
tsscan: Debug: ====> tsscan --debug --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels_ISDB-S_freq_0.xml
tsscan: Debug: loading XML file /usr/share/tsduck/tsduck.lnbs.xml
tsscan: Debug: loading XML file /usr/share/tsduck/tsduck.lnbs.model.xml
tsscan: Debug: dvb0.frontend0/device/idVendor = "" (failure)
tsscan: Debug: dvb0.frontend0/device/idProduct = "" (failure)
tsscan: Debug: dvb0.frontend0/device/manufacturer = "" (failure)
tsscan: Debug: dvb0.frontend0/device/product = "" (failure)
tsscan: Debug: dvb0.frontend0/device/version = "" (failure)
tsscan: Debug: dvb0.frontend0/device/serial = "" (failure)
tsscan: Debug: loading XML file tsduck_channels_ISDB-S_freq_0.xml
tsscan: Debug: loading XML file /usr/share/tsduck/tsduck.channels.model.xml
tsscan: Debug: adding standards ISDB to Japan
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 6 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 9 (0x00000009)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 1049480 (0x00100388)
tsscan: Debug: [2] cmd = 8 (DTV_SYMBOL_RATE), data = 28860000 (0x01B85E60)
tsscan: Debug: [3] cmd = 9 (DTV_INNER_FEC), data = 9 (0x00000009)
tsscan: Debug: [4] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [5] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: got 10000 packets
tsscan: Debug: adding standards DVB to ISDB, Japan
tsscan: Debug: got table id 0x42 on PID 0x0011
tsscan: Debug: adding standards MPEG to DVB, ISDB, Japan
tsscan: Debug: got table id 0x00 on PID 0x0000
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x40 on PID 0x0010
tsscan: Debug: * tuning to --delivery-system ISDB-S --frequency 11,727,480,000 --symbol-rate 28,860,000 --fec-inner auto --polarity right --stream-id 16400
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
tsscan: Debug: * tuning to --delivery-system ISDB-S --frequency 11,727,480,000 --symbol-rate 28,860,000 --fec-inner auto --polarity right --stream-id 16401
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
[... logs repeat for other frequencies ...]
~~~

With the config file tuning to a frequency looks like this:
~~~
  Transport stream id: 16400, 0x4010
tsscan: Debug: * tuning to --delivery-system ISDB-S --frequency 11,727,480,000 --symbol-rate 28,860,000 --fec-inner auto --polarity right --stream-id 16401
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 7 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 9 (0x00000009)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 1049480 (0x00100388)
tsscan: Debug: [2] cmd = 8 (DTV_SYMBOL_RATE), data = 28860000 (0x01B85E60)
tsscan: Debug: [3] cmd = 9 (DTV_INNER_FEC), data = 9 (0x00000009)
tsscan: Debug: [4] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [5] cmd = 42 (DTV_STREAM_ID), data = 16401 (0x00004011)
tsscan: Debug: [6] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
* Frequency: 11,727,480,000 Hz R
tsscan: Debug: got 10000 packets
[...]
~~~

#### Environment:
 - OS: Linux
 - OS version: Debian bullseye on arm64
 - TSDuck full version: 3.28-2457
 - Installation type: Built from git

lelegard, 2021-08-08:
Hi @APTX 

It is quite hard for me to characterise this issue. I cannot reproduce it, having no ISDB-S tuner and no ISDB-S satellite above my head in western Europe.

Using the tuner emulator (user's guide, section 7.1.4), I can reproduce the logical part of the ISDB-S scanning using [the only ISDB-S file in the stream repository](https://tsduck.io/streams/?name=japan-nippontv), a capture of BS-13 if I interpret the frequency correctly. But I cannot analyse the behaviour of a physical ISDB-S tuner.

I have found another issue, but nothing about the LNB not being set from the command line. In fact, because of the issue I found, `tsscan` using the NIT-scan method on an ISDB-S network should have never worked correctly.

The problem is that the command `tsscan` did not have the options `--isdb` and `--atsc`. On the other hand, a synthetic option such as `--japan` is a shortcut for many options _which are otherwise present in the command_. This means that `--japan` includes `--isdb` only if `--isdb` is a supported option for the command. A standard such as ATSC is automatically recognised because it defines very specific tables. ISDB, on the other hand, is mostly based on DVB (as far as the signalisation is concerned) with a number of variations. So, without `--isdb` or other synthetic options which include it such as `--japan` or `--brazil`, most ISDB streams will be initially interpreted as DVB. This was the problem with `tsscan`.

But, I haven't seen any symptom of an LNB not being recognised from the command line.

I have fixed the `--isdb` option issue. Now, `tsscan --japan` correctly assumes that the stream is ISDB from the beginning. I can now perform a logical ISDB-S scan using the tuner emulator and the BS-13 capture while it failed before the fix.

I have also added a few debug traces on LNB.

Could you try with the latest state of the repo? Sorry for not being more assertive, but you described an issue with a physical ISDB-S tuner on Linux, a configuration I cannot reproduce. You may also check if the issue #841 is still present.

APTX, 2021-08-09:
I checked `3.28-2458` and I think there's no difference.

The logs suggest that the `--lnb` option is only applied to the first tune, but not the subsequent ones:
~~~
tsscan: Debug: debug level set to 1
tsscan: Debug: ====> tsscan --japan --isdb --lnb japan --debug -a 0 -c BS-1 --nit-scan -g --save-channels tsduck_channels_ISDB-S_freq_0.xml
tsscan: Debug: loading XML file /usr/share/tsduck/tsduck.lnbs.xml
tsscan: Debug: loading XML file /usr/share/tsduck/tsduck.lnbs.model.xml
tsscan: Debug: loaded LNB "Japan" from command line
tsscan: Debug: dvb0.frontend0/device/idVendor = "" (failure)
tsscan: Debug: dvb0.frontend0/device/idProduct = "" (failure)
tsscan: Debug: dvb0.frontend0/device/manufacturer = "" (failure)
tsscan: Debug: dvb0.frontend0/device/product = "" (failure)
tsscan: Debug: dvb0.frontend0/device/version = "" (failure)
tsscan: Debug: dvb0.frontend0/device/serial = "" (failure)
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: using LNB Japan
tsscan: Debug: LNB uses stacked transposition, no dish control required
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 6 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 9 (0x00000009)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 1049480 (0x00100388)
tsscan: Debug: [2] cmd = 8 (DTV_SYMBOL_RATE), data = 28860000 (0x01B85E60)
tsscan: Debug: [3] cmd = 9 (DTV_INNER_FEC), data = 9 (0x00000009)
tsscan: Debug: [4] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [5] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: got 10000 packets
tsscan: Debug: adding standards MPEG to ISDB, Japan
tsscan: Debug: got table id 0x00 on PID 0x0000
tsscan: Debug: adding standards DVB to MPEG, ISDB, Japan
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x42 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x40 on PID 0x0010
tsscan: Debug: * tuning to --delivery-system ISDB-S --frequency 11,727,480,000 --symbol-rate 28,860,000 --fec-inner auto --polarity right --stream-id 16400
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: using LNB Extended
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
tsscan: Debug: * tuning to --delivery-system ISDB-S --frequency 11,727,480,000 --symbol-rate 28,860,000 --fec-inner auto --polarity right --stream-id 16401
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: using LNB Extended
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
[... logs repeat with different frequencies ...]
~~~

First tune:
```
tsscan: Debug: using LNB Japan
```
Subsequent tune:
```
tsscan: Debug: using LNB Extended
```

It seems the `--lnb japan` option is not forwarded to the subsequent tune operations, but if the option is set in the config it is used for all tune operations.

If  `--lnb japan` is not present the first tune fails as well.

The frequency in the channel xml file is still `0`.

The NIT table seems to be read correctly, or at least the frequency is not `0`:
```
$ tsp --japan -I dvb -c BS-1 -P until -s 10 -P tables -t 0x40 -O drop
* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 14, sections: 1, total size: 784 bytes
  - Section 0:
    Network Id: 4 (0x0004)
    Network information:
    - Descriptor 0: Network Name (0x40, 64), 12 bytes
      Name: ".BS Digital"
    - Descriptor 1: unknown (0xFE, 254), 2 bytes
      0000:  02 01                                      ..
    Transport Stream Id: 16400 (0x4010), Original Network Id: 4 (0x0004)
    - Descriptor 0: Service List (0x41, 65), 21 bytes
      Service id: 151 (0x0097), Type: 0x01 (Digital television service)
      Service id: 152 (0x0098), Type: 0x01 (Digital television service)
      Service id: 153 (0x0099), Type: 0x01 (Digital television service)
      Service id: 753 (0x02F1), Type: 0xC0 (Data service)
      Service id: 755 (0x02F3), Type: 0xC0 (Data service)
      Service id: 756 (0x02F4), Type: 0xC0 (Data service)
      Service id: 757 (0x02F5), Type: 0xC0 (Data service)
    - Descriptor 1: Satellite Delivery System (0x43, 67), 11 bytes
      Frequency: 11.72748 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
      Delivery system: DVB-S
      Modulation: auto (0)
      Symbol rate: 28.8600 Msymbol/s
      Inner FEC: 4/5 (8)
[...]
```
NOTE: that unknown descriptor `0xFE` is the _System management descriptor_, ARIB STD-B10, Part 2, Section 6.2.21 and the value is as specified per ARIB TR-B15 Fascicle 2, Part 1, Volume 4, Section 31.1.2.

lelegard, 2021-08-09:
> It seems the --lnb japan option is not forwarded to the subsequent tune operations, but if the option is set in the config it is used for all tune operations.

Thanks, that was the issue. I just fixed it.

> NOTE: that unknown descriptor 0xFE is the System management descriptor, ARIB STD-B10,

Most ISDB descriptors, including this one, have been implemented for a while now and are correctly reported as long as the stream is correctly identified as ISDB (forced using `--isdb` or  `--japan`). In your command `tsp --japan ... -P tables -t 0x40`, the global `--japan` option should be propagated to all plugins, including `tables`. This is another bug. The plugins `tables` and `psi` incorrectly reset their set of standards upon start, thus dropping the global settings. This is now fixed as well.

Note that if your command had been `tsp ... -P tables --japan -t 0x40`, the problem wouldn't have been seen since `--japan` is locally set in the plugin. But your command is correct, the issue is fixed.

Please let us know if this works for you.

APTX, 2021-08-09:
I can confirm that with the latest build (`3.28-2461`) scanning ISDB-S BS works again with the following command and no config:
```
tsscan --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels.xml
```
In addition I checked that the `0xFE` descriptor is showing up correctly.

closed 2021-08-09 by APTX