tsduck/tsduck #841: [BUG] Services scanned on ISDB-S have frequency set to 0

APTX, 2021-08-08:
#### Bug description:
When scanning ISDB-S channels always have `0` as the frequency in the channels xml file:
```xml
  <network id="0x0004" type="ISDB-S">
    <ts id="0x4010" onid="0x0004">
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <service id="0x0097" name="BS朝日1" provider="ビーエス朝日" PMTPID="0x0101" type="0x01" cas="false"/>
      <service id="0x0098" name="BS朝日2" provider="ビーエス朝日" PMTPID="0x0201" type="0x01" cas="false"/>
      <service id="0x0099" name="BS朝日3" provider="ビーエス朝日" PMTPID="0x0203" type="0x01" cas="false"/>
    </ts>
  <!-- more entries -->
  </network>
```
This is the case for all frequencies:
```
$  grep frequency tsduck_channels_ISDB-S_freq_0.xml
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
```

With frequency `0` it's not possible to tune to any of the ISDB-S channels. ISDB-T scanning still works as before.

The frequencies are printed in debug logs when scanning, they just do not appear in the xml file.
#### How to reproduce:
Scan ISDB-S:
```
tsscan --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels_ISDB-S_freq_0.xml
```

#### Expected behavior:
The frequency should not be `0`.

#### Errors and logs:
It's the same scanning operation as in issue #840.

#### Environment:
 - OS: Linux
 - OS version: Debian bullseye on arm64
 - TSDuck full version: 3.28-2457
 - Installation type: Built from git

lelegard, 2021-08-09:
Hi @APT,

I think I have identified and fixed this issue. I cannot reproduce it, having no ISDB-S environment to work on. Please let us know if the latest repo fixes the issue for you.

The issue was due to the way tuning parameters were collected to fill the channel file. Each TS is scanned to get the list of services. After a successful TS scanning, the physical tuner is queried to report its current modulation parameters. These parameters are used instead of the parameters from the NIT (which were originally used to scan the TS). This is useful on DVB-T where most tuning parameters can be omitted or incorrect (FEC, guard interval, etc). The tuning will succeed anyway and many DVB-T tuners are able to report the exact actual parameters. So, the parameters from the tuner are usually more accurate than the parameters from the NIT (or at least identical).

In the case of satellite, the tuner cannot report the satellite frequency, only the intermediate frequency, at best. This is why the frequency was reset to zero. This is fixed. In the general case, when the parameters which were collected from the tuner do not have a valid frequency, the original frequency is used.

APTX, 2021-08-09:
I can confirm that with the latest build (`3.28-2461`) the frequency is no longer `0` in the channels xml file:
```
tsscan --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels.xml
```
```
$ grep frequency tsduck_channels.xml
      <isdbs frequency="11,727,480,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,727,480,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,727,480,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,765,840,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,880,920,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,957,640,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,957,640,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,996,000,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,804,200,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,804,200,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,919,280,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,072,720,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,111,080,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,765,840,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,880,920,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,880,920,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,996,000,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,919,280,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,919,280,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="11,957,640,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,072,720,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,072,720,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,072,720,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,111,080,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <isdbs frequency="12,111,080,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
```

Unfortunately, I encountered an error when tuning to the frequencies:
Tuning a frequency directly:
```
$ tsp --japan -I dvb --lnb japan -a 0 -f "11,727,480,000" -P until -s 10 -P analyze -O drop
* Error: dvb: satellite frequency 11,727,480,000 Hz cannot be transposed using LNB type Japan 110BS/CS
```
Tuning a service:
```
$ tsp --japan -I dvb --lnb japan -a 0 -c "BS朝日1" --tuning-file tsduck_channels.xml -P until -s 10 -P analyze -O drop
* Error: dvb: satellite frequency 11,727,480,000 Hz cannot be transposed using LNB type Japan 110BS/CS
```

In practice this means that the channels/services from the tuning file can not be tuned.

Tuning the frequency "by name" still works:
```
$ tsp --japan -I dvb --lnb japan -a 0 -c "BS-1" -P until -s 10 -P analyze -O drop
```

If `--lnb japan` is not present, then the `FE_SET_TONE` error appears in all cases. Is the switch supposed to be required even when using a tuning file? Shouldn't the file contain that information?

If this tuning issue is unrelated I can create a new issue and close this one.

----
> In the case of satellite, the tuner cannot report the satellite frequency, only the intermediate frequency, at best.

For BS/CS it may be more practical to store the intermediate frequency instead of the real satellite frequency, due to the [special way these frequencies are handled](https://github.com/tsduck/tsduck/issues/582#issuecomment-629636167).

I don't know how DVB tuners handle missing tuning parameters (do they just try combinations until it works?), but with ISDB all the parameters (except frequency, but all of them are known) are in TMCC. I guess the PT3 tuner driver does not report the tuning parameters on Linux, but they don't really need to be provided either. The PT3 SDK doesn't even allow setting arbitrary frequencies, only enum values that represent the fixed frequencies.

lelegard, 2021-08-09:
Hi @APTX,

> I can confirm that with the latest build (3.28-2461) the frequency is no longer 0 in the channels xml file:

Thanks for the feedback, we are progressing.

> `$ tsp --japan -I dvb --lnb japan -a 0 -f "11,727,480,000" -P until -s 10 -P analyze -O drop`
> `* Error: dvb: satellite frequency 11,727,480,000 Hz cannot be transposed using LNB type Japan 110BS/CS`

The message is a bit misleading, sorry. The problem probably comes from the polarity, which is unspecified, not the frequency. In the LNB configuration file, this type of LNB is declared as handling circular polarity only (left or right), not linear polarity (horizontal or vertical). As documented in the `dvb` plugin, the default value for `polarity` is `vertical` (there must be one default value, there is no "good" one, just one...) The message just means: cannot find the intermediate frequency for this satellite frequency _and vertical polarity_.

Please retry with `--polarity right` (since BS-1 has right polarity).

I looked at the NIT in the [ISDB-S BS-13 stream from the streams repository](https://tsduck.io/streams/?name=japan-nippontv). The polarity are all declared as "right". So, they looks correct. Strangely enough, there are only odd BS channels (right polarity), no even BS channels (left polarity).

> `$ grep frequency tsduck_channels.xml`
> `     <isdbs frequency="11,727,480,000" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>`

The polarity should not be "vertical" but "right" since the polarity comes from the NIT. I can only see the NIT from BS-13 but I assume that the polarity is also correctly set in the BS-1 NIT. In fact, it seems that this is the same problem as the frequency being set to zero. The polarity being a dish configuration parameter and not a strict tuning parameter, it is not returned by the tuner. I just applied the same fix as the frequency.

Please retry with the latest repo. Again, I am sorry for all those iterations. There are some significant differences in the Japanese context that I cannot test myself and I must rely on you to debug.

> Tuning the frequency "by name" still works

Using `-c "BS-1"` works because this is an HF band channel name, not a TV channel name:
~~~
  -c name
  --channel-transponder name
      Tune to the transponder containing the specified channel. The channel
      name is not case-sensitive and blanks are ignored. It is either an "HF
      band channel" or a "TV channel".

      An "HF band channel" has the format "band-number" such as "UHF-22"
      (terrestrial) or "BS-12" (Japanese satellite). See also option
      --offset-count.

      A "TV channel" name is searched in a "tuning file" and the corresponding
      tuning information in this file is used. See also option --tuning-file.
~~~
An the HF band configuration file `tsduck.hfbands.xml` contains:
~~~
  <hfband type="BS">
    <region name="japan"/>
    <channels first_channel="1" last_channel="24" base_frequency="11,727,480,000" channel_width="19,180,000"
              even_polarity="left" odd_polarity="right"/>
  </hfband>

  <hfband type="CS">
    <region name="japan"/>
    <channels first_channel="1" last_channel="24" base_frequency="12,271,000,000" channel_width="20,000,000"
              even_polarity="right" odd_polarity="left"/>
  </hfband>
~~~
This means that BS-1 is an HF band name and its polarity is "right" (1 is odd).

> If `--lnb japan` is not present, then the FE_SET_TONE error appears in all cases. Is the switch supposed to be required even when using a tuning file? Shouldn't the file contain that information?

It cannot. The tuning file contains "external" tuning information, characteristics of the signal coming down from the satellite. The description of the LNB characterizes your local reception equipment. You only know what is your equipment while channel files are generic and can be distributed to everyone in the reception area. Different people may receive the same signal from the same satellite with different types of dish. This may sounds silly in a Japanese context where everyone use the same type of dish (in fact I have no idea if this is really the case), but TSDuck must remain generic.

Theoretically, you may have two satellite tuners on your system which are connected to two different types of dish. In that case, the `--lnb` parameter shall be different depending on the tuner device.

I understand the frustration of being obliged to always specify non-default values in a context like yours. This is where the configuration file is useful to setup other defaults (see appendix A.5.2 in the user's guide for a typical Japanese setup).

> If this tuning issue is unrelated I can create a new issue and close this one.

No need to open a new one.

> For BS/CS it may be more practical to store the intermediate frequency instead of the real satellite frequency

Same remark as above. The intermediate frequency is a characteristics of your local reception equipment, not a generic characteristics of the signal which can be used by everyone.

> I don't know how DVB tuners handle missing tuning parameters (do they just try combinations until it works?),

Quite simple: it does not work :) You have to provide exact values for all parameters in DVB-S/S2. With DVB-T, however, it is different, the tuner can retrieve the actual values if the provided values are incorrect.

> with ISDB all the parameters (except frequency, but all of them are known) are in TMCC

[... cut ...]
But note that the polarity is different. Although it is a characteristics of the signal, it impacts the configuration of the dish. So, even with ISDB-S, specifying the wrong polarity will drive you to another satellite frequency and you cannot receive the TMCC either (or there would be some magic I guess).

> The PT3 SDK doesn't even allow setting arbitrary frequencies, only enum values that represent the fixed frequencies.

I guess that the BS and CS frequencies and polarities are hard-coded. This device must be specific to the Japanese market.

APTX, 2021-08-09:
Thank you for all your help @lelegard .

> Please retry with `--polarity right` (since BS-1 has right polarity).

I can confirm that frequency + polarity works. This also works:
```
tsp --japan -I dvb --lnb japan -a 0 -c "BS朝日1" --polarity right --tuning-file tsduck_channels.xml -P until -s 10 -P analyze -O drop
```
I tested this with the older version (`3.28-2461`) before the latest fix.

I ran the scan again with the new version (`3.28-2462`) but the polarity still ends up being vertical in the channel xml file (same as [previous comment](https://github.com/tsduck/tsduck/issues/841#issuecomment-895258354)).

I checked that the value is correct in nit:
```
$ tsp --japan -I dvb --lnb japan -c BS-1 -P until -s 10 -P tables -t 0x40 -O drop | grep -B2 Polarization
      Frequency: 11.72748 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.72748 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.72748 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.76584 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.76584 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.88092 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.88092 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.88092 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.95764 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.95764 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.99600 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.80420 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.99600 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.80420 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.95764 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.14944 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.91928 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.91928 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 11.91928 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.07272 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.07272 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.07272 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.07272 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.11108 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.11108 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
--
      Frequency: 12.11108 GHz
      Orbital position: 110.0 degree, east
      Polarization: circular - right (3)
```

----

> It cannot. The tuning file contains "external" tuning information, characteristics of the signal coming down from the satellite. The description of the LNB characterizes your local reception [...]

Thank you for the explanation. It does make sense.

> I guess that the BS and CS frequencies and polarities are hard-coded. This device must be specific to the Japanese market.

I believe the polarities are fixed. If I recall correctly, the BS and CS channels are transmitted from the satellite in the order they are numbered and they have opposite polarization to not interfere with each other. In the signal available in the apartment socket the channels are separated by polarization:
[image]
I don't know if the polarization is still relevant after the split.

The reason there are only right polarized channels in my scans is that PT3 does not support the higher frequencies used by the left polarized channels. It can only only tune to the frequencies used by the right polarized channels. (And I also have no CS channels because that signal is not available in my building).

I checked and polarization is not one of the parameters of TMCC for ISDB-S, so I assume it has to be fixed.

There are some more advanced parameters in TMCC that I don't think anyone would want to input by hand. This is not strictly signal related, but as mentioned in #582 the Service ID's are in TMCC, but there is also a mapping of which of the 48 individual slots belongs to which service (not only the count, see [comment](https://github.com/tsduck/tsduck/issues/582#issuecomment-646566080) #582). The modulation/error correction scheme is assigned per service. See ARIB STD-B20, Part 1, Section 2.9.

@lelegard If you wish, I can collect sample TS files from all the services I can receive.

lelegard, 2021-08-09:
> I ran the scan again with the new version (3.28-2462) but the polarity still ends up being vertical in the channel xml file 

My fault, sorry. I did the polarity fix on DVB-S/S2 and forgot ISDB-S. I need some holidays. Too bad, I am just coming back from holidays...

Fixed again...

> I don't know if the polarization is still relevant after the split.

No, it isn't. The polarity is just a configuration of the dish to grab the signal, either software-controlled in most dishes (tone, voltage or DiSEqC) or hard-coded in Japanese dishes. After frequency transposition, all signals are equal for the tuner.

The tuner has no polarization parameter. The polarisation parameter is used by the software _before_ tuning, either to compute the intermediate frequency (Japan) or to send a command to the dish (rest of the world).

> I checked and polarization is not one of the parameters of TMCC for ISDB-S, so I assume it has to be fixed.

It sounds logical. To get TMCC, you need to know the frequency on which to listen to, the tuner frequency, ie. the intermediate frequency. And you need to know the polarity to compute the right intermediate frequency. You cannot find an information (the polarity) in a data set (TMCC) that you can read only if you already know the polarity...