tsduck/tsduck #842: [FEAT] RIST Support

enricovittorini, 2021-08-09:
#### Is your feature request related to a problem?
NO

#### Proposed solution:
Would be great so see RIST Support on TSDuck

https://code.videolan.org/rist

lelegard, 2021-08-30:
Hi @enricovittorini 

Supporting RIST on TSDuck would certainly be great. We just need someone who is sufficiently knowledgeable with RIST to either implement it or at least provide technical support and guidelines.

This is what happened with SRT. A user who needed it implemented it. Afterwards, the support for the plugin was merged with the rest of the TSDuck activities.

I have zero knowledge on RIST. I had a brief look at the library. It seems to have a callback-based API. TSDuck better works with synchronous read/write model because it is more generic. It is possible to work with a callback model but before going to that kind of push/pull adapter model, a discussion with some RIST expert would avoid going the wrong way from the beginning.

@enricovittorini, are you fluent with RIST? Anyone else here?

lelegard, 2021-08-30:
Just assessing the feasibility of using RIST, I cannot event compile it on Windows. For reference, I started [a thread in librist repo](https://code.videolan.org/rist/librist/-/issues/122) in case some volunteers for RIST support in TSDuck would be interested.

Build on Linux is fine. But no major distro (Ubuntu, Debian, Fedora, CentOS) provides librist packages. I have built `.rpm` and `.deb` packages but managing dependencies will become an issue for users if TSDuck binaries are made available with a dependency on non-standard packages.

lelegard, 2021-09-11:
Hi @enricovittorini 

I finally managed to build `librist` everywhere and an initial version of a `rist` plugin is now available. It is announced and discussed [here](https://code.videolan.org/rist/librist/-/issues/125). Please note from the announce that the `rist` plugin is not available by default since `librist` is not available on any Linux distro or macOS. You must build it yourself.

enricovittorini, 2021-09-12:
Great news @lelegard ! Thanks. Will try to make a test soon

lelegard, 2021-09-20:
TSDuck version 3.28 is out. RIST support is enabled by default on macOS and Windows. On Linux, no distro has RIST support by default (except Arch via AUR) but you can rebuild TSDuck on Linux with RIST support if you manually installed librist beforehand.

tipexfm, 2021-10-03:
This is great @lelegard, I'm going to rebuild TSDuck so it can handle RIST.
Many thanks for all the work !!

closed 2025-02-27 by lelegard