tsduck/tsduck #843: [Q] control-port option on windows

enricovittorini, 2021-08-09:
Hello all, i am trying the --control-port option:

`tsp -v --control-port 5555 -I ip 239.0.0.100:4000 -l 10.0.0.100  -O drop`

but when i run the netstat command, i do not see anything listening on that port:

`netstat -na | find ":5555"`
noting is returned
I am using windows 10 Pro

lelegard, 2021-08-09:
Did you try `tspcontrol`? Does it work?

`netstat` is a classical Unix command, this is not a native Windows command. If you run some Cygwin or Msys port, it won't probably have access to the Windows kernel and its specific use of TCP/IP. I wouldn't bet that `netstat` can display all open ports on Windows. So, the `tsp` control port may be successfully listening but `netstat` can't see it on Windows.

If `tspcontrol` does not work, then this may be a firewall issue. Firewalls don't really like unknown programs opening a listening port without permission. So, check the firewall.

enricovittorini, 2021-08-09:
thanks @lelegard for the reply,

tspcontrol does not work.

```
tsp -v -d -i --control-port 5555 -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d -i --control-port 5555
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes

```
```
tspcontrol -v -d --tsp 5555 exit
tspcontrol: Debug: debug level set to 1
tspcontrol: Debug: ====> tspcontrol -v -d --tsp 5555 exit
tspcontrol: Debug: debug level set to 1
tspcontrol: Debug: ====> tspcontrol exit
tspcontrol: Debug: binding socket to 0.0.0.0
tspcontrol: Debug: connecting to 127.0.0.1:5555
tspcontrol: error connecting socket: No connection could be made because the target machine actively refused it.
```

I tried with firewall and antivirus disabled.

netstat shows all open ports but not the tsp one.

lelegard, 2021-08-10:
OK, I see the problem.

I was surprised by the absence of debug trace _"control command thread started"_. This is due to the sequence of operations during the startup of `tsp`:

1. All plugins, except output, are initialized (command line options parsing, opening/creating resources).
2. Preload the global buffer (half of it by default) using the input plugin. Most of the time, this is enough to determine the TS bitrate using PCR's.
3. Initialize output plugin.
4. Create one thread per plugin and start them (the initial phase is sequential in the context of the main thread).
5. Create the control server thread, the one which handles commands from `tspcontrol`.

Your command has no specific input plugin, defaulting to plugin `file` reading on standard input. Since there is no input TS packet (standard input is the keyboard), the process is still waiting at step 2 above. The control server thread is not yet created.

The rationale for this startup order is that the control server manages plugin threads. It cannot be started before the plugin threads are created.

If you use an input plugin which has something to read (use `-I null` for instance), then the whole system is started and the control port is opened.

enricovittorini, 2021-08-10:
@lelegard understood. it works but I have some concerns:.
if i start an IP input plugin but do not have nothing at the input, i will not be able to terminate it wiith tspcontrol.
Is there a workaround for this scenario?

thanks,
E

lelegard, 2021-08-10:
> Is there a workaround for this scenario?

If this is an IP input, you can send a few null packets on the port to force the process to start:
~~~
tsp -I null -P until --packets 100 -O ip localhost:PORT
tspcontrol ...
~~~
Not perfect, but it should work.

closed 2021-11-13 by lelegard