tsduck/tsduck #844: TSEMMG stress test cause the MUX to crash

nikpakar, 2021-08-10:
Im simulating an EMM towards a commercial MUX/Head-end using tsemmg for a STB test bed. In exactly 2 hrs the EMM stream cause the mux to crash with no further info.

./tsemmg --mux 192.168.1.5:5000 -v --log-protocol --log-data --client-id 0x00100001

1. Scenario 1 - tseemg without xml file sending dummy emm data
2. Scenario 2 - tsemmg with a simple xml file as below containing two lines
3. Scenario 3 - same file with --section-mode

<tsduck>
<generic_short_table table_id="0x82">
  AA BB CC DD EE FF AA BB AA 01 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
</generic_short_table>
<generic_short_table table_id="0x82">
  BB CC DD EE FF AA BB CC AA 02 22 22 22 22 22 22 22 22 22 22 22 22 22 22 22 22
</generic_short_table>
</tsduck>

All 3 scenarios result the same identical mux crash in exactly 2 hours. Can this be a fault in MUX or do we need any extra parameters to send with emm.

Would highly appreciate any hint.

lelegard, 2021-08-10:
Hi @nikpakar 

Does the MUX equipment actually crash or does it simply close the connection with the EMMG? A commercial production equipment, MUX or whatever, shall never crash. Even in front of a faulty peer equipment, it should survive. A crash is a bug, there is no room for arguentation. Now, how to characterize it?

Since you use `--log-protocol --log-data`, did you see any abnormal  message from the MUX? Any stream_error message? Maybe you can try ` --log-protocol=info --log-data=debug` to avoid the flood of data_provision messages, to concentrate on the protocol itself. Be sure to redirect standard output and standard error to a text file to be able to analyze it afterwards.

`tsemmg` is just a test tool. It tries to respect the bandwidth which was actually allocated by the MUX. But, after a couple of hours, maybe the amount of data is slightly larger than the allocated bandwidth. Normally, the MUX shall reply with a stream_error message with an exceeded bandwidth error (0x000F). But there is maybe a bug in the MUX in case of input overflow (which would be a critical bug for a production equipment).

To stress the capability of the MUX to resist to overflow, you may try `--bandwidth XXX --ignore-allocated` with various XXX bandwidth. You can see the stream_BW_request and stream_BW_allocation response. At some point, you will ask for too much and the returned allocation will be less than the requested value. The option `--ignore-allocated` will then force `tsemmg` to ignore the allocation and overflow the MUX with the initially requested EMM bandwidth. This will show you how the MUX behaves in case of overflow. If it crashes immediately, there is a bug in the MUX and your initial crash is maybe due to a slow drift in `tsemmg` output. But if the MUX behaves properly, sends a stream_error message with an exceeded bandwidth error and drop EMM's in excess, then your initial problem is probably elsewhere.

nikpakar, 2021-08-11:
Hi Thierry,

Many thanks for your detailed reply. Below are the observations with your suggestions.

1. Initial crash identification - tsemmg just lost the connection, and the MUX gone to non responsive mode stopping multiplexing and output as well and had to restart to bring it back. Hence identified as a sort of crash
2. At the point of crash MUX's terminal frozen and shows Data Provision B/W as 2900 Kb/s where it was normally around 100 Kb/s. (MUX maximum allocated is 300 Kb/s while tseemg runs with 100 Kb/s
2. Ran with protocol=info and data=debug for more clarity with different bandwidth with --ignore-allocated. After 300 Kb/s, I can see MUX properly sending error message "error_status = 0x000F (exceeded bandwidth)"
3. When reach 2000 Kb/s MUX crashes

So the conclusion is with the running time, tsemmg build some extra BW and the mux continue to give error 0x000F till it reach around 2000 Kb/s but crashes afterward. 

Question: Is there anyway we can handle this bandwidth exceed from tsemmg ? i.e. forcing it not to exceed certain.
I will also try set MUX's max limit to 3000 Kb/s to see how well it can tolerate and whether tsemmg goes further beyond that as well. Its only 2 lines of emm data its sending for the test bed.

Once again many thanks for your detailed response and would highly appreciate any further advise.

lelegard, 2021-08-11:
"Normally" (a mythic adverb in computing), `tsemmg` sends fake EMM's at a fixed bitrate. This bitrate is initially given by the `--bandwidth` option (100 kb/s by default) and is reduced when the _stream_BW_allocation_ response message from the MUX allocates less. The option `--ignore-allocated` is here to test the capabilities of the MUX and stays with the `--bandwidth` value, even if the MUX allocates less.

But, in all cases, the data should be sent at a fixed bitrate, all the time. What you see on the MUX is that the received EMM bandwidth was 100 kb/s for a while and then increases to 2900 kb/s. The question is now: is the MUX right or wrong? Does `tsemmg` suddenly go crazy and multiply its output speed by 29? Or does the MUX incorrectly evaluate the input bandwidth?

As usual, the response is on the wire. I suggest that you retry your original test and capture packets in parallel using wireshark (use a capture filter based on the MUX address and port). After the MUX crashes, it should be relatively easy to check the time between two TCP packets using wireshark. If the EMM bandwidth jumps from 100 to 2900 kb/s, the duration between packet time stamps should be divided by 29. You should be able to see that quite easily on wireshark GUI. Take care that TCP packets may be sent in bursts. So, you have to identify the groups of TCP packets (very short interval between them) and the interval between them.

If you can confirm that the EMM bandwidth actually jumps to 2900 kb/s, this is a bug in `tsemmg` and I will investigate (there is also a bug in the MUX which does not behave properly).

nikpakar, 2021-08-11:
Thank you so much again for your clarification and got your point.

Im running the first case now with tcpdump on the interface. Will analyse the capture from wireshark and post the result.

nikpakar, 2021-08-12:
I have captured the crash point and struggling a bit to identify the point as you have described. Timestamps seems quite closer in every packet. Below is the crash point on Wireshark. For some reason Wireshark doesn't decode them as simulcrypt even the protocol is enabled.
tsemmg - 192.168.101.101
mux - 192.168.101.18

 ```
 60041 2920.845249    192.168.101.101       192.168.101.18        RSL Cell Broadcast 1570   SMS BroadCast CoMmanD  (CBS Page)
 60042 2920.845252    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43343132 Ack=84 Win=64066 Len=2920
 60043 2920.845276    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
 60045 2920.845473    192.168.101.101       192.168.101.18        RSL      1514   BCCH INFOrmation 
 60046 2920.845480    192.168.101.101       192.168.101.18        RSL      1514   DATA INDication 
 60048 2920.845697    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43351892 Ack=84 Win=64066 Len=1460
 60050 2920.845912    192.168.101.101       192.168.101.18        RSL      1514   BCCH INFOrmation 
 60052 2920.846197    192.168.101.101       192.168.101.18        RSL      2974   DATA INDication (DTAP) (BCC) 
 60054 2920.846400    192.168.101.101       192.168.101.18        RSL      1514   ESTablish CONFirm 
 60056 2920.846685    192.168.101.101       192.168.101.18        RSL      2350   [Malformed Packet]
 60058 2920.846909    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43361488 Ack=84 Win=64066 Len=1460
 60060 2920.847149    192.168.101.101       192.168.101.18        RSL      1514   [Malformed Packet]
 60062 2920.847467    192.168.101.101       192.168.101.18        RSL      1514   unknown 0 
 60064 2920.847666    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43365868 Ack=84 Win=64066 Len=2920
 60066 2920.847935    192.168.101.101       192.168.101.18        RSL      1514   ip.access CRCX 
 60068 2920.848234    192.168.101.101       192.168.101.18        TCP      2994   42756 → 5000 [PSH, ACK] Seq=43370248 Ack=84 Win=64066 Len=2940
 60070 2920.848472    192.168.101.101       192.168.101.18        TCP      658    42756 → 5000 [ACK] Seq=43373188 Ack=84 Win=64066 Len=604
 60072 2920.848718    192.168.101.101       192.168.101.18        RSL      2974   MultiRate CODEC MOD PERformed 
 60074 2920.848973    192.168.101.101       192.168.101.18        RSL      1514   [Malformed Packet]
 60076 2920.849265    192.168.101.101       192.168.101.18        RSL      2254   CCCH LOAD INDication 
 60078 2920.849500    192.168.101.101       192.168.101.18        RSL      1514   MultiRate CODEC MOD PERformed 
 60080 2920.849888    192.168.101.101       192.168.101.18        RSL      2614   BCCH INFOrmation 
 60081 2921.065900    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60082 2921.505923    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60083 2922.369970    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60085 2924.097922    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60088 2927.681933    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60107 2934.593917    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
```

nikpakar, 2021-08-12:
I have captured the crash point and struggling a bit to identify the point as you have described. Timestamps seems quite closer in every packet. Below is the crash point on Wireshark. For some reason Wireshark doesn't decode them as simulcrypt even the protocol is enabled.
tsemmg - 192.168.101.101
mux - 192.168.101.18, port - 5000
filtered for ip.dst == 192.168.101.18

```
59939 2920.242661    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43209496 Ack=1 Win=64149 Len=1460
59941 2920.243052    192.168.101.101       192.168.101.18        TCP      2102   42756 → 5000 [PSH, ACK] Seq=43210956 Ack=1 Win=64149 Len=2048
59943 2920.434770    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43213004 Ack=84 Win=64066 Len=2920
59944 2920.434778    192.168.101.101       192.168.101.18        RSL      2974   unknown 0 
59945 2920.434879    192.168.101.101       192.168.101.18        RSL      1702   BCCH INFOrmation 
59946 2920.434889    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43220492 Ack=84 Win=64066 Len=2920
59947 2920.434894    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
59948 2920.435056    192.168.101.101       192.168.101.18        RSL      2974   DATA REQuest (DTAP) (BCC) unknown 0x4d 
59949 2920.435064    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43229252 Ack=84 Win=64066 Len=2920
59950 2920.435239    192.168.101.101       192.168.101.18        RSL      2974   unknown 0 
59951 2920.435247    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
59953 2920.435445    192.168.101.101       192.168.101.18        RSL      1514   BCCH INFOrmation 
59954 2920.435452    192.168.101.101       192.168.101.18        RSL      1514   BCCH INFOrmation 
59956 2920.435517    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43240932 Ack=84 Win=64066 Len=2920
59958 2920.435784    192.168.101.101       192.168.101.18        RSL Cell Broadcast 1706   SMS BroadCast CoMmanD  (CBS Page)
59960 2920.436042    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43245504 Ack=84 Win=64066 Len=1460
59962 2920.436284    192.168.101.101       192.168.101.18        RSL      1514   BCCH INFOrmation 
59964 2920.436501    192.168.101.101       192.168.101.18        RSL      1514   unknown 0 
59965 2920.436506    192.168.101.101       192.168.101.18        RSL      1322   DATA REQuest (DTAP) (BCC) unknown 0x4d 
59967 2920.436754    192.168.101.101       192.168.101.18        RSL      1514   ESTablish CONFirm 
59969 2920.436982    192.168.101.101       192.168.101.18        RSL      1514   unknown 0 
59971 2920.437240    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43254072 Ack=84 Win=64066 Len=1460
59973 2920.437470    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43255532 Ack=84 Win=64066 Len=2920
59975 2920.437703    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43258452 Ack=84 Win=64066 Len=1460
59977 2920.438060    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43259912 Ack=84 Win=64066 Len=1460
59979 2920.438369    192.168.101.101       192.168.101.18        TCP      3582   42756 → 5000 [PSH, ACK] Seq=43261372 Ack=84 Win=64066 Len=3528
59981 2920.438605    192.168.101.101       192.168.101.18        RSL      1514   BCCH INFOrmation 
59983 2920.438894    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43266360 Ack=84 Win=64066 Len=1460
59985 2920.439120    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43267820 Ack=84 Win=64066 Len=1460
59987 2920.439355    192.168.101.101       192.168.101.18        RSL      2974   DATA INDication [Malformed Packet]
59989 2920.439646    192.168.101.101       192.168.101.18        TCP      2330   42756 → 5000 [PSH, ACK] Seq=43272200 Ack=84 Win=64066 Len=2276
59991 2920.439908    192.168.101.101       192.168.101.18        RSL      1514   unknown 0 
59993 2920.440165    192.168.101.101       192.168.101.18        RSL      1514   CCCH LOAD INDication 
59995 2920.440540    192.168.101.101       192.168.101.18        RSL      3126   BCCH INFOrmation 
59997 2920.441042    192.168.101.101       192.168.101.18        RSL      1398   unknown 125 
59999 2920.640898    192.168.101.101       192.168.101.18        RSL Cell Broadcast 2974   SMS BroadCast CoMmanD  (CBS Page)
60000 2920.640907    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
60001 2920.640994    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
60002 2920.641002    192.168.101.101       192.168.101.18        RSL      2974   unknown 0 
60003 2920.641186    192.168.101.101       192.168.101.18        RSL      2974   DATA INDication [Malformed Packet]
60004 2920.641193    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
60006 2920.641435    192.168.101.101       192.168.101.18        RSL      2862   BCCH INFOrmation 
60008 2920.641697    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43302140 Ack=84 Win=64066 Len=1460
60010 2920.641925    192.168.101.101       192.168.101.18        RSL      1514   [Malformed Packet]
[... cut ...]
 60083 2922.369970    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60085 2924.097922    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60088 2927.681933    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
 60107 2934.593917    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
```

lelegard, 2021-08-12:
Hi,

Is is indeed strange that DVB SimulCrypt is not decoded as such by Wireshark. The protocol is however correct, otherwise you wouldn't be able to communicate with the MUX. Did you explicitly enable DVB SimulCrypt in Wireshark? I don't think it will detect it automatically.

There are some interesting information in you dump. As I predicted in a previous post, you get bursts of 20 to 30 IP packets, with a 200 ms gap between bursts.
~~~
59939 2920.242661    192.168.101.101       192.168.101.18        TCP      1514   42756 → 5000 [ACK] Seq=43209496 Ack=1 Win=64149 Len=1460
59941 2920.243052    192.168.101.101       192.168.101.18        TCP      2102   42756 → 5000 [PSH, ACK] Seq=43210956 Ack=1 Win=64149 Len=2048
-- 200 ms gap
59943 2920.434770    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43213004 Ack=84 Win=64066 Len=2920
59944 2920.434778    192.168.101.101       192.168.101.18        RSL      2974   unknown 0 
.... more messages
59995 2920.440540    192.168.101.101       192.168.101.18        RSL      3126   BCCH INFOrmation 
59997 2920.441042    192.168.101.101       192.168.101.18        RSL      1398   unknown 125 
-- 200 ms gap
59999 2920.640898    192.168.101.101       192.168.101.18        RSL Cell Broadcast 2974   SMS BroadCast CoMmanD  (CBS Page)
60000 2920.640907    192.168.101.101       192.168.101.18        RSL      2974   BCCH INFOrmation 
.... more messages
60031 2920.644478    192.168.101.101       192.168.101.18        RSL Cell Broadcast 1514   SMS BroadCast CoMmanD  (CBS Page)
60033 2920.644794    192.168.101.101       192.168.101.18        RSL      2614   [Malformed Packet]
-- 200 ms gap
60036 2920.844904    192.168.101.101       192.168.101.18        TCP      2974   42756 → 5000 [PSH, ACK] Seq=43327016 Ack=84 Win=64066 Len=2920
60037 2920.844912    192.168.101.101       192.168.101.18        RSL      2974   unknown 0 
.... more messages
60078 2920.849500    192.168.101.101       192.168.101.18        RSL      1514   MultiRate CODEC MOD PERformed 
60080 2920.849888    192.168.101.101       192.168.101.18        RSL      2614   BCCH INFOrmation 
-- 200 ms gap
60081 2921.065900    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
60082 2921.505923    192.168.101.101       192.168.101.18        TCP      1514   [TCP Retransmission] 42756 → 5000 [ACK] Seq=43381832 Ack=84 Win=64066 Len=1460
~~~

First observation, the session ends with endless TCP retransmissions at the same TCP sequence number. This means that the MUX is hung either at process or operating system level and no longer reads packets. The MUX "application process" did not crash, otherwise the OS of the MUX would have probably closed the socket, which is not the case (no TCP RST). There is clearly a bug in the MUX.

Second, this dump is the end of the transmission. Have a look at the beginning of the session, after the setup, or in the middle of the session, when EMM's are sent on a nominal bitrate. Do you observe the same patterns of 20-30 messages with 200 ms gap?
- If yes, then the transmission speed has not significantly changed, and certainly not by a factor of 29. So, the jump to 2900 kb/s is an interpretation error from the MUX which seems to be otherwise confused anyway. You have to report the issue to the MUX vendor, it is on their side.
- If no (just one or two message per burst or much longer gaps), then `tsemmg` went crazy. I will have to investigate. You may also report to the MUX vendor that their device hangs in the middle of operation but they will probably respond that the operational conditions are not nominal, (which is not an excuse to crash a production system).

Please let us know the result.

nikpakar, 2021-08-12:
Thank you so much again for the investigation.

Simulcrypt is enabled in wareshark, but didn't decode. I have specifically enabled it while disabling all others. Still didnt worked. Wireshark is the latest downloaded from the site yesterday. 

The trace was on for some time even after the crash and the 6 retransmission we see are the only ones at the end. This is probably the MUX process initially hung and immediately died after few seconds. There are no comms afterward. and mux was not responsive at all.

There are no such 200ms gap at the very beginning, but can see towards the end after around 2 - 2 1/2 hrs.

MUX vendor has investigated and identified that we are sending larger bursts often at the later stage and its not that mux crashed from the first burst, but after many. and the bursts are not exactly 2900, they are varied from 1800 - 2900 kb/s most.

They have also confirmed the mux has crashed due to quite a few bursts it reached which are much larger than it set for. Mux is set to accept 300 Kb/s.

lelegard, 2021-08-12:
> They have also confirmed the mux has crashed due to quite a few bursts it reached which are much larger than it set for. Mux is set to accept 300 Kb/s.

There is a confusion between "burst" and "bitrate". A bitrate is never smooth, we never send bits or even packets one by one. There is always some kind of burst. The actual question is how many bits over which period? In other words, what is the duration of a so-called burst?

The EMMG/PDS <=> MUX protocol does not specify the granularity of the burst. This is implementation dependent. The protocol only specify bitrates and a bitrate is an _average_, a number of kb over one second. So, having bursts is normal. They just need to be "reasonable".

>  the bursts are not exactly 2900, they are varied from 1800 - 2900 kb/s most.

What does this mean exactly? Does this mean that 2900 kb are received _every_ second? In that case, the average bandwidth is 2900 kb/s, which is incorrect. Or does this mean that 2900 kb are received in one second and then nothing during 28 seconds? In that case, the "average" bitrate is 100 kb/s but implemented in a rather stupid way (by `tsemmg` in that case).

nikpakar, 2021-08-12:
I think I have misinterpreted the word by saying a burst. My mistake, extremely sorry about it.

What they say is they see higher bitrates ranging from 1800 to 2900 Kb/s from the mux.

If its ok I can post two screenshots from the mux terminal which shows when its normal and when its receiving higher.

lelegard, 2021-08-12:
> What they say is they see higher bitrates ranging from 1800 to 2900 Kb/s from the mux.

In that case, we are back to the initial question: is the MUX right or wrong when it evaluates the bitrate?

Could you provide two pcap files from your Wireshark capture? The idea is to get a 5-second capture of the middle of the session (when the bitrate is considered normal) and the last 5 seconds before the MUX hangs.

If `tsemmg` goes crazy with bitrates at some point, could you give the exact TSDuck version and operating system type and version, as well as if this is a virtual or physical machine?

nikpakar, 2021-08-12:
Thank you so much again. Im preparing to isolate the network and run the test again and will capture those 5 seconds and send it.

Meanwhile, TSDuck is the latest version from git runs on Ubuntu 20.04 in a LXC container. LXC container runs on a Proxmox PVE. Would that be the reason for such behaviour ?

lelegard, 2021-08-12:
> LXC container runs on a Proxmox PVE. Would that be the reason for such behaviour ?

It could be. In fact, I had this in mind when I asked the question. I have no experience with Promox PVE but I have quite a lot with VMware on a macOS host and some with VirtualBox on a Windows host. In short, real-time response and time precision are not so good on virtual machines. This is my experience but specific tuning methods may exist.

As an example, there is a unitary test program with TSDuck in subdirectory `src/utest`. It tests all the low-level features of most base C++ classes in TSDuck. All the classes which encapsulate system time features are tested (timers, monotonic clock, synchronization timeouts, etc). It is difficult to test time in an automatic and reproducible manner. So, there are some precautions. For instance, a 500 ms timeout tests succeeds if the difference of measured times is greater than 500 ms and less than 500 ms + a reasonable margin (say +400 ms to be generous). The same cautious approach is used everywhere for tests involving time. On physical machines, macOS, Linux, Windows, all tests always pass. On virtual machines, however, some of these tests fail from time to time. For instance, a 500 ms timeout is woken up after 450 ms (or 450 measured ms). Some other timers are much longer than expected. I do not know if the problem comes from the _action_ (timer, timeout) or its _measurement_ but in all cases there is some inconsistency in the system time management of the virtual machine. The failure rate is not very high, say less than 5 or 10%, but exclusively on virtual machines, never on physical machines. So, I had to create a "weak failure" status and a different class of unitary test assertions for tests involving timers. If the unitary test suite completes with weak failures only, all those failures are reported but the overall test suite is considered successful. Without that class of unitary test weak failures, automating the build on many virtual machines was close to impossible, too many failures were reported.

I think that have seen that on Linux VM's only but I have only one Windows VM, no macOS VM (just a real Mac) but more than 10 Linux VM's because of the large number of different distros. One interesting point: On GitHub, all push actions trigger an automatic compilation, unitary test suite and system test suite on Linux, Windows and macOS. All of these run in GitHub-hosted virtual environment. I am notified only in case of failure, so weak failures are unnoticed unless I explicitly examine the log. I rarely examine the logs of successful CI/CD operations but I do from time to time and I noticed that kind of time-related weak-failures sometimes. So, yes, virtual machines matter.

Containers use the host kernel, so they should be less impacted, at least if the kernel runs on a physical machine. Virtual machine + containers seem a bad idea to run time-critical systems. But I may be wrong and others may have some better tuning experience of that environment.

As a conclusion, I would say that virtualization is a great IT tool for data processing servers, application servers, web servers, etc. but very poor for applications which require some time accuracy in the millisecond magnitude.

nikpakar, 2021-08-12:
Thanks a lot again for the very comprehensive inputs. Indeed lots of valuable information. Thanks again.

My LXCs runs on the Proxmox and the proxmox runs on a bare-metal Sun Fire X4175. But i think it ll be really worth while to test it on a bare-metal server based on your inputs. So im setting up a new environment with a new hardware server.

Will run the same test from that and update the result first.

nikpakar, 2021-08-13:
Result with a physical server was also same. Below are the captures from the start, at 1st hr, at 2nd hr and at the end and also tsemmg output.

[tsemmg-pcap.zip](https://github.com/tsduck/tsduck/files/6981260/tsemmg-pcap.zip)

lelegard, 2021-08-13:
Thanks, I will have a look at your capture files.

For your information, Wireshark can format the DVB SimulCrypt protocol on this session. As mentioned in a previous post, you have to enable it on this specific TCP port because there is no standard port for this protocol (unlike SSH, FTP, etc) and it cannot be recognized automatically. Use Analyze -> Decode As... and then, considering that your MUX uses the server port 5000:

[image]

And the packets are displayed as follow:

[image]

lelegard, 2021-08-13:
Hi @nikpakar 

I did a quick analysis tool for pcap files (based on the TSDuck `pcap` plugin code). It gives the data size (and bitrates based on timestamps) of the TCP data size. Filtering on `tsemmg` -> MUX, the data size includes the DVB SimulCrypt overhead (a few percents).

From your capture files:
- In the "after 1 hour" file, the bitrate is 76.967 kb/s
- In the "end" file, the bitrate is 94.356 kb/s

So, my conclusion is that the EMMG bandwidth consistently remains under 100 kb/s. Your MUX must have a severe bug somewhere with multiple consequences: incorrect input bitrate evaluation, hang, crash, etc.

The capture files prove that the incoming data are correct and the MUX is not. You may use them to report the bug to the MUX vendor. A link to this page would be probably useful as well.

nikpakar, 2021-08-14:
Hi @lelegard 

Thank you so much for the comprehensive analysis. I will take it up with the MUX vendor to further troubleshoot the issue.

Thanks about the tcp port mapping on Wireshark. It works now and decoding as simulcrypt.

Thank you so much again.

nikpakar, 2021-08-14:
Hi @lelegard 

After investigating, MUX vendor suggested to restrict number of datagrams in a tcp packet to 7. So I have used tsemmg's --bytes-per-send parameter to control the datagrams to 7.

Interestingly this changed the behaviour.

Now in exact time duration, tsemmg reports "tsemmg: too many invalid messages from mux:5000, disconnecting" in the log and exit.
MUX doesn't crash now and stays in good health.
When restart tsemmg, everything works fine again

This repeats every 2 1/2 hrs.

Wireshark now shows tsemmg sends few Malformed packets towards the end and the MUX replies with Stream error messages to the tsemmg just before the tsemmg exit with "too many invalid messages:"

Attached is the new capture at the exit point. Im troubleshooting further. If you see any thing out of this development so far, would highly appreciate your comment.

[new-end.pcapng.zip](https://github.com/tsduck/tsduck/files/6987000/new-end.pcapng.zip)

nikpakar, 2021-08-14:
Hi @lelegard

After investigating, MUX vendor suggested to restrict number of datagrams in a tcp packet to 7. So I have used tsemmg's --bytes-per-send parameter to control the datagrams to 7.

Interestingly this changed the behaviour.

Now in exact time duration, tsemmg reports "tsemmg: too many invalid messages from mux:5000, disconnecting" in the log and exit.
MUX doesn't crash now and stays in good health.
When restart tsemmg, everything works fine again

This repeats every 2 1/2 hrs.

Wireshark now shows tsemmg sends few Malformed packets towards the end and the MUX replies with Stream error messages to the tsemmg just before the tsemmg exit with "too many invalid messages:"

Attached is the new capture at the start and exit point. It shows no malformed packets at the beginning but only appear at the end.

Im troubleshooting further. If you see any thing out of this development so far, would highly appreciate your comment.

[New start & end.zip](https://github.com/tsduck/tsduck/files/6987015/New.start.end.zip)

nikpakar, 2021-08-15:
Hi @lelegard

After investigating, MUX vendor suggested to restrict number of datagrams in a tcp packet to 7. So I have used tsemmg's --bytes-per-send parameter to control the datagrams to 7.

Interestingly this changed the behaviour.

Now in exact time duration, tsemmg reports "tsemmg: too many invalid messages from mux.., disconnecting" in the log and exit.
MUX doesn't crash now and stays in good health.
When restart tsemmg, everything works fine again

This repeats every 2 1/2 hrs.

Wireshark now shows the MUX replies with Stream error messages to the tsemmg just before the tsemmg exit with "too many invalid messages:" and error message says "Exceeded BW (peak 1949184bps) : forcing stream close"

MUX vendor confirms that they read this correct and this is the actual cause.

Attached is the new capture of the last 1+ hour till it exit with this error.

Im troubleshooting further. If you see any thing out of this development so far, would highly appreciate your comment.

[5. New end.pcapng.zip](https://github.com/tsduck/tsduck/files/6987458/5.New.end.pcapng.zip)

pwaruna, 2021-08-16:
Hi @lelegard 

I have tested the same scenario with a completely different MUX (Different model, software but same vendor)

And the result appeared to be identical. Were you able to identify any thing from my last findings.

Thanks again for kind support.

lelegard, 2021-08-16:
Thanks for the latest feedback. I am away for a couple of days without any computer. I will check by the end of the week.

lelegard, 2021-08-18:
Hi @nikpakar,

I analyzed the file "5. New end.pcapng" by ranges of captured packets. The TCP data bitrate from 192.168.201.100 to 192.168.201.18:5000 is consistently 95-96 kb/s except in the end phase. After packet number 51600 (approximately), the data bitrate suddently jumps higher.

By ranges of 100 captured packets:
~~~
Packet 51000:     96,734 b/s
Packet 51100:     96,540 b/s
Packet 51200:     96,542 b/s
Packet 51300:     96,730 b/s
Packet 51400:     97,516 b/s
Packet 51500:     96,540 b/s
Packet 51600:    123,170 b/s
Packet 51700: 13,684,932 b/s
Packet 51800:  3,886,039 b/s
Packet 51900:  3,595,472 b/s
Packet 52000:  2,839,176 b/s
Packet 52100:  3,212,059 b/s
Packet 52200:  2,909,507 b/s
Packet 52300:  2,572,278 b/s
~~~

So, yes, there is a weird issue in `tsemmg`. I will investigate but it may be hard to reproduce.

- Which version of TSDuck do you use?
- What is the exact `tsemmg` command which was used for capture file "5. New end.pcapng"?
- Could you please provide the end of the `tsemmg` log, including the "too many invalid messages from mux" (it shouldn't happen).

nikpakar, 2021-08-19:
Hi @lelegard 

Thanks again for investigating and looks like we have finally found the reason.

This jump is always consistence and it happens around that very same time after tsemmg starts. you can easily reproduce it by letting the tsemmg run for about 2 1/2 hrs. This always happens around that time.

Version
tsemmg: TSDuck - The MPEG Transport Stream Toolkit - version 3.28-2470

and this is the exact command i run now
./tsemmg -v --mux 192.168.201.1:5000 --log-protocol --log-data --client-id 0x00FF0001 --section-mode --bytes-per-send 203 emm.xml
(bytes per send was introduced later, and  without that and --section-mode also the result was the same.)

End of tsemmg does not show any other logs or errors other than
tsemmg: too many invalid messages from mux.., disconnecting
just before it exit.

lelegard, 2021-08-19:
Hi @nikpakar,

I fixed the issue (or at least I think it is fixed). Could you please rebuild from the latest repo and check if the fix is correct in your case?

nikpakar, 2021-08-20:
Hi @lelegard 

Thank you so much for the fix. Im now running tseemg for 4hrs without that issue. It used to come in at 2 1/2hrs.

So looks like its fixed. Thanks a lot again and will keep it running for longer period and update the result again.

nikpakar, 2021-08-21:
Hi @lelegard 

Its been running for more than 24hrs now without the issue. We can confirm its been fixed now.

closed 2021-08-30 by lelegard