tsduck/tsduck #847: [BUG] PSI json ouput last node not closed

enricovittorini, 2021-08-16:
#### Bug description:

tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.27-2438

Command used:
tsp -I ip xxxxx -l yyyyy -P psi  -a -j - -O drop.

The last node does not have the closing "}"

```
[
  {
    "#name": "PAT",
    "#nodes": [
      {
        "#name": "metadata",
        "pid": 0
      },
      {
        "#name": "service",
        "program_map_pid": 32,
        "service_id": 1
      }
    ],
    "current": true,
    "network_pid": 16,
    "transport_stream_id": 1,
    "version": 14
  },
  {
    "#name": "CAT",
    "#nodes": [
      {
        "#name": "metadata",
        "pid": 1
      }
    ],
    "current": true,
    "version": 1
  },
  {
    "#name": "PMT",
    "#nodes": [
      {
        "#name": "metadata",
        "pid": 32
      },
      {
        "#name": "component",
        "#nodes": [
          {
            "#name": "ISO_639_language_descriptor",
            "#nodes": [
              {
                "#name": "language",
                "audio_type": 0,
                "code": "ita"
              }
            ]
          }
        ],
        "elementary_pid": 2541,
        "stream_type": 4
      },
      {
        "#name": "component",
        "elementary_pid": 2542,
        "stream_type": 27
      }
    ],
    "current": true,
    "service_id": 1,
    "version": 2
  },
  {
    "#name": "NIT",
    "#nodes": [
      {
        "#name": "metadata",
        "pid": 16
      },
      {
        "#name": "network_name_descriptor",
        "network_name": "Test"
      },
      {
        "#name": "transport_stream",
        "#nodes": [
          {
            "#name": "service_list_descriptor",
            "#nodes": [
              {
                "#name": "service",
                "service_id": 1,
                "service_type": 1
              }
            ]
          }
        ],
        "original_network_id": 1,
        "preferred_section": 0,
        "transport_stream_id": 1
      }
    ],
    "actual": true,
    "current": true,
    "network_id": 1,
    "version": 3
  },
  {
    "#name": "SDT",
    "#nodes": [
      {
        "#name": "metadata",
        "pid": 17
      },
      {
        "#name": "service",
        "#nodes": [
          {
            "#name": "service_descriptor",
            "service_name": "Srv_2",
            "service_provider_name": "",
            "service_type": 1
          }
        ],
        "ca_mode": false,
        "eit_present_following": false,
        "eit_schedule": false,
        "running_status": "undefined",
        "service_id": 1
      }
    ],
    "actual": true,
    "current": true,
    "original_network_id": 1,
    "transport_stream_id": 1,
    "version": 8
```

lelegard, 2021-08-17:
Hi @enricovittorini 

This is not a bug. The input processing is not terminated, the output is not terminated either. The `ip` input is endless, it reads packets endlessly. If some new table is processed, you will see the closing `},` of the current table and then the opening `{` of the next table. If there is some terminating condition such as `-P until ...` or if you interrupt the processing using Ctrl-C, then you will see the two terminating lines `}` and `]`.

The reason why you don't see the closing `}` of the current table while the processing is still in progress is because the `psi` plugin doesn't know yet if the next line will be `},` (before a new table) or `}` (without comma, at end of list of tables, at end of processing).

enricovittorini, 2021-08-17:
thanks @lelegard , but how would you process the a live analysis?
i explain what i am doing: i am reading the stout to parse object by object the received JSON.

If the JSON is not closed properly then it is not considered valid by the parser (and not taken)

In a live analysis, there should always be a "," after each node in the stout. don't you agree?

would it be possible to create a patch to the logger to say "always close a JSON node for live": --live-json ?

thanks
Enrico

lelegard, 2021-08-17:
> In a live analysis, there should always be a "," after each node in the stout. don't you agree?

No. Each plugin is independent of the other and cannot determine if the output is endless or not. The presence of `-P until ...` before or after `-P psi` makes a difference.

> would it be possible to create a patch to the logger to say "always close a JSON node for live": --live-json ?

It should also close the list with `]`. Otherwise the parse would complain about an unterminated list.

I will consider something like this for this use case. But it is a very difference type of output. Each table should be a complete JSON file.

enricovittorini, 2021-08-17:
understood, in my case i am reading the stout as a "stream", and extract node by node what is received. {... } after {...},
would be be able to indicate which file should i look at to arrive to what i need? I will try to understand if i will be able to change it by myself for the time being. 
thanks!

lelegard, 2021-08-17:
In fact, your use case is already addresses by option `--log-json-line` (there is also an option named `--log-xml-line` for those who prefer parsing XML). 

Please read the description of `--log-json-line`.

Example:
~~~
$ tsp -I ip 6666 -P psi -a --log-json-line=@@HERE@@:  -O drop
* psi: @@HERE@@:{ "#name": "PAT", "#nodes": [ { "#name": "metadata", "pid": 0 }, { "#name": "service", "program_map_pid": 33, "service_id": 800 } ], "current": true, "transport_stream_id": 930, "version": 11 }
* psi: @@HERE@@:{ "#name": "PMT", "#nodes": [ { "#name": "metadata", "pid": 33 }, { "#name": "component", "#nodes": [ { "#name": "T2MI_descriptor", "num_t2mi_streams_minus_one": 0, "pcr_iscr_common_clock_flag": false, "t2mi_stream_id": 0 } ], "elementary_pid": 64, "stream_type": 6 } ], "current": true, "service_id": 800, "version": 11 }
~~~

This is much easier than reading stdout as a stream. Read stderr and filter lines containing the string that you specified to `--log-json-line`, in the example it is `@@HERE@@:`. Just specify something that you are confident to uniquely identify the lines to parse. Drop the beginning of the line, up to your marker string. The rest of the line is a complete JSON structure. There is one line per table.

Similar options are available in plugins `tables`, `analyze` and a few others.

lelegard, 2021-08-17:
And if you parse the JSON output using Python, you do not even need to use `tsp`. You can activate the plugin directly from your Python application and handle the logged lines directly from a Python class without creating any process, no pipe, no stream I/O.

See an [introduction here](https://tsduck.io/doxy/jpbindings.html), the list of TSDuck [Python classes here](https://tsduck.io/doxy/group__python.html) and [sample Python programs here](https://github.com/tsduck/tsduck/tree/master/sample/sample-python).

The same features exist for Java.

closed 2022-09-23 by lelegard