tsduck/tsduck #848: [Q] Watch multiple PIDs with AbstractTablePlugin

nimblin, 2021-08-20:
Hello @lelegard - What would be the suggested way to use `AbstractTablePlugin` to modify tables on multiple PIDs ? As it is, `AbstractTablePlugin` can only watch for one PID at a time through: 

``` c++ 
void ts::AbstractTablePlugin::setPID(PID pid);
```

Or passing it directly to the ctor. I would find it convenient for it to expose a method similar to:
``` c++
ts::AbstractDemux::addPID(PID pid);
```
I would like a solution that does not imply running multiple instances of the same plugin inhereting from `AbstractTablePlugin` as I want to do this programmatically.

Thanks.

lelegard, 2021-08-20:
Hi @nimblin 

>  I would find it convenient for it to expose a method similar to: `ts::AbstractDemux::addPID(PID pid);`

The short answer is no.

This would be against TSDuck principles: design the simplest possible building blocks with complete freedom to combine them in any order (borrowed from the original Unix philosphy). This is one of the reasons why, unlike many software, TSDuck did not collapse on itself after 15 years of continuous development: keep a low internal complexity.

The idea of TSDuck has always been to process "one thing" and create multiple instances to process multiple similar things. Typically, when processing multiple PMT's, you add one `pmt` plugin per PMT to process. This is done on purpose. If you want to do it programmatically, just create new instances of the same subclass of `AbstractTablePlugin`.

To do what you request, the internal complexity of the class would increase to manage bitrates, repetitions, versions, etc, per PID, ending in something that would be like several instances of  `AbstractTablePlugin` inside one. So, why not just keep multiple instances?

If you can exhibit one use case which cannot be implemented using several instances of the plugin, we will try to find some solution, other than duplicating everything inside `AbstractTablePlugin`.

nimblin, 2021-08-21:
> The short answer is no. This would be against TSDuck principles...

I must admit I feared this would be the answer, although I support your development principles - as you said they prove to be great.

> Typically, when processing multiple PMT's, you add one pmt plugin per PMT to process. This is done on purpose. If you want to do it programmatically, just create new instances of the same subclass of AbstractTablePlugin.

Lets say that I want to modify all instances of a proprietary table (for which I have built support into TSDuck), and this table appears on non predictable PIDs in the TS. If I understood you correctly, the way to do this would require two plugins and one custom TSP application where: 
* The application stars with plugin (A) which looks for the target PIDs (by following *PAT->PMT->etc*) and signals an event back to the TSP application when a new target PID is found
* The plugin (A) event handler in the application adds a new plugin (B) for every new target PID found by plugin (A). Plugin (B) is the one that is a subclass of 'AbstractTablePlugin'.

A comment on this approach: `TSProcessor` options can not change dynamically - so the plugin Event Handler for (A) would have to stop and re-run the `TSProcessor`. It would be great if It was possible modify `ts::TSProcessorArgs` in the application whenever I wished and do something along the lines of  `tsproc.reload()` to be able to add and remove plugins to the plugin pipeline without having to stop and re-run TSP.

Is my reasoning correct? 

On a side note, if my reasoning happens to be correct, I think that a Processor Plugin to extract all PIDs for any given TID would be a great contribution to the project.

lelegard, 2021-08-21:
> Lets say that I want to modify all instances of a proprietary table (for which I have built support into TSDuck), and this table appears on non predictable PIDs in the TS. If I understood you correctly, the way to do this would require two plugins and one custom TSP application

This may not be necessary.

Let's take the example of the `pmt` plugin again. The command `-P pmt --service foo ...` means that we want to work on the PMT of the service named "foo". In that case, we do not know the PID where the PMT is. The plugin has to wait for an SDT, grab the service id of "foo", wait for the PAT, grab the PMT PID for this service id. At that point, the plugin can start working on the PMT.

You may implement the same type of initial discovery phase, based on your signalisation, before working on the actual PID of the table.

Otherwise, it may be possible that `AbstractTablePlugin` is not the right superclass, typically if your table is dynamic, moving from one PID to another based on some proprietary signalisation. A plugin manipulating tables typically uses a `SectionDemux` to intercept the right information, adding and removing PID's to the demux based on the received information. At the beginning, plugins such as `pat`, `pmt`, `sdt` and the like were implemented that way. At some point, I realised that they shared a lot of similar code and I decided to do some refactoring to move the common code to a superclass. But, it is quite understandable that other proprietary processing does not fit in that common superclass and you need to implement your own processing (based on the way other plugins were already coded, no need to reinvent the wheel).

nimblin, 2021-08-21:
> You may implement the same type of initial discovery phase, based on your signalisation, before working on the actual PID of the table.

I had already accomplished this - my plugin inherits from `AbstractTablePlugin` and overrides `handleTable()` from `TableHandlerInterface` as well as the `modifyTable()` methods. `handleTable` does the discovery by following *PAT->PMT->etc*, and calls `setPID()` once it finds a target PID. This lets me find **all** target PIDs, yet I can only modify **one** at a time (and modifying one works fine, so `AbstractTablePlugin` suits my case). I got stuck trying to make it modify all. 

> This may not be necessary.

The only other place I can think of to create multiple instances of `TableHandlerInterface` is within a plugin itself, however, I do not see how I could possibly create multiple instances of `AbstractTablePlugin`within one plugin, aren't plugins instantiated by an executor class?. `PMTPlugin` does not help me as a reference because it can only work on one service at a time, and, as far as I know, there is not a single plugin that creates multiple instances of `AbstractTablePlugin`.

A summary to make sure we remain on the good track as these messages are getting dense:
* (1) I can discover all my target PID with one plugin
* (2) I can also modify *one* target PID with that same plugin
* (3) I proposed a way to modify all target PIDs using two plugins
* (4) You suggested that the two plugin approach might not be necessary (which hints me that modifying **all** target PIDs within the same plugin might be possible) yet your explanation used `PMTPlugin` as an example - which can only modify one target at a time, something that I had already accomplished (2).

Thanks a lot

lelegard, 2021-08-21:
There is no need to create several instances of `TableHandlerInterface`. Just use one. But in the `handleTable()` method you must be prepared to do different things, on different tables, based on the current state of the plugin.

What you want to do is perfectly standard in TSDuck. The only point is whether or not `AbstractTablePlugin` is the appropriate superclass for your plugin. Maybe `ProcessorPlugin` (the superclass) is more appropriate.

If you want to go further, you have to give more details on how your proprietary tables are located, why there are several of them (are they linked or independent), how do you want to update them (no details on what your table contains, just how it is integrated in the signalisation).

nimblin, 2021-08-21:
Sorry that was a typo, I meant to say *create multiple instances of `AbstractTablePlugin`*. My table is essentially an [AIT](https://www.etsi.org/deliver/etsi_ts/102800_102899/102809/01.03.01_60/ts_102809v010301p.pdf) with some extra fields. There is also no support for broadcast delivery of applications so do not worry about the DSMCC OC. I want to manipulate fields which can affect section size (consider for example the application_name_descriptor) and we can assume there is sufficient stuffing.

lelegard, 2021-08-21:
So, if your table is, like an AIT, part of a service, your plugin is very similar to the `pmt` plugin. There is only one additional step in the initial phase: grab the PMT to get the PID of your table. And, just like the `pmt` plugin, one instance of the plugin manipulates one instance of your table. Use several instances to manipulate several tables. Use the same skeleton as the `pmt` plugin. It will work.

nimblin, 2021-08-21:
Understood. Thank you @lelegard, very helpful, as always. To finally close this issue, could you please comment on this:
> TSProcessor options can not change dynamically - so the plugin Event Handler for (A) would have to stop and re-run the TSProcessor. **It would be great if It was possible modify ts::TSProcessorArgs [...]

Would such a feature make sense to you? perhaps it is something that I can contribute myself.

lelegard, 2021-08-22:
> the plugin Event Handler for (A) would have to stop and re-run the TSProcessor.

I am not sure to understand what you mean. What would be the exact use case ?

> It would be great if It was possible modify ts::TSProcessorArgs in the application whenever I wished and do something along the lines of tsproc.reload() to be able to add and remove plugins to the plugin pipeline without having to stop and re-run TSP.

This could be technically possible but you enter the danger zone. Implementing this would not be trivial and would require intensive testing to be sure that nothing is broken. This represents quite a lot of work, mostly designing test cases. You should really try to find solutions with the current infrastructure first.

nimblin, 2021-08-23:
> I am not sure to understand what you mean. What would be the exact use case ?

My end goal is to have a standalone C++ application that I can just fire and have it replace all of my tables, without the human factor of first figuring out the target PIDs and then adding as many table-modifying plugins to `tsp` as necessary. This is why an `addPID()` method would be useful in the `AbstractTablePlugin` class - I could have a single plugin instance that does the discovery AND modification of all of my tables in one shot.

Without that method, my only other option seems to be creating a TSP application that replicates that human factor: it would run two independent tsp instances, one with the target-discovery plugin, and another with the target-modification plugins, which would have to be shut down every time a new target is found to add another target-modification plugin in the plugin pipeline, and be run again.

This last solution is obviously less clean (a whole new tsp instance + slower response times between a new target is found and it starts to be modified, potentially letting some unmodified tables slip through).

nimblin, 2021-08-23:
To clarify - by "replace all of my tables" I mean modify something in those tables.

lelegard, 2021-08-23:
Do not over-complicate things. Everything can be done in one plugin. I have solved more complicated problems with one plugin, never by modifying tsp. I think that you initially focused on `AbstractTablePlugin` because it is used by some existing plugins that _you thought_ were similar but in fact your structure is different.

The overall idea is the following. It is based on the implementation of several other plugins. You will have to adjust the details if my assumptions about your table are not true.

- Define your plugin as a direct subclass of `ProcessorPlugin` (regular packet processing plugin).
- Implement `TableHandlerInterface` in your plugin.
- Define one "PID context" class which holds the context of a PID containing your table.
  - Use a `CyclingPacketizer` in each PID context.
  - In the plugin, use an `std::map` or `SafePtr` to PID context, indexed by PID value.
- In the plugin, use a `SectionDemux`.
  - In your constructor plugin, use your plugin instance as table handler
  - In the `start()` method, reset the demux and add  `PID_PAT` to it.
  - In your `handleTable()`, use a big switch on the table id of the received table.
  - When this is a PAT, add all PMT PID to the demux
  - When this is a PMT, look for a reference to your table (your private signalization) and add its PID to the demux.
  - When this is an instance of your table, deserialize it, modify it and replace it in the packetizer of the PID context for this PID.
- In the method `processPacket()` of your plugin:
  - Feed the demux with the TS packet.
  - If the packet PID is from one of your tables (= if the PID is used as index in the map of PID context), replace the TS packet by the next packet from the packetizer of this PID context.

That's all, Really.

If you work on live streams and do not want to let previous versions of your table pass, typically in the very early phase, before you got the PMT and identified your table's PID, then mute all PID's as long as you did not get all PMT's. Here, "mute" means replace all TS packets by null packets in `processPacket()` as long as you got all PMT's.

Otherwise, make sure to increment the version of a table when you modify it to make sure the receiver will get the updated version.

All these techniques are used in various places in TSDuck source code. In case of doubt, just look for them and adapt them to your context.

nimblin, 2021-08-23:
Thanks for the detailed explanation, the map of PID contexts is eye opening, I was confused by the idea of having a container of `CyclingPacketizer` as they are no-copy. Closing this issue now.

closed 2021-08-23 by nimblin