tsduck/tsduck #850: [BUG] Can't suspend and resume SRT output plugin

wongfei2009, 2021-08-23:
#### Bug description:
Using tspcontrol to suspend and resume SRT output plugin will result in invalid socket error.

#### How to reproduce:
1. tsp -v --control-port 10080 -I file <TS file path> --infinite -P regulate --pcr-synchronous -O srt --caller <SRT URL> --transtype live --messageapi --multiple --restart-delay 500
2. tspcontrol -t 10080 suspend 2
3. tspcontrol -t 10080 resume 2

#### Expected behavior:
SRT output can be resumed succesfully.

#### Errors and logs:
~~~
* control commands: received from 127.0.0.1:65427: list
* control commands: received from 127.0.0.1:65429: suspend 2
* control commands: received from 127.0.0.1:65434: resume 2
* Error: srt: error during srt_send(): Operation not supported: Invalid socket ID
~~~

#### Environment:
 - OS: macOS
 - OS version: 11.5.2
 - TSDuck full version: 3.27-2438
 - Installation type: brew install tsduck

lelegard, 2021-08-23:
Hi @wongfei2009,

I cannot reproduce your problem. I tried on macOS 11.5.2 (with TSDuck 3.28-2481 built on libsrt 1.4.3) and on Linux. There is no issue, suspending and resuming the output plugin does not raise any error.

In fact, there no reason why suspending and resuming the output plugin should have any effect on the plugin. There is one simple reason for that, suspending and resuming the output plugin _does nothing at all_ on the output plugin.

When `tsp` receives a suspend request for the output plugin, it simply ceases to call the `send()` method of the plugin. When TS packets reach the end of the plugin chain and the output plugin is suspended, `tsp` simply drops the packets.

From an output plugin standpoint, suspend/resume is transparent. The plugin is simply not called when the output is suspended.

I also tried to crash the receiver and restart it, just in case your issue would be due to something else (without any relation with a suspend/resume which was there by coincidence). But there was no issue, as long as the restart of the receiver occurs within the connection timeout of the sender.

For information, my receiver command is:
~~~
tsp -t -d -I srt -l 5000 --transtype live --messageapi -P filter --every 1000 -P debug -O drop
~~~
The idea is that a debug trace is displayed every 1000 packets, just to make sure that a stream is continuously received.

The sender command is:
~~~
tsp --control-port 5555 -v -b 1,000,000 -I null -P regulate -O srt --caller localhost:5000 --transtype live --messageapi --multiple --restart-delay 500
~~~
The sender connects to the receiver and the receiver displays traces. I suspend/resume the sender from another session and the receiver no longer displays traces during the suspend period. After resume, the traces in the receiver are back.

I also crashed the receiver and restarted it. The sender detects the disconnection and reconnects. If the receiver is not restarted without 3 seconds, then the sender fails on timeout since the default connection timeout is 3000 ms. Using `--conn-timeout 10000` in the sender, for instance, increases the allowed restart period to 10 seconds.

Could you please be more specific on your issue? 

Is your PR #851 related to this issue? I don't see how it could impact the behaviour of the suspend/resume operations since the output plugin is not called at all in these operations.

I do not think that TSDuck 3.27 vs. 3.28 makes a difference but you should use the latest version anyway. The suspend/resume behaviour of the output plugin was never changed from the beginning, so it won't change anything. If your issue is due to some other cause (restart of the receiver for instance), maybe an older version of libsrt can make a difference.

wongfei2009, 2021-08-23:
The issue I encounter is:

After I suspend the SRT output for a while (~1 min), then I resume the output using tsp control. But as soon as I do that, I will encounter the error "Error: srt: error during srt_send(): Operation not supported: Invalid socket ID" and the SRT output can't be resumed.

I highly suspect the SRT connection needs some data to keep the connection alive. If there is no data sending via SRT connection for a period of time, the SRT connection will be closed by the underlying SRT library & results in the error - invalid socket ID.

wongfei2009, 2021-08-23:
I think my PR makes a difference because the retry in SRTOutputPlugin only act on a "clean" disconnection, which is quite limited.

        // Send error.
        if (!_sock.peerDisconnected()) {
            // Actual error, not a clean disconnection from the receiver, do not retry, even with --multiple.
            return false;
        }

lelegard, 2021-08-23:
> After I suspend the SRT output for a while (~1 min), then I resume the output using tsp control. 

I just did that, on macOS, with the commands from my previous post. I resumed the output plugin after more than 5 minutes and everything continued as if nothing happened. There was no disconnection on either side. The output SRT socket was quietly waiting for more packets to send.

I can merge your plugin because the idea is not bad anyway. But I think this is not a real solution for your problem. It just hides the problem in your configuration. There is something else that may remain latent.

What is the receiver? Another instance of `tsp` or some dedicated SRT application? Does the receiver report any issue?

lelegard, 2021-08-23:
Also try to `brew upgrade`. You will get the latest versions of libsrt and TSDuck.

What is your output of `tsversion --version=all`?

lelegard, 2021-08-23:
I think I now understand the conditions under which the problem occurs. If I crash and restart the receiver while the output plugin of the sender is suspended and wait "some time" before resuming, the problem occurs.

It looks like `srt_send()` returns the error `SRT_ECONNLOST` when the peer "recently" disconnected and `SRT_EINVSOCK` when the peer disconnected "some longer time ago". The suspend/resume is here just a way to exhibit this weird behaviour.

This looks like a bug in libsrt. I will check with the SRT dev team.

In the meantime, your PR seems a valid workaround to avoid the issue. I will merge it.

lelegard, 2021-08-25:
Hi @wongfei2009 

See https://github.com/Haivision/srt/issues/2098. The SRT dev team confirms that they arbitrarily close an inactive socket after some time. I consider this as a design bug in the SRT API (they may disconnect the session but not invalidate the socket handle). So, your PR is a valid workaround for that bug.

wongfei2009, 2021-08-25:
Interesting discussion 👍

closed 2021-08-25 by wongfei2009