tsduck/tsduck #852: [FEAT] Read cell Identifier

cristianpuddu, 2021-08-24:
#### Is your feature request related to a problem?
Hi @lelegard,
Is it possible to read the cell identifier on TSSCAN? 
In Italy the 8 bits S40-$47 are used by RAI also in DVB-T, for identify the cell.
So it would be very useful to be able to receive this parameter.
Thanks
Cristian

lelegard, 2021-08-25:
> In Italy the 8 bits S40-$47 are used by RAI also in DVB-T, for identify the cell.

What do you mean by "8 bits S40-$47"? Looking at the [Italian DVB-T files you previously provided](https://tsduck.io/streams/?name=italy-sardinia-dttv), I do not see any standard DVB descriptor to signal the cell_id in the NIT.

cristianpuddu, 2021-08-25:
I refer to the TPS carrier
RAI and others operators in Italy have studied a "trick" to differentiate the locations, taking inspiration from the cell-id created for DVB-H. Some analyzer is able to receive it correctly.
This link explains in English the project of RAI : [http://www.crit.rai.it/CritPortal/progetti/?p=707](http://www.crit.rai.it/CritPortal/progetti/?p=707)
[image]
[image]

lelegard, 2021-08-25:
OK, it is in the DVB-T TPS. To my knowledge, there is no standard API to get the TPS from tuners on Linux and Windows. Maybe some proprietary application is delivered with a given tuner to get the TPS. But I do not know how to get the TPS in the general case.

cristianpuddu, 2021-08-25:
thanks @lelegard,
I just thought so. I read that the specification was born for DVB-H and then later used without any standard also for DVB-T.
With the arrival of DVB-T2 it will be easier to implement?

lelegard, 2021-08-25:
> With the arrival of DVB-T2 it will be easier to implement?

It is not a matter of DVB-T vs. DVB-T2. It is a matter of standard software API on Linux and Windows. The two operating systems provide two very different uniform API's to access tuners. The Linux API is well alive. The Windows API is more or less in an abandoned state. Operations such as tuning and receiving packets are implemented in both systems (fortunately!) Getting the signal state is not so well implemented (see #853). Getting the TPS is not implemented at all. So, even if a hardware tuner can receive the TPS and its driver can read its content, there exist no standard way to read it from an application such as TSDuck. And I do not really expect this to change. DVB-T and TPS exist for more than 20 years now.

EnoSat, 2021-11-01:
[CrazyCat](url=https://sourceforge.net/u/crazycat69/profile/
[image]
[image]
[image]
[image]
) implementoval čítanie Cell_ID do svojho [CrazyScan2](url=https://sourceforge.net/p/crazyscan/code/HEAD/tree/trunk/)