tsduck/tsduck #853: [Q] signal and quality in dBm

cristianpuddu, 2021-08-24:
I am trying to save the values that indicate the signal quality in dB to use them for example on a csv file or a chart.

I found the dvb-fe-tool --femon utility, which returns while tsduck is searching the channels, a continuous loop of data but does not report the tuning frequency. Also, the whole thing in this way is very long and complicated. This is the output:
 dvb-fe-tool --femon
`"Lock   (0x1f) Quality= Good Signal= -63.12dBm C/N= 33.61dB UCB= 0 postBER= 0 preBER= 256x10^-6 PER= 0"`
I wanted to know if with tsduck there could be an easier way to know the signal without using percentages?
Thanks
Cristian

lelegard, 2021-08-25:
dvb-fe-tool is a Linux specific tool. TSDuck tries to be generic and works indifferently on Linux and Windows. Signal strength and other statistics were badly managed on many drivers, Linux and Windows. The semantics of the returned value was unclear. Only recent versions of Linux kernel introduced a new API to report precise values (subject to cooperation from the driver). I will see what I can do.

cristianpuddu, 2021-08-25:
thanks @lelegard 
Obviously I don't expect measurements worthy of professional instruments, but be sure of a signal variation compared to other daily measurements

cristianpuddu, 2021-08-28:
I did some testing with tsp --verbose
These are values that for my area are reasonable
```
* Debug: dvb: 6 DTVProperties (statistics result):
* Debug: dvb: [0] cmd = 62 (DTV_STAT_SIGNAL_STRENGTH), count = 1, stat = {scale: 1, value: -64250}
* Debug: dvb: [1] cmd = 63 (DTV_STAT_CNR), count = 1, stat = {scale: 1, value: 31632}
* Debug: dvb: [2] cmd = 66 (DTV_STAT_POST_ERROR_BIT_COUNT), count = 1, stat = {scale: 0, value: 0}
* Debug: dvb: [3] cmd = 67 (DTV_STAT_POST_TOTAL_BIT_COUNT), count = 1, stat = {scale: 0, value: 0}
* Debug: dvb: [4] cmd = 68 (DTV_STAT_ERROR_BLOCK_COUNT), count = 1, stat = {scale: 0, value: 0}
* Debug: dvb: [5] cmd = 69 (DTV_STAT_TOTAL_BLOCK_COUNT), count = 1, stat = {scale: 0, value: 0}

* dvb: locked: yes, strength: -64.25 dB, SNR: 31.632 dB

```

cristianpuddu, 2021-09-14:
Hi @lelegard,
Can you insert in the xml file generated by ttscan the value of the signal (strength)?
For example 
* UHF channel 36 (594 MHz), locked: no, strength: -69.5 dB

and in ttscan when I use the option --save-channel: 
...
<dvbt frequency="594,000,000" modulation="64-QAM" HPFEC="3/4" LPFEC="1/2" bandwidth="8,000,000" transmission="8K" guard="1/4" hierarchy="none" inversion="off"/ **strength="-69.5dB"** >
...
Thanks

lelegard, 2021-09-15:
HI @cristianpuddu 

The channel XML file describes the permanent characteristics of the broadcast in a given area. The content of these files never changes (as long as the channel map is not updated by operators). These files can be freely exchanged by people in the same reception area or published on the net.

The signal strength, on the other hand, cab change all the time. It depends on the weather (especially for satellite reception), the quality of your antenna, downlink cable, tuner, driver, etc.

So, no, the signal strengrh has nothing to do in the channel file, sorry.

cristianpuddu, 2021-09-15:
I totally understand, your valid observations.

About DVB-T, when I do a scan with tsscan I expect to find local muxes in the results.
But this is not always true.
For example, I scanned some frequency yesterday, and surprisingly I found new muxes.
These signals were much weaker than the others and only receivable for a few hours, because they were due tropospheric propagation phenomena (https://en.wikipedia.org/wiki/Tropospheric_propagation) and they were over 600 km away from my location.
A reference to the signals would have allowed me to know right that they weren't coming from the usual location
where I get all the others.

I'll try to extrapolate the signal from stdout...

closed 2022-09-23 by lelegard