tsduck/tsduck #860: [Q] Java and merge plugin

lyncher0, 2021-08-30:
Hi @lelegard,

I'm trying to use merge plugin over Java integration. I'm able to create and start the processing pipeline but when I terminate it (tsp.abort) the process started by merge plugin (another tsp.exe) keeps running forerever.

What is the correct strategy to terminate processes started by merge when using Java access layer?
Thank you
--
lyncher

lelegard, 2021-08-31:
Hi @lyncher0 

This should not happen. I just tried on Windows (since you mention ".exe" in tsp.exe, I assume that you use Windows) and the merge process is terminated right after `tsp.abort()`. See code below.

When `tsp.abort()` is called, it terminates all plugins. When the `merge` plugin terminates, it closes the pipe from which it reads the packets to merge. This closes the standard output of the created process. The process is not killed by the merge plugin because it has no right to do so. The created process may need to do some cleanup. The created process just sees that its output is closed and it is up to this process to decide what to do. Generally, this means terminate. This is the normal way piped applications work. In the case of `tsp` (this is the application you run in the created process), closing the output, means that the output plugin terminates on error. As a consequence, all plugins are aborted. When all plugins have completed their close, the process terminates.

In your case, if the `tsp` process does not terminate, this is because some plugin is stuck somewhere during its close sequence. Maybe a packet processing plugin is stuck in a wait. Maybe the input plugin is stuck in a blocking I/O that never completes.

So, your problem is not in the parent Java application, it is in one of the plugins of the created process.

Just as a demo, I used the following code on Windows (run on Eclipse context):
~~~
import io.tsduck.AsyncReport;
import io.tsduck.Report;
import io.tsduck.TSProcessor;

public class TestMerge {
    public static void main(String[] args) throws Exception {

        AsyncReport rep = new AsyncReport(Report.Debug, false, true, 512);
        TSProcessor tsp = new TSProcessor(rep);
        tsp.bitrate = 1000000;
        tsp.input = new String[] {"null"};
        tsp.plugins = new String[][] {
            {"regulate"},
            {"merge", "tsp -b 1000000 -I null -P regulate"},
        };
        tsp.output = new String[] {"drop"};

        rep.info("Starting tsp");
        tsp.start();

        rep.info("Waiting 20 seconds to check the presence of the created process");
        Thread.sleep(20000);

        rep.info("Aborting tsp");
        tsp.abort();

        rep.info("Waiting 20 seconds to check the absence of the created process");
        Thread.sleep(20000);

        rep.info("Waiting for tsp to terminate");
        tsp.waitForTermination();

        rep.info("Terminated, cleanup");
        tsp.delete();
        rep.terminate();
        rep.delete();
    }
}
~~~
During the first wait, we can see in the task manager that the `tsp` process is created. During the second wait, after `tsp.abort()` but before cleaning up the tsp object, we can see that the `tsp` process is gone.

lyncher0, 2021-08-31:
Thanks for the reply,
Indeed the problem was not related with the Java access layer.
My input was from a UDP socket (low bitrate contents) and the input plugin was stuck in initAllBuffers (default size is 16MB). After adding the option "--input-iniital-packets 1" I'm able to start/stop without any problem.

lelegard, 2021-08-31:
If `tsp` is still in the startup phase, waiting for enough initial packets to figure out what the bitrate could be, the output plugin is not even started. So it won't realize that the output (the pipe) is closed and nothing happens. For very low bitrate streams, it is indeed necessary to reduce the buffering which is tuned by default for high performance streams. But the same problem would occur if an input stream is inactive, even long after the initial phase. No input means that all plugins are waiting for packets to process, including the output plugin. As long as the output plugin does not try to write something on its output, it will not realize that the output is closed and nothing will happen.

closed 2021-09-29 by lyncher0