tsduck/tsduck #861: [BUG] tsswitch fails to stop forked input

nimblin, 2021-09-01:
#### Bug description:
Tool tsswitch fails to switch between forked inputs `tsp -I null -P regulate --bitrate 100` with default or `--delayed-switch` modes as it can not stop the previous input.

#### How to reproduce:
> tsswitch --debug --remote 8888 --event-command "echo ==== EVENT" -I fork "tsp -I null -P regulate --bitrate 100" -I fork "tsp -I null -P regulate --bitrate 200" -O drop

Or with  `--delayed-switch` option.

#### Errors and logs:
~~~
$ tsswitch --debug --remote 8888 --event-command "echo ==== EVENT" -I fork "tsp -I null -P regulate --bitrate 100" -I fork "tsp -I null -P regulate --bitrate 200" -O drop
[...]
==== EVENT newinput 0 0
* received command "next", from 127.0.0.1:60601 (5 bytes)
* Debug: switch input 0 to 1
* Debug: enqueue action SUSPEND_TIMEOUT, 0, false
* Debug: enqueue action ABORT_INPUT, 0, false
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action START, 1, true
* Debug: enqueue action WAIT_STARTED, 1, false
* Debug: enqueue action RESTART_TIMEOUT, 0, false
* Debug: executing action SUSPEND_TIMEOUT, 0, false
* Debug: executing action ABORT_INPUT, 0, false
* Debug: executing action STOP, 0, false
* Debug: fork[0]: received stop request
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: not ready, waiting: WAIT_STOPPED, 0, false
* received command "next", from 127.0.0.1:42238 (5 bytes)
~~~

~~~
$ tsswitch --debug --remote 8888 --event-command "echo ==== EVENT" --delayed-switch -I fork "tsp -I null -P regulate --bitrate 100" -I fork "tsp -I null -P regulate --bitrate 200" -O drop
[...]
==== EVENT newinput 0 0
* received command "next", from 127.0.0.1:54047 (5 bytes)
* Debug: switch input 0 to 1
* Debug: enqueue action SUSPEND_TIMEOUT, 0, false
* Debug: enqueue action START, 1, false
* Debug: enqueue action WAIT_INPUT, 1, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action NOTIF_CURRENT, 1, true
* Debug: enqueue action RESTART_TIMEOUT, 0, false
* Debug: enqueue action ABORT_INPUT, 0, false
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: executing action SUSPEND_TIMEOUT, 0, false
* Debug: executing action START, 1, false
* Debug: fork[1]: received start request, current: false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: fork[1]: starting input plugin
* Debug: fork[1]: creating process "tsp -I null -P regulate --bitrate 200"
* Debug: fork[1]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 1, true
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
~~~

#### Environment:
 - OS:  Linux
 - OS version: Ubuntu 20.04.2
 - TSDuck full version: 3.28-2449
 - Installation type: rebuilt binary with `make install`

#### Additional information:
Switch mode `--fast-switch` works fine

lelegard, 2021-09-01:
Hi @nimblin,

Thanks for reporting this. This bug was a nasty one. Well done.

I fixed it. But there are in fact two things here.

First, there is a problem with the provided command, although I suspect that this command was just provided for the sake of example. The plugin `-P regulate --bitrate 100` in the forked process means that you regulate at 100 bits per seconds, which is super-low. Considering that `tsswitch` reads `--max-input-packets` (default: 128 packets) per transaction, it takes 128 * 188 * 8 / 100 = 1952 seconds (32 minutes) to pass. With the second input using `-P regulate --bitrate 200`, one transaction lasts 16 minutes. Since `--delayed-switch` means that the actual switch occurs after some input is received from the next plugin, this means that the actual switch will occur only 16 or 32 minutes after the switch command.

That being said, even using more realistic bitrates (say 100 and 200 kb/s instead of 100 and 200 b/s), there is still another Unix-specific issue (Linux and macOS). I assume that the super-low bitrates were just an error in the sample command and that you ran into this actual issue with more realistic bitrates. I just fixed it but it was a really nasty one...

If you are interested in the gory details, the problem was in the handling of the file descriptors of the pipes to the forked command. In short, the flag `FD_CLOEXEC` was not set on the file descriptors. With `--delayed-switch`, the second command is started first, while the first command is still running. Here, the file descriptors to the pipe of the first command were erroneously inherited in the second process. When the switch was completed, `tsswitch` closed the pipe to the first process. Normally, this sends an error ("no more reader on that pipe") to the output pipe of the first command, which then terminates. However, because the second process inherited the pipe file descriptors of the first process, the second process still had a read file descriptor on the pipe of the first process, unused but still open. As a consequence, there was no error on the output pipe of the first process (there was still a opened reader on the pipe) and the first process never terminated. Since the plugin `fork` synchronously terminates on the process, when the process terminates, the first input never terminated. Note that this did not prevent the switch to occur on the second input plugin, as soon as some input was produced. But since the first plugin was still stuck in its close operation, it was no longer possible to switch back to it later.

Without `--delayed-switch`, the first plugin is stopped first (closing the pipe and its file descriptors) and then the second process is created. Therefore, there is no unused pending reader on the output pipe of the forked command.

nimblin, 2021-09-01:
The fix seems to work fine.  I can see how that must have been some nasty debugging, thanks for the speedy and detailed fix/explanation - you're superhuman :laughing:.

>  I assume that the super-low bitrates were just an error in the sample command and that you ran into this actual issue with more realistic bitrates.

Correct. If you're interested, I use `tsp -I null -P regulate --bitrate 100000` sometimes while testing because `tsp -I null` alone uses too much CPU. Would you say my approach is appropriate?

lelegard, 2021-09-01:
> that must have been some nasty debugging

Yes, in the sense of careful and thorough analysis of the symptoms and correlations with possible causes. But no in the sense of using a debugger. I almost never need to use a debugger. Recompiling in debug mode and stepping into the code usually takes longer that simply observing and cogitating.

In that specific case, creating processes means using `ps faxw` to get an idea of the process hierarchy. Then, I was puzzled about the remaining presence of the first process after the switch (that was consistent with the absence of debug trace reporting the completion of the `WAIT_PROCESS 0`). Finally, `ls -l /proc/PID/fd` on each PID, the parent `tsswitch` and the two children `tsp`, showed that the second child process had a exotic file descriptor 5 on the same pipe as the the output of the first child while the parent had only a file descriptor on the output of the second one. The immediate cause for the blocked situation became clear. Thinking about the differences in the order of operations with and without `--delayed-switch`, it became obvious that the file descriptor was erroneously inherited, which immediately led to `FD_CLOEXEC` (after `man fcntl` to be honest). As you can see, no debugging per se, just observation...

> I use tsp -I null -P regulate --bitrate 100000 sometimes while testing because tsp -I null alone uses too much CPU. Would you say my approach is appropriate?

Absolutely. I use the same kind of trick when I need to generate a flow of packets when the content is not important.

nimblin, 2021-09-01:
Thanks @lelegard , I appreciate the insight. Closing this issue now.

closed 2021-09-01 by nimblin