tsduck/tsduck #862: [FEAT] PSI plugin - AC3 descriptor

enricovittorini, 2021-09-02:
#### Proposed solution:
Include AC3 descriptor in PMT according to [ETSI EN 300 468 V1.16.1 Anned D3](https://www.etsi.org/deliver/etsi_en/300400_300499/300468/01.16.01_60/en_300468v011601p.pdf)

lelegard, 2021-09-02:
Hi @enricovittorini 

What do you mean by _"include AC3 descriptor in PMT"_?
This descriptor is supported by TSDuck. See the user's guide, section D.8.2.

Do you have a sample PMT containing an AC3 descriptor which is not correctly formatted by the `psi` plugin?

lelegard, 2021-09-02:
Thanks for the stream. The full PMT is:
~~~
02 B0 1C 00 01 C5 00 00 FF FF F0 00 06 EA 1B F0 0A 0A 04 69 74 61 00 6A 02 80 44 DC 85 C2 6C
~~~
The AC3 descriptor is:
~~~
6A 02 80 44
~~~
The way TSDuck formats it is:
~~~
    - Descriptor 1: AC-3 (0x6A, 106), 2 bytes
      Component type: 0x44 (AC-3, full, complete main, multichannel > 2)
~~~
XML:
~~~
      <DVB_AC3_descriptor component_type="0x44"/>
~~~
Everything looks normal to me.
Could you please explain what do you expect from this issue?

enricovittorini, 2021-09-02:
i think  the following info are missing.
It is reported only the  "if" condition of the component_type.
[image]

lelegard, 2021-09-02:
OK, I see, you mention fields such as `component_type_flag`, `bsid_flag` and other `_flag` fields.

This is a misinterpretation of the role of these fields. The `_flag` fields do not carry any semantics regarding AC-3. In the binary representation of the descriptor, they are just syntactic indicators of the presence of the actual meaningful fields: `component_type`, `bsid`, etc.

XML and binary data have there own syntax requirements. XML is not a binary equivalent of the descriptor. It is a non-ambiguous logical representation of the binary descriptor.

In XML, the syntactic presence of a `component_type` or `bsid` attribute is simply assessed by ... its presence in the XML structure.

In XML, a `component_type_flag` attribute would be useless, redundant and counter-productive (risk of inconsistency). In short, the presence of a `component_type` XML attribute means that the binary flag `component_type_flag` is set to 1 and the binary `component_type` field is present and contains the attribute value. Similarly, the absence of a `component_type` XML attribute means that the binary flag `component_type_flag` is set to 0 and there is no binary `component_type` field.

This principle is consistently applied to all tables and descriptors. A binary `_flag` field which exists only to indicate the presence of a corresponding binary field carrying actual information never has an XML `_flag` attribute, nowhere.

enricovittorini, 2021-09-02:
@lelegard , the meaning of those flags are well understood. But I think it is important to have a view of them: what if the flag 
bsid_flag is set to 1 by the upstream device, but there is no data then (or viceversa)?
Generally speaking, the probes I usually see, show the full descriptor syntax.

enricovittorini, 2021-09-02:
@lelegard  

I see it in another way:

Let's say that:
componeny_flag =1
bsid_flag = 1

I should then get two bytes:
- X for component_type
- Y for bsid

if I get only one byte, let' say Z, this should be linked to component_type regardless if it is the byte of the bsid. And this is because the syntax of the descriptor has an order of reading and interpreting the bytes, and a proble should represent what is transmitted (then maybe with no-sense of the value).
Then eventually, since the second byte is not present, but signaled as "1" in the flags, the probe should report "missing"

lelegard, 2021-09-02:
There is a misunderstanding on the primary role of each tool.

You may expect that from a probe, the purpose of which is to analyse defects on a network (although I do not completely agree on the arbitrary interpretation of an incorrect data area).

But, again, XML files serve a different purpose here. They are not a probe, they are not used to analyse the defect of a network. They are used to build or modify the (valid) structure of a network. Their primary role is engineering tools, not analysis tools. What is decompiled as XML shall be recompiled identically, with the same semantics as the original (even if not always bit-to-bit identical). The only usable XML representation of an incorrect descriptor is a `<generic_descriptor>` structure. It contains raw and uninterpreted data. If you want to detect incorrect descriptors in an XML table, look for `<generic_descriptor>` structures. The descriptor id is in the `tag` attribute.

EricBerendsen, 2021-09-13:
@enricovittorini , just a small warning, do not rely on DVB Inspector to show you correct result when "the flag
bsid_flag is set to 1 by the upstream device, but there is no data then (or viceversa)?"

This issue triggered me to test how DVB inspector would interpret missing bsid data. while bsid_flag is set. DVB Inspector does not check the descriptor length while reading bytes, so it reads past the end of the descriptor into whatever happens to be there (in this case it reads the CRC32 bytes)

In this example I forced the bsid_flag and mainid_flag to 1, while the length of the descriptor was still 2 (only flags and component_type present):

[image]

There is no visible indication something is wrong.

This is a structural bug in DVB Inspector, and I created a ticket for it. So thank you for helping me find a problem.

@lelegard 's code is much more robust, it checks for size for each field, and falls back to a generic descriptor. So in his code you see a generic descriptor, which is a clear sign something is wrong. In my code everything looks OK, which is much more dangerous.

I like his handling of handling inconsistent/wrong data; fall back to the raw data and do not try to interpret nonsense.  I tried to do the same, but missed this one.