tsduck/tsduck #864: [BUG] Building on Centos 7 fails

akkermansadriaan, 2021-09-03:
#### Bug description:
Building on Centos 7 fails with current master branch.

#### How to reproduce:
Checkout current master branch (https://github.com/tsduck/tsduck/commit/f305faf64a2cd8dd2fd88e9cf85b14bbc85ade51)
Follow build instructions for Linux:
- Get prerequisites: `build/install-prerequisites.sh`
- Build: `make NOTEST=1 NOCURL=1`

#### Expected behavior:
Build is successful. 

#### Errors and logs:
Full log in [output.log](https://github.com/tsduck/tsduck/files/7106616/output.log). Error extract:
~~~
In file included from /root/tsduck/src/libtsduck/base/types/tsFraction.h:334:0,
                 from dtv/private/tsDektec.h:37,
                 from dtv/private/tsDektecArgsUtils.h:42,
                 from dtv/private/tsDektecArgsUtils.cpp:30:
/root/tsduck/src/libtsduck/base/types/tsFractionTemplate.h:46:31: error: specialization of ‘template<class INT_T, typename std::enable_if<std::is_integral<_Tp>::value, int>::type <anonymous> > template<bool is_signed, bool dummy> struct ts::Fraction<INT_T, <anonymous> >::SignWrapper’ in different namespace [-fpermissive]
 struct ts::Fraction<INT_T,N>::SignWrapper<false, dummy> {
                               ^
In file included from dtv/private/tsDektec.h:37:0,
                 from dtv/private/tsDektecArgsUtils.h:42,
                 from dtv/private/tsDektecArgsUtils.cpp:30:
/root/tsduck/src/libtsduck/base/types/tsFraction.h:76:16: error:   from definition of ‘template<class INT_T, typename std::enable_if<std::is_integral<_Tp>::value, int>::type <anonymous> > template<bool is_signed, bool dummy> struct ts::Fraction<INT_T, <anonymous> >::SignWrapper’ [-fpermissive]
         struct SignWrapper {
                ^
In file included from /root/tsduck/src/libtsduck/base/types/tsFraction.h:334:0,
                 from dtv/private/tsDektec.h:37,
                 from dtv/private/tsDektecArgsUtils.h:42,
                 from dtv/private/tsDektecArgsUtils.cpp:30:
/root/tsduck/src/libtsduck/base/types/tsFractionTemplate.h:53:31: error: specialization of ‘template<class INT_T, typename std::enable_if<std::is_integral<_Tp>::value, int>::type <anonymous> > template<bool is_signed, bool dummy> struct ts::Fraction<INT_T, <anonymous> >::SignWrapper’ in different namespace [-fpermissive]
 struct ts::Fraction<INT_T,N>::SignWrapper<true, dummy> {
                               ^
In file included from dtv/private/tsDektec.h:37:0,
                 from dtv/private/tsDektecArgsUtils.h:42,
                 from dtv/private/tsDektecArgsUtils.cpp:30:
/root/tsduck/src/libtsduck/base/types/tsFraction.h:76:16: error:   from definition of ‘template<class INT_T, typename std::enable_if<std::is_integral<_Tp>::value, int>::type <anonymous> > template<bool is_signed, bool dummy> struct ts::Fraction<INT_T, <anonymous> >::SignWrapper’ [-fpermissive]
         struct SignWrapper {
                ^
make[2]: *** [/root/tsduck/bin/release-x86_64-localhost/objs-libtsduck/tsDektecArgsUtils.o] Error 1
make[1]: *** [default] Error 2
make: *** [default] Error 2
~~~

#### Environment:
 - OS: Linux
 - OS version: CentOS 7

lelegard, 2021-09-03:
This looks like a gcc issue. Which version of gcc is this?
I only have CentOS 8 VM's but with a few older versions og gcc side by side.

lelegard, 2021-09-04:
Hi, 

Looking at the repo sites for CentOS 7, it seems that the gcc version is 4.8.5. This is a completely outdated version of the compiler. The current version is 11.x. Fedora has gcc 11, Ubuntu has gcc 10 and CentOS 8, the "oldest" distro I have, has gcc 8.

I tried various older versions of gcc I rebuilt some time ago on CentOS. I can reproduce your compilation issue with gcc 4.8.5 (presumably the same as yours on CentOS 7) and gcc 6.3.0. With gcc 7.5.0 and later, the compilation is fine.

It seems to be a bug in old versions of gcc. I will see if I can find some workaround in the code to bypass that compiler bug.

lelegard, 2021-09-04:
Hi @akkermansadriaan 
A workaround has been pushed. Please check that it compiles in your environment.

closed 2021-10-12 by lelegard