tsduck/tsduck #865: [Q] psi plugin behavior

enricovittorini, 2021-09-05:
Would like to clarify the following use case to understand if this is a bug or not

When using psi -a plugin, let's say there is a service with PMT version 1.
The first time all PSI/SI table of the TS are reported.

Remove the service:
Update of PAT and SDT are correctly reported.

recreate the same service previously deleted. This will appear as a new service in the TS with version 1 (same as before).
This is not logged from PSI plugin but once again only PAT and SDT are reported.

I think the PMT should be added too as this is a new service (identical to the one deleted before).
My understanding is that psi plugin reports only when the version of a table change, but does not cover this corner case.

lelegard, 2021-09-05:
Hi @enricovittorini 

This is a corner case indeed.

A service being removed and back again in the PAT can happen. The use case is a service being suspended or _"not running"_, _"pausing"_ , _"off-air"_ in DVB parlance. If the same service (same service id, same PMT content on same PMT PID) reappears again later, this is not a "new" service. It is an existing service which resumes.

In your case, the core question is: does the structure of the service change after resuming? If the PMT section is slightly different, then the version must be changed (this is a basic MPEG rule). But if the PMT section is exactly identical as previously, same service id, same content, same version, then this is correct. If a receiver is constantly tuned on the service, before and after the suspension (and the user falling asleep on the couch during the back screen of the suspension), then there is no need for the receiver to get the PMT again. The STB or TV continues to demux the same audio and video PID's as previously.

When the "receiver" is the TSDuck `psi` plugin, well, this is a matter of specification... This plugin is supposed to report the structure of the TS. If an existing service is suspended and back with the same identical structure, there is no new information about the structure of that service and there is no need to output the same PMT again. The new information is that 1) the service disappeared and 2) it reappeared. These new information are reported with the new versions of the PAT. But, there is no new information in the structure of the service.

So, I would say that the current behaviour has its own logic. But I agree we may argue about it.

enricovittorini, 2021-09-05:
@lelegard 
I was talking specifically about a new service and not about a service "Stopped"

Immagine the following scenario

Setvice1 with version 1 is deleted from the mux 

Then service 1 is created with same service ID and  PMT  PID as previous service. Its version, being a new service, would be 1. 

The priblem is that something is considered "new" only if table version change. But there are cases where this is not enough. 
The new service could have more audio and data PID compared to old deleted one, and if I understood correctly, this would not be elaborated by the plugin since it's version is still one. Will try this case and let you know...

To me in general, any PAT change should trigger PMTs check and compare vs previous PMT knowledge.

lelegard, 2021-09-05:
> I was talking specifically about a new service and not about a service "Stopped"
> [...] Then service 1 is created with same service ID and PMT PID as previous service. Its version, being a new service, would be 1.

You do not have the right to do that. It violates the basic rule of PSI as defined is IS0 13818-1. On a given PID, when two consecutive sections of the same _subtable_ (ie. same table id and same table id extension) have the same version, they must have the same content. This means that when the content changes, the version must change.

If your new service has a different structure, first it is a bad idea to reuse the same service id (same table id extension). Two different entities should have different ids. And if you decide to reuse the same service id anyway, you __MUST__ use a different version number. If you do not follow this rule, anything can happen on a receiver.

Moreover, the statement _"Its version, being a new service, would be 1"_ is incorrect. The version field of a table is not a version in the sense of a product version, starting at 1 and then incrementing. It is a trigger to announce something different. You can start at any value, not specifically 1, and you are not obliged to increment. You can use any sequence, 3, 23, 30, 12, etc., at long as the value changes when the content changes. There are 31 possible values, not just a simple 0, 1 trigger in case the changes are rapid and you may miss more than one on the receiver.

enricovittorini, 2021-09-05:
@lelegard 
> Then service 1 is created with same service ID and PMT PID as previous service. Its version, being a new service, would be 1
This is a case that can happen to someone operating a mux.

> You do not have the right to do that. It violates the basic rule of PSI as defined is IS0 13818-1. On a given PID, when two consecutive sections of the same subtable (ie. same table id and same [...]

I do not think I am violating anything here: I would agree with you if I was doing changes on  a service that remains on-air during the changes. But it was not the corner case described.

> If your new service has a different structure, first it is a bad idea to reuse the same service id (same table id extension). Two different entities should have different ids. And if you decide to [...]

This is a more frequent practice than what you can imagine table versions are usually manages automatically but the multiplexers. And note: when i remove the service, the reference on the PMT is removed by by the PAT too... STBs take this into consideration.

> Moreover, the statement "Its version, being a new service, would be 1" is incorrect. The version field of a table is not a version in the sense of a product version, starting at 1 and then [...]

Agree on this, I was just describing the specific corner case.

My point Thierry is that if  when the service at T1 is removed, psi plugin seems to keep a reference of it for what it was at T0. When at T2 the service is recreated it is not managed as something new that happened in the TS. When the PMT is not referenced in the PAT, "psi plugin should remove any reference to it form its memory.

In general, it would be great to add a check on each table even when the "version" does not change. This would help to analyze issue caused by buggy multiplexers

lelegard, 2021-09-05:
> I do not think I am violating anything here: I would agree with you if I was doing changes on a service that remains on-air during the changes. But it was not the corner case described.

No way. Just looking at what is going on on that specific PID, this is a violation of one of the most basic rules of MPEG PSI: same PID + same adjacent subtable + same version = same content, regardless of the table id, PMT or whatever. Full stop. No consideration on the chronology of changes in the PAT won't change this. This is a lower-level rule.

If this is your network, feel free to do whatever you want. But some day, if some new model of receiver does not handle that correctly, you may have a hard time convincing the manufacturer to change its DVB firmware.

But I am just the author of TSDuck and I am only concerned about it. For the sake of robustness, I will change the PSI plugin to reset PMT PID's which are removed from the PAT, with the consequence of restarting from scratch on that PID if comes back later in the PAT. This is not the first time TSDuck adds features to handle non-standard cases. As long as it does not break or endanger the structure of the code, this is fine with me.

enricovittorini, 2021-09-05:
Thierry, 
thanks for your answer: always interesting exchanging point of views...

To further clarify:
> No way. Just looking at what is going on on that specific PID, this is a violation of one of the most basic rules of MPEG PSI: same PID + same adjacent subtable + same version = same content, [...]

No violation of any standard or recommendation here: when a service is removed from a TS, It does not exist from any point of view and in any point in time (in the present and in the past). STB, probes, TV, analyzer, tsduck have to forget it. 
When a new service is added (even if using same SID, PID or PMT PID of previous existing service),  this is a new event in the TS and as such "shall" be interpreted by any STB, probe, analyzer etc.

> If this is your network, feel free to do whatever you want. But some day, if some new model of receiver does not handle that correctly, you may have a hard time convincing the manufacturer to change its DVB firmware.

Full of sh***y hardware and even worst firmware out there. but this should not prevent the analysis to follow the rules.

.

lelegard, 2021-09-05:
I just pushed a small update to handle that (invalid!) case. I haven't tested it on the specific case since I do not have such a stream. Please test it. You may close the issue if it works.

enricovittorini, 2021-09-05:
thanks @lelegard ,
will try the fix for this case.

here it is a stream with for your test.
[image]

[new_sid_pmt.zip](https://github.com/tsduck/tsduck/files/7112241/new_sid_pmt.zip)

lelegard, 2021-09-08:
I confirm that the PMT is displayed again after the service goes out and later back in the PAT, even though the PMT service id and version remain identical.

closed 2021-11-13 by lelegard