tsduck/tsduck #866: [Q] tspcontrol with no running input plugin

enricovittorini, 2021-09-07:
When input plugin is not running (for example ip input plugin with no source received), tsp control command fails.

```
tspcontrol -t 127.0.0.1:7001 list
tspcontrol: error connecting socket: No connection could be made because the target machine actively refused it.
```

lelegard, 2021-09-07:
Hi @enricovittorini 

This is the same issue as #843. There is a required startup phase during which `tsp` fills a part of its global buffer (half of it by default). As long as this startup phase is not completed, `tsp` is not really started: the output plugin is not started, the control port is not open, etc. You must receive at least one input packet (and use `--initial-input-packets 1`) to start.

This is arguably a limitation, `tsp` was designed with active stream processing in mind.

enricovittorini, 2021-09-09:
@lelegard understood.
would it be possible  to implement an option to tsp that send 1 fake packet to the input plugin  so that it starts in any case and could be controlled remotely?

lelegard, 2021-09-09:
> would it be possible to implement an option to tsp that send 1 fake packet to the input plugin so that it starts in any case and could be controlled remotely?

With the latest state of the repo, `--add-start-stuffing 1 ----initial-input-packets 1` does that. This adds one initial input artificial null packet and the initial buffer load is precisely one packet.

These options have existed for a long time. However, before a recent modification, partly to address this point, the initial buffer load needed at least some packets from the input plugin. The initial artificial stuffing was not sufficient. Some kind of "proof of life" from the input plugin was needed by `tsp` before starting.

This "proof of life" is still necessary with the default parameters because many packet processing plugins work much better when they have a valid evaluation of the bitrate from the beginning. The difference is that, now, this kind of "extreme tuning" allows to start with an initially inactive input plugin. The price to pay is that packet processing plugins start without any valid information on the stream and may need some time to adapt. So, it really depends on what you want to do.

closed 2021-11-13 by lelegard