tsduck/tsduck #867: [Q] analyse plugin - bitrate evaluation

enricovittorini, 2021-09-10:
version: 3.28-2520

When I use the analyze plugin with ip input:
`tsp -I ip xxxx:yyy -P analyze  -i 3 --json-line  -O drop`

I see a  problem when the bitrate of the TS at the input change:

the ts.pcr-bitrate is updaed almost immediately. and this is fine
the ts.bitrate does not update, or to better say, seems to take a very long time
This is true also for the pids.xx.bitrate: as there is not a pcr_based evalaution,

lelegard, 2021-09-10:
This TS analysis module is used in the command `tsanalyze` and in the plugin `analyze`.

Two sources of bitrate evaluations are used: the "user bitrate" and the "PCR bitrate". The former is optional and is provided from outside the TS analysis module. The latter is evaluated from inside the TS analysis module, based on the PCR's from all PID's.

The "bitrate" is the "user bitrate" if specified, the "PCR bitrate" otherwise.

Depending on the context, the "user bitrate" or the "PCR bitrate" can be more precise or undefined. It depends. Only the user knows which one is better.

In the command `tsanalyze`, the "user bitrate" is given by the `--bitrate` option. So, if unspecified, only the "PCR bitrate" is used.

In the context of `tsp`, the plugin `analyze` receives the "user bitrate" from `tsp`. This is the bitrate value which is propagated from plugins to plugins (and which can be modified by plugins). As documented, its preferred value is the option `tsp --bitrate` (if the user decides, it is considered the right value). If omitted, this is the value which is reported by the input plugin, if supported by the plugin. Input plugins with a hardware input (ASI, demodulators) have a precise evaluation of the bitrate and this value is considered more precise than any PCR-based evaluation. If the input plugin has no intrinsic input bitrate (this is the case for all network input plugins), then `tsp` evaluates the bitrate from the PCR's at the input plugin level. The first evaluation is taken from the initial buffer load. Then from time to time, the evaluation is adjusted, based on _all packets from the beginning_.

In your case, if your input is intermittent or irregular and you take a snapshot of the analysis every 3 seconds, only the PCR-based bitrate is significant because it is evaluated on each 3-second period. The "user bitrate" is some average from the beginning. But only you can decide to use the "PCR-based" bitrate, not the "user bitrate" because you know the context while the `analyze` plugin doesn't.

Anas92230, 2021-12-08:
Hello,
I use similar command with -P until -s 60 but the result is very different when i use bitrate-monitor(which is conform with reality).
My stream is in CBR. What's the good way to get the right bitrate for all component in one pass?Otherwise I have to get firstly json result with -P analyse after that run bitrate monitor for all pid and put result on the json?

thanks a lot

lelegard, 2021-12-08:
The plugins `bitrate_monitor` and `analyze` measure two different things.
- `bitrate_monitor` evaluates the actual bitrate of a _live stream only_, based on the system clock, Moreover, the bitrate is measured over successive intervals, to monitor the variations. Even if the TS is CBR, each individual PID can be VBR (video PID's typically).
- `analyze` gives an overall _average_ of the TS and PID's, all over the measurement period. It can work on files or live streams equally, the system clock is not relevant.

In short, `bitrate_monitor` reports instant bitrates and `analyze` reports global average bitrates. This is of course different.

I have made a modification in the way the `analyze` plugin works. I added a "level of confidence" for each bitrate. When the "user bitrate" (see the explanation on my previous post) is not a hardware device, a system clock or a user-specified value, the bitrates are always average values based on PCR all over the stream.

yaneti, 2022-01-08:
Here is my little experiment with a live multicast stream, where `analyze` reports very very different things from `bitrate_monitor` for the same pid for roughly the same time and same stream. 
```
# timeout 20 tsp --realtime -I ip -r --default-interface 224.10.10.10:5000 -P analyze -i4 --normalized -O drop | grep pid:pid=4144
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=1694870:bitrate204=1839114:packets=15850:clear=15850:scrambled=0:invalidscrambling=0:af=4270:pcr=4205:discontinuities=0:duplicated=0:pes=83:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=2239593:bitrate204=2430196:packets=15284:clear=15284:scrambled=0:invalidscrambling=0:af=4117:pcr=4051:discontinuities=0:duplicated=0:pes=83:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=2217137:bitrate204=2405830:packets=14991:clear=14991:scrambled=0:invalidscrambling=0:af=4195:pcr=4127:discontinuities=0:duplicated=0:pes=84:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=2558363:bitrate204=2776096:packets=15527:clear=15527:scrambled=0:invalidscrambling=0:af=4230:pcr=4159:discontinuities=0:duplicated=0:pes=83:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
* tsp: user interrupt, terminating...
Terminated
# timeoute 20 tsp --realtime  -I ip -r --default-interface 224.10.10.10:5000 -P bitrate_monitor -p 4 --pid 4144 -O drop 
-bash: timeoute: command not found
[root@tvnano ~]# timeout 20 tsp --realtime  -I ip -r --default-interface 224.156.69.100:5000 -P bitrate_monitor -p 4 --pid 4144 -O drop 
* bitrate_monitor: 2022/01/08 20:32:12, PID 0x1030 (4144) bitrate: 5,924,858 bits/s
* bitrate_monitor: 2022/01/08 20:32:16, PID 0x1030 (4144) bitrate: 5,707,379 bits/s
* bitrate_monitor: 2022/01/08 20:32:20, PID 0x1030 (4144) bitrate: 5,768,742 bits/s
* tsp: user interrupt, terminating...
```

yaneti, 2022-01-08:
So I tried setting the "user" bitrate for the same stream (which is heavily null padded CBR that  I know is around 16Mbps) and the `analyze` result seems more reasonable now. 
```
#  tsp --bitrate 16000000 --realtime -I ip -r --default-interface 224.10.10.10:5000 -P analyze -i4 --normalized -O drop | grep pid:pid=4144
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=5591752:bitrate204=6067646:packets=14779:clear=14779:scrambled=0:invalidscrambling=0:af=4146:pcr=4080:discontinuities=0:duplicated=0:pes=82:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=5782929:bitrate204=6275093:packets=15375:clear=15375:scrambled=0:invalidscrambling=0:af=4227:pcr=4164:discontinuities=0:duplicated=0:pes=84:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
pid:pid=4144:access=clear:streamid=224:video:servcount=1:servlist=100:bitrate=5687236:bitrate204=6171256:packets=15138:clear=15138:scrambled=0:invalidscrambling=0:af=4163:pcr=4097:discontinuities=0:duplicated=0:pes=83:invalidpesprefix=0:description=AVC video (1920x1080, baseline profile, level 4.2, 4:2:0)
^C* tsp: user interrupt, terminating...

```
This still looks like something `analyze` is not getting right for the case where the bitrate is not always known.

closed 2022-09-23 by lelegard