tsduck/tsduck #868: Error processor when run sample Python-scripts

dvbshka, 2021-09-10:
Hi! when i try to run any scripts from the folder /tsduck/sample/sample-python/ then the script stop with error:
File "/usr/share/tsduck/python/tsduck.py", line 1416, in start
raise TSPStartError("Error starting TS processor").
I tried any system Debian 10, Ubuntu 19, but all the same.
TSDuck version 3.28-2520.
Thanks for the answer.

dvbshka, 2021-09-10:
[https://dvb.by/photo_2021-09-10_11-28-46.jpg](url)
Python version 3.8.10

lelegard, 2021-09-10:
Hi @dvbshka 

These scripts are _sample_ scripts. They are here to provide guidelines for the Python programmer. Did you read them and check what they do? For example, the script `sample-analyze-ts.py` that you use on your picture, specifies the input plugin `file` and specify the parameter `file.ts`:
~~~
tsp.input = ['file', 'file.ts']
tsp.plugins = [ ['analyze', '--json-line=' + json_marker] ]
tsp.output = ['drop']
~~~
As you see, this is just an _example_. Do you have a TS file named `file.ts` in your current directory? If yes, the script will probably work. If not, the ugly Python error reporting will blow up with its usual useless traceback. 

BTW, it would have been more practical to post a copy/paste of your command and its output rather than a truncated photo of your screen.

closed 2021-11-13 by lelegard