tsduck/tsduck #870: [Q] Help for optimisation eitinject with a lot of eit

antoineagscom, 2021-09-15:
Hi,

I have a small problem with my eit injection process.

I set my TVs with unique identifiers 
ex:
For my TV 1 ==> pat/nit/sdt/eit : service_id = 0x0001 = Channel Fr2 TV1
For my TV 100 ==> pat/nit/sdt/eit : service_id = 0x0100 = Channel Fr2 TV100
....
-- eit_x.xml
<EIT actual="true" current="true" last_table_id="0x4E" original_network_id="0x20FA" service_id="0x0001" transport_stream_id="0x0001" type="pf" version="14">
		<event event_id="0x0080" start_time="2021-09-15 08:45:00" duration="00:30:00" running_status="running" CA_mode="false">
                </event>
</EIT>
<EIT actual="true" current="true" last_table_id="0x4E" original_network_id="0x20FA" service_id="0x0100" transport_stream_id="0x0001" type="pf" version="14">
		<event event_id="0x0081" start_time="2021-09-15 08:45:00" duration="00:30:00" running_status="running" CA_mode="false">
                </event>
</EIT>

I inject my files via the "eitinject" plugin:

> tsp -I dvb XXX -P eitinject -f "eit\eit_x.xml" --bitrate 10000 --terrestrial -O dektec XXX

The problem is that I have to multiply the number of service_id in the eit_.xml with a lot of service_id.
I cannot  using a  "pmt_pid" for replace service_id.
How can I optimize to have only one service_id common to all TV in order to avoid duplicate ?

antoineagscom, 2021-09-15:
An example of eit_x.xml

[image]

lelegard, 2021-09-15:
Hi @antoineagscom 

By "TV", you mean "service", right?

> The problem is that I have to multiply the number of service_id in the eit_.xml with a lot of service_id.
> It's not working with most data (> 200 services_id)

Could you be more precise? What is the problem exactly? What would you expect?

If you have many events, you have to list them all. If you mean that repeating the `<EIT ...>` structure for each event is a pain, then you can group multiple events _for the same service_ in one single `<EIT ...>` structure. As indicated in the user's guide, this plugin just needs a database of preformatted events. The `<EIT ...>` structure is just a convenient container for events, nothing more. All events will be extracted from those fake EIT's and spread over real EIT's according to the DVB EIT broadcast rules.

> I cannot using a "pmt_pid" for replace service_id.

????

> How can I optimize to have only one service_id common to all TV in order to avoid duplicate ?

If a "TV" is a service, you cannot have the same service_id on distinct "TV".

antoineagscom, 2021-09-16:
Hi,

Exactly TV is a service.

I have multiplex R1 in input : tsp -I dvb freq MY_FREQ_R1 ...

EIT , TDT/TOT are present in original stream.

```` 
INPUT ANALYZE
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 452, PID's: 6 (clear: 6, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      111,721 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      111,721 b/s  |
|  0x0000  PAT .......................................... C       15,077 b/s  |
|  0x0001  CAT .......................................... C        3,213 b/s  |
|  0x0010  NIT .......................................... C        2,966 b/s  |
|  0x0011  SDT/BAT ...................................... C        1,483 b/s  |
|  0x0012  EIT .......................................... C       88,734 b/s  |
|  0x0014  TDT/TOT ...................................... C          247 b/s  |
|=============================================================================|
|  Service: 0x0101 (257), TS: 0x0001 (1), Original Netw: 0x20FA (8442)        |
|  Service name: France 2, provider: GR1 A                                    |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 23,296, PID's: 7 (clear: 7, scrambled: 0)                      |
|  PMT PID: 0x006E (110), PCR PID: 0x0078 (120)                               |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C    5,758,111 b/s  |
|  0x006E  PMT .......................................... C        7,415 b/s  |
|  0x0078  AVC video (1920x1080, high profile, level 4.0  C    5,338,167 b/s  |
|  0x0082  E-AC-3 Audio (fre, E-AC-3, stereo (L,R), @48,  C      132,978 b/s  |
|  0x0083  E-AC-3 Audio (qad, E-AC-3, stereo (L,R), @48,  C      133,225 b/s  |
|  0x0084  E-AC-3 Audio (qaa, E-AC-3, stereo (L,R), @48,  C      133,225 b/s  |
|  0x008C  Subtitles (fra, DVB subtitles for hard of hea  C       10,134 b/s  |
|  0x008E  Subtitles (fra, DVB subtitles, high definitio  C        2,966 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
|=============================================================================|

|=============================================================================|
|  PID: 0x0012 (18)                                                      EIT  |
|-----------------------------------------------------------------------------|
|  0x4E (78, EIT p/f Actual), TID ext: 0x0101 (257)                           |
|      Repetition: .... 1805  ms  Section cnt: ........ 7                     |
|      Min repet.: .... 1800  ms  Table cnt: .......... 3                     |
|      Max repet.: .... 1811  ms  Version: ............ 8                     |
|-----------------------------------------------------------------------------|
````

In my example I patch input stream with custom pat/pmt for add most services it's works like a charm !

````
CUSTOM PMT
|=============================================================================|
|  Service: 0x07D1 (2001), TS: 0x0001 (1), Original Netw: 0x20FA (8442)       |
|  Service name: France 2 (SVC 1), provider: GR1 B                             |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 6, PID's: 3 (clear: 3, scrambled: 0)                           |
|  PMT PID: 0x03E8 (1000), PCR PID: 0x0078 (120)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C          Unknown  |
|  0x0078  PCR (not otherwise referenced) ............... C          Unknown  |
|  0x03E8  PMT .......................................... C          Unknown  |
|  0x04D2  AVC video .................................... C+         Unknown  |
[... 5 lines cut ...]
|  Service name: France 2 (SVC 100), provider: GR1 B                           |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 6, PID's: 3 (clear: 3, scrambled: 0)                           |
|  PMT PID: 0x03E8 (1000), PCR PID: 0x0078 (120)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C          Unknown  |
|  0x0078  PCR (not otherwise referenced) ............... C          Unknown  |
|  0x03E8  PMT .......................................... C          Unknown  |
|  0x04D2  AVC video .................................... C+         Unknown  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
[... cut ...]
I have two choices i think
- Reinject my custom EIT but i have a lot of structure with service_id
- Remap the eit present on original stream input

The problem is when I want remap directly the eit are present in the original stream (input), how remap original for my all new services (SVC 1 - SVC 100) ? 

For "cannot using a "pmt_pid" for replace service_id" 
==> All service_id are distinct, but i have same PMT_PID for service 1 and 100, i cannot use PMT_PID on EIT structure ? (i have no idea for this possible)

antoineagscom, 2021-09-16:
I have already sort all events on one block service_id

[image]

lelegard, 2021-09-16:
If you change the service_id of a service, use plugin `svrename`. It will update the signalization consistently, including PAT, PMT and EIT's. So, you probably won't need to reinject a new PAT. If your alterations of the PMT are minor, use the plugin `pmt` instead of reinjecting a new PMT. If you want to reinject a new PMT anyway, at least the service was properly renamed first in the rest of the signalization.