tsduck/tsduck #871: [BUG] Compiling tables from xml to bin or using eitinject, doesn't respect table version attribute

peter-stanton, 2021-09-15:
#### Bug description:
I'm specifically interested in the EIT, but I expect this is the same for all tables. 
When injecting an EIT directly from an xml file (tsp -P inject), the resulting table version number is correct, however section numbers can't be specified, causing problems with the ETSI spec on present/following being one event per section.

So I've tried using tstabcomp and injecting the binary, plus I've also tried directly passing the xml to eitinject. Both of these methods wipe out the table version number specified in the xml and set it to zero. This then breaks streams with multiple EIT versions.

#### How to reproduce:
Create an eit xml file. Specify a version attribute other than 0.
run tstabcomp -c --eit-normalization --output eit.bin eit.xml
run tsp -I file input.ts -P inject --pid 0x12 eit.bin=1300 -s -O file output.ts
Inspect with DVB Inspector and note version number of the EIT is 0.

#### Expected behavior:
I expect the table compiler to set the version to the value in the xml file.

#### Errors and logs:
No errors.

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 21.04
 - TSDuck full version: 3.27-2438
 - Installation type: official binary

lelegard, 2021-09-15:
Hi @peter-stanton 

Indeed EIT's are special beasts. They are not complete tables according to MPEG rules. XML/JSON files can only represent complete tables, not aggregates of sparse and discontinuous sections as EIT's are. 

EIT's cannot be injected by the plugin `inject`, at least in a DVB-conformant way. The specialized plugin `eitinject` shall be used to inject EIT's. It manipulates events as input and rebuilds EIT sections at the right time according to DVB rules. Be sure to read its description in the user's guide. It is a bit complicated (just as DVB rules are on this topic).

peter-stanton, 2021-09-16:
My problem is that eitinject always sets the version to 0. You can set the attribute in the xml, but it ignores it

lelegard, 2021-09-17:
> My problem is that eitinject always sets the version to 0. 

You just found a bug, sorry about that. When the EIT generator changes the content of a section, the version number was not updated. I fixed it. So, you need to upgrade to the latest state of the repo.

It is still not possible to force a given version number from the XML file because it does not make sense. The pseudo-EIT from the XML file are just event containers. In the EIT structure, only the DVB-triplet (service id, ts id, original network id) is used to classify events in that container.

The first time a section is used, it is assigned version 0. Then, each time a section matching the same table id, service id and section number is updated, the section version is updated.

lelegard, 2021-09-17:
> For what I'm doing, it's easier to use tstabcomp --eit-normalization and then inject the binaries, which does work much better than eitinject. But that has the same issue with the version.

The `--eit-normalization` option also uses an instance of the EIT generator and consequently suffered from the same bug and now benefits from the same fix.

The command `tstabcomp --eit-normalization` produces a snapshot which is valid once only and not really suitable for injection since EIT sections are supposed to be continuously reorganized and regenerated. You need the `eitinject` plugin for continuous adjustment of the EIT's.

peter-stanton, 2021-09-23:
There is still another issue with eitinject. The p/f is good now, but something goes wrong with the schedule sections when an event changes.
Here's a screenshot of DVB inspector looking at the EIT schedule of my stream. The section at the top is the schedule for the first 60 seconds of the stream, at that point "First Event" ends and "Second Event" starts. The schedule sections then change, without changing the version number.

[image]
e 

On the transition, a copy of the three remaining events are added. I don't know if this output is caused by the version not changing (DVB Inspector thinks it's the same section, so it just appends the data?), but something is wrong.
By the way, the schedule doesn't have to change at all, it can keep events after they end.

lelegard, 2021-09-23:
> The schedule sections then change, without changing the version number.

Could you provide data and command to reproduce this? It will be impossible to analyze without a reproducer.

The output shows that the events are in the wrong order. This should not happen. Could you provide the EIT binary section so that we can see if this is an artifact from DVBinspector or a real error.

> By the way, the schedule doesn't have to change at all, it can keep events after they end.

It does not have to, but it can, as far as I know.

peter-stanton, 2021-09-24:
I've uploaded a test stream and the eit xml file I'm using.
https://file.io/ab2B4H7TPbfK

tsp -I file input.ts -P eitinject --terrestrial -f eit1.xml -b 451200 -w -O file eit.ts

Each service has 11 events: one before the stream starts, one that ends 30 seconds into the stream, then a load in the future.
What I see is that at 30 seconds the schedule changes without updating the version number. Playing it in VLC is fine, possibly because it doesn't care about version numbers, but DVB inspector shows weird things in the schedule and modulating it to a TV doesn't display the schedule properly.

peter-stanton, 2021-09-28:
That's great. Thanks for all your work.

closed 2021-09-29 by peter-stanton