tsduck/tsduck #873: [BUG] Impossible to install 3.28-2551 on RaspberryPi. But 3.28-2509 is ok.

moving-digital, 2021-09-23:
#### Bug description:
 Impossible to install 3.28-2551 on RaspberryPi. But 3.28-2509 is ok.
Error message reported with 3.28-2551 is:

```
Preparing to unpack .../tsduck_3.28-2551.raspbian11_armhf.deb ...
Unpacking tsduck (3.28-2551.raspbian11) over (3.28-2551.raspbian11) ...
dpkg: dependency problems prevent configuration of tsduck:
 tsduck depends on libsrt | libsrt1 | libsrt1-openssl | libsrt1.4-openssl; however:
  Package libsrt is not installed.
  Package libsrt1 is not installed.
  Package libsrt1-openssl is not installed.
  Package libsrt1.4-openssl is not installed.
```

#### Expected behavior:
Pretty confused as SRT is already included in 3.28-2509. Is that really a dependency issue?

lelegard, 2021-09-23:
Hi @moving-digital 

The message is pretty clear: ` tsduck depends on libsrt | libsrt1 | libsrt1-openssl | libsrt1.4-openssl`

So you need to install one of these packages. The actual name of the package unfortunately depends on the distro (Debian, Ubuntu, Raspbian) and its version. On Raspbian 11, the package is `libsrt1.4-openssl` (don't ask me why).

SRT support is implemented in the TSDuck `srt` plugin but it needs the SRT library underneath. The intermediate version 3.28-2509 was compiled on a Raspberry Pi where libsrt was not installed for some reason. When TSDuck is built on systems where DTAPI, SRT or RIST is not installed, the build succeeds but without support for the corresponding feature. If you try to use the `srt` plugin with your 3.28-2509, you will get an error message.

The final 3.28 version (3.28-2551) was built on a Raspberry Pi where `build/install-prerequisites.sh` was properly executed to make sure all dependencies were up to date. Libsrt was installed and SRT support automatically included.

If you want to check the dependencies of a package, use `dpkg -I`. See the differences between the two packages:
~~~
$ dpkg -I tsduck_3.28-2509.raspbian11_armhf.deb
 new Debian package, version 2.0.
 size 3854156 bytes: control archive=944 bytes.
     484 bytes,    16 lines      control
    1763 bytes,    14 lines   *  postinst             #!/usr/bin/env
 Package: tsduck
 Source: tsduck
 Version: 3.28-2509.raspbian11
 Maintainer: Thierry Lelegard <thierry@lelegard.fr>
 Architecture: armhf
 Section: free/video
 Priority: optional
 Depends: libc6,
  libpcsclite1,
  libcurl3 | libcurl4,
  libstdc++6,
  libgcc1
 Conflicts: tsduck-dev (<= 2.28.2441)
 Description: MPEG Transport Stream Toolkit
  TSDuck, the MPEG Transport Stream Toolkit, provides some simple utilities to
  process MPEG Transport Streams (TS), either as recorded files or live streams.
$
$
$
$ dpkg -I tsduck_3.28-2551.raspbian11_armhf.deb
 new Debian package, version 2.0.
 size 3861112 bytes: control archive=544 bytes.
     541 bytes,    17 lines      control
 Package: tsduck
 Source: tsduck
 Version: 3.28-2551.raspbian11
 Maintainer: Thierry Lelegard <thierry@lelegard.fr>
 Architecture: armhf
 Section: free/video
 Priority: optional
 Depends: libc6,
  libsrt | libsrt1 | libsrt1-openssl | libsrt1.4-openssl,
  libpcsclite1,
  libcurl3 | libcurl4,
  libstdc++6,
  libgcc1
 Conflicts: tsduck-dev (<= 2.28.2441)
 Description: MPEG Transport Stream Toolkit
  TSDuck, the MPEG Transport Stream Toolkit, provides some simple utilities to
  process MPEG Transport Streams (TS), either as recorded files or live streams.~~~

moving-digital, 2021-09-23:
Thanks again @lelegard 

After more thorough investigations, I found the root cause:

Even though I was using the command `sudo apt dist-upgrade -y`, somehow the OS would not upgrade to bullseye.

So I had to manually modify the `/etc/apt/sources.list` file, and now I can install TSDuck ! Thanks :-)

moving-digital, 2021-10-13:
Hello @lelegard 

After upgrading the Rpi to Debian 11 (Bulleseye), I can now indeed use TSDuck compiled for Raspbian11.

However, using Debian 11 or the Rpi4 is actually creating network issues when having more than 1 NIC interface connected to the Rpi. `dhcpcd.conf` does not work anymore, except for the on-board NIC.

Because of the above, I can't use my Rpi4 with Debian11 as I need more than 1 NIC per Rpi.

Yet, I would love being able to use the latest TSDuck version, but compiled for Raspbian10.

Would it be possible for you to look at compiling the latest TSDuck version for not only Raspbian11, but also for Raspbian10 ?

Thanks for your help in advance.

lelegard, 2021-10-13:
> Would it be possible for you to look at compiling the latest TSDuck version for not only Raspbian11, but also for Raspbian10 ?

Unfortunately not. I have one single Raspberry Pi and it runs Raspbian 11. All pre-built binaries are produced on the latest version of their respective distros. I do not have the resources the maintain more than that. You have to realize that TSDuck is a [one-person project](https://tsduck.io/download/docs/tsduck-project.pdf) and I already run 9 virtual machines on my Mac to build those binaries.

When using older versions or other distros, you have to rebuild TSDuck yourself ([instructions here](https://tsduck.io/doxy/building.html)).

closed 2021-10-31 by moving-digital